davidrapan/ha-solarman #1: [Bug] Solarman AC Temperature - Sensor missing

PV-Joe, 2024-07-02:
Hi David,

Just testing your integration in my productive system.

My Configuration:
DEYE Hybrid 8K SG04LP3 EU
PYTES E-Box Battery

I installed your integration parallel to the Solarman integration 1.4.1 by Stephen.

Result:
Many sensors have different names. That’s not a big problem and can be fixed manually.

I realized that the following sensor is missing:
Solarman AC Temperature
Could you please add it?

I‘m impressed by the speed of reloading the values. This integration is much faster.

Great work and many thanks!

PV-Joe, 2024-07-02:
Meanwhile I helped myself and added these lines to make AC and DC temperature for the Deye Hybrid work again:

`      - name: "DC Temperature"
        # disabled:
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        scale: 0.1
        rule: 2
        offset: 1000
        registers: [0x021C]
        icon: "mdi:thermometer"
        range:
          min: 0
          max: 3000
          
      - name: "AC Temperature"
        # disabled:
        class: "temperature"
        state_class: "measurement"      
        uom: "°C"
        scale: 0.1
        rule: 2
        offset: 1000      
        registers: [0x021D]
        icon: 'mdi:thermometer'
        range:
          min: 0
          max: 3000`

davidrapan, 2024-07-02:
Yes I did change names of some of the sensors. Now i see that I should have stated that! Thanks for pointing it out! Anyway naming of the sensors is in my opinion totally personal thing so feel free to adjust them accordingly to suit your needs!

I have planned for the future to transfer this customization into the configuration from HA.

PV-Joe, 2024-07-02:
Thanks for your reply.
Your integration works like a charm!

You are right. The DC Temperature seems to be stuck at 25 C but the AC temperature works:

[image]

Does anyone know the right DC temperature Modbus register value for the Deye Hybrid SG04LP3?

davidrapan, 2024-07-02:
It is the right one, at least according to the modbus doc available. My guess is that the sensor (in terms of hardware) is just not there. :)

davidrapan, 2024-07-02:
I just begin to look into that "sensor naming" matter and that DC Temperature sensor stayed in fact in the definition file but i marked it with "disabled:". That means that the sensor entity won't be added automatically with configuring of the device, but you can then enable this entity in the HA and it will stay enabled. ;)

PV-Joe, 2024-07-02:
I did enable it by putting a “#” in front of “disable:”

davidrapan, 2024-07-02:
Thats also an option. 😄 This feature works with HAs 'Disabled by integration' feature.
Sensor entity, which is the added into the HA looks like this:
[image]
And when you click on it you can enable it, by clicking on ENABLE:
[image]

bsafh, 2024-07-03:
I presume this does not work with SUN-10K-SG01HP3-EU-AM2 ?

That's a high voltage inverter,  serial no. starts with 23xxx, so presumably it's not protocol Version 5, but 4? Or another?

davidrapan, 2024-07-03:
@PV-Joe I made the changes to the definition file in the: [c6dfe87](https://github.com/davidrapan/ha-solarman/commit/c6dfe87f242afe43ca1535b4757f74b28a3057a6)

davidrapan, 2024-07-07:
So latest version of 'definition' files should match the original integration. If you still find any inconsistencies please open new issue with specific example.

Thank you and have fun!

closed 2024-07-07 by davidrapan

jouster1974, 2024-07-16:
just wanted to piost here that your new integration is fantastic and working better than the previous one, even just the update speed makes it all feel better.

Im running mine on a Solis inverter so if youfd like anythihng tested or info from me, please do let me know

Here are my sensors...no batteriries yet but that will hopefully be coming in the not to distant future

[image]

davidrapan, 2024-07-16:
Thank you for your kind words. And I'm glad it works nicely for you!

If you want you can open discussion at [Discussions](https://github.com/davidrapan/ha-solarman/discussions) and we can together polish some of your sensors to be more descriptive if you want (like for example status sensor etc.)

Anyway, thanks again and as always have fun!