davidrapan/ha-solarman #2: Overriding Stephen integration

frsantos, 2024-07-02:
Hi David,

Just saw your new integration and I'd like to test it, but then I just read this comment

> And about the override, it works like this: Inverter with same serial number will always override the existing one. This way it works also in the version from Stephen and i kept it that way. Sorry about the inconvenience!

Would be any way of testing this in parallel? I don't like the idea of this overwriting names or values from the other working integration.

Thank you!

PV-Joe, 2024-07-02:
Yeah, it took me quiete a while to change the sensor names to the integration by David so my dashboards work again.

davidrapan, 2024-07-02:
Before you start it up with my code, you can go through the deye_sg04lp3.yaml and "rematch" the sensor names to suit your needs.

But running both versions side by side (i.e. to compare them) is not possible at the moment. But switching between them should be possible as described above.

_As I am now thinking about it, you know what I could add more versions of the definition file for this particular device (which would also involve the original one) ;) I look into it when I have some time to spare. ;)_

davidrapan, 2024-07-02:
I have a question on how to approach this... Do you want also definition file which will contain only original list of sensors? Or file (which will be the 'default' one) that will contain sensors with original naming + all the new sensors?

PV-Joe, 2024-07-04:
Hi David,

after 3 days of testing an rematching the sensor names I realized that all my historical data from the energy dashboard was gone.

Obviously I don’t want to loose all the history, so I restored my backup to the date before I installed your version.

All my historical data from the energy dashboard is there again with a loss of 3 days but I can live with that, rather than loosing all of it.
Rematching the sensors to the original names by Stephen was to confusing for me.

I thought to try your latest version and hoped that this would solve the problem overwriting the old sensor names.
It did not, so I restored once again back to the original from: https://github.com/StephanJoubert/home_assistant_solarman

As I much prefer your version as it has more sensors and is much faster pulling the data I would kindly ask you to use the original sensor names by Stephen.

This would make it much easier for me and others to upgrade to your version without doing all the conversion of sensor names.

I also realized that the Solarman configuration page is faulty:

[image]

I wasn’t able to select my “custom_components.yaml” in the Options.

I hope you can make it more convenient for non professionals to use your integration overwriting the original by Stephen.

I will come back to your integration and give it another try when it is more hassle free. 

Keep up the good work and hope to be able to use your Solarman in future. I might even go for a name conversion myself if I have more time but I’m scared that the deye_SG04LP3.yaml will be overwritten again by future updates.

Joe

PV-Joe, 2024-07-04:
If it will help you to fix compatibility for the Solarman Deye Hybrid integration , here is the YAML I use with the integration 1.4.1 by Stephen Joubert:

`# SUN-8/12K-SG04LP3-EU | 8/12KW | Three Phase | 2 MPPT | Hybrid Inverter | Low Voltage Battery
# tested with LSW3_15_FFFF_1.0.91R + LSW3_15_FFFF_1.0.84

requests:
  - start: 0x0003
    end: 0x0059
    mb_functioncode: 0x03
  - start: 0x0202  
    end: 0x022E
    mb_functioncode: 0x03 
  - start: 0x024A  
    end: 0x024F
    mb_functioncode: 0x03  
  - start: 0x0256
    end: 0x027C
    mb_functioncode: 0x03    
  - start: 0x0284  
    end: 0x028D
    mb_functioncode: 0x03   
  - start: 0x02A0  
    end: 0x02A7
    mb_functioncode: 0x03
  
parameters:
 - group: solar
   items: 
    - name: "PV1 Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x02A0]
      icon: 'mdi:solar-power'

    - name: "PV2 Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x02A1]
      icon: 'mdi:solar-power'

    - name: "PV1 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x02A4]
      icon: 'mdi:solar-power'

    - name: "PV2 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x02A6]
      icon: 'mdi:solar-power'

    - name: "PV1 Current"
      class: "current"
      state_class: "measurement"
      uom: "A"
      scale: 0.1
      rule: 1
[... 431 lines cut ...]
      uom: ""
      scale: 1
      rule: 1
      registers: [0x0011]
      isstr: true

    - name: "Control Board Version No."
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 1
      registers: [0x000D]
      isstr: true

 - group: Alert
   items: 
    - name: "Alert"
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 6
      registers: [0x0229,0x022A,0x22B,0x022C,0x022D,0x022E]
`

davidrapan, 2024-07-04:
Are you sure you used the latest version? Cause you can look in the file and see that the sensor names are the same though. :) (ofc if I didn't forget any)

PV-Joe, 2024-07-04:
Hi David,

That sounds great! You are tempting me to give it another try. I’m ready to check it out as soon as possible and will let you know if it works well. I’m backup before, so if anything goes wrong it will be possible to roll back. 

Thanks for letting me know!

davidrapan, 2024-07-07:
So latest version of 'definition' files should match the original integration. If you still find any inconsistencies please open new issue with specific example.

Thank you and have fun!

closed 2024-07-07 by davidrapan

frsantos, 2024-07-14:
Hi @davidrapan 

Sorry, I went on vacation and just looking back at this.

My initial suggestion was to have both integrations working side by side, so I could test it before migrating to this one. However, HACS installs both under the same directory as far I can see, so the code (and not only inverter definitions) will overwrite the old integration, making comparing sensors impossible.

[image]
[image]

Could it be possible that HACS downloads this component into a different path? 

Thanks!

davidrapan, 2024-07-14:
Hi, just simply downloading the integration into different path probably won't work due to how are unique entity ids generated (It probably won't add new entities...).

But if you still want to try it you will have to fork this repository and:
- Change the folder name in 'custom_components/solarman' to 'custom_components/anything_else'
- Change 'DOMAIN' and 'SENSOR_PREFIX' from const.py to match the change above
- Change 'domain' and 'name' to match the changes above
- In readme.md change link under INSTALLATION to match link to your forked repository

Then you will able to use the BLUE button :)

Hope it help and as always have fun!

davidrapan, 2024-07-18:
Hi @PV-Joe, as I remember you wasn't really happy from some changes in sensor naming schemes cause of loosing history etc.

So I decided to start wiki and in your honor start with [Naming Scheme ‐ sensor renaming](https://github.com/davidrapan/ha-solarman/wiki/Naming-Scheme-%E2%80%90-sensor-renaming) as an help to mitigate any inconveniences (or shortcomings) of said changes!

Hope it helps and as always have fun guys!

PV-Joe, 2024-07-18:
> Hi @PV-Joe, as I remember you wasn't really happy from some changes in sensor naming schemes cause of loosing history etc.

Thx, that will help once I have time to give it another try.

I reverted back to Stephen Jouberts Version 1.4.3. This Version worked very stable for my Deye Hybrid 8K EU for almost 2 years even though it just updates every 30 seconds.

Onother thing is that I have InfluxDB with Grafana as Frontend running and I‘m curious if my database will be blown up by collecting data every 5 seconds.

I’m still watching your integration since you started and hope for more features and sensors to make it even more tempting to upgrade.

Is it possible to use the original definition file from Stephen and will it be updated automatically after upgrading to your integration? I‘m a bit scared to mess it up again.

Keep going,
Joe

davidrapan, 2024-07-18:
Hi @PV-Joe, as sensors are defined only using said profiles it will only create sensors (with names) according to that profile. Profile (definition file) is what names and controls sensors. That means my integration + your profile will create exactly the same sensors as you have with Stephan's integration.

When comes to data update intervals in DB world especially Time Series DBs as InfluxDB is difference between 30s and 5s almost non existent.

And even then the frequency between state changes can be in HA controlled using 'recorder:' in 'configuration.yaml'

Hope it answers all your questions and clearly as possible!

Have fun!