davidrapan/ha-solarman #4: [request] Solarman - Status last update is missing - please add it to next release

PV-Joe, 2024-07-03:
Hi David,

could you please add the sensor “Status last update” into your integration?
I had it before from Stephen’s integration and it is missing now.
Not a bug but more of a request.

Keep up the good work!
Joe

davidrapan, 2024-07-03:
Hi Joe,
Look into readme under Diagnostic section, maybe that helps. 😉

_Of course you have to update those entity names according to your name of the device._

PV-Joe, 2024-07-03:
I have seen that code but I thought that it would be easier to implement it to your integration.

Your Integration works very stable by the way. 👍

davidrapan, 2024-07-03:
The thing is, I'm using for info about "last updated" HA's built in "last_updated" param of the sensor states. So you have the info there already. The way how I approached the implementation was to maximally utilize the scaffolds of the HA itself, rather than reinvent and create "my own" things and so on.

When comes to the "replication" of the provided config inside the integration itself I would be really happy if I knew how to do that, some way easily. But I don't yet.. 😄 and I didn't had the time to investigate more. Now I'm more invested in the figuring out why one of my older versions of the code works way better than the one committed here (seemingly without any major changes) lol, that troubles me somewhat more.

But I'll put it on my list! ;)

Have fun.

davidrapan, 2024-07-07:
Last update of the data can be seen here:
[image]

Have fun.

closed 2024-07-07 by davidrapan