davidrapan/ha-solarman #5: Does it work with high voltage inverters?

bsafh, 2024-07-03:
I presume this does not work with SUN-10K-SG01HP3-EU-AM2 ?

That's a high voltage inverter, serial no. starts with 23xxx, so presumably it's not protocol Version 5, but 4? Or another?

Futhermore the sensors/registers are different for the HV inverters...

davidrapan, 2024-07-03:
Hi,
first I have to ask where are you coming from? Are you a "first time user" or you are already using implementation from the @StephanJoubert? Cause theoretically this integration should work for all the same devices.

That being sad, it's really tested only for the device I own. So it has to be tried to be really sure.

On the other hand if definition file for your device is missing but you have the registry list i could try to add it. ;)

Wish I could help you more.

Have fun.

bsafh, 2024-07-03:
I'm trying to use the implementation from the @StephanJoubert, but that does not work for me either. 

Over in his repository there is a discussion about that, and someone has supplied an incomplete custom_parameters.yaml for the high voltage inverters (everything related to time of use is missing,  e.g.)

But I can't get that far, as the integration does not connect to the wifi dongle at all, the serial number of the inverter starts with 23xx, the serial number of the dongle starts with 29xx - both are not listed as working with protocol V5 in the readme from https://github.com/StephanJoubert. There is an omnious remark to try v4, but it's not explained anywhere how to do this.

The inverter is registered with Solarman Business App, and all values show up there.

bsafh, 2024-07-03:
```
Inverter information
Inverter serial number 2312122xxx
Firmware version (main)
Firmware version (slave)
```
```
Device information
Device serial number  2908548xxx
Firmware version LSW3_15_FFFF_1.0.9E
```

bsafh, 2024-07-03:
I'll try tomorrow ...

bsafh, 2024-07-03:
I don't have the register list directly, but the definitions can be found in
https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_common.py and https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_hv.py combined.

bsafh, 2024-07-03:
And there is someting here:

- https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg01hp3.md
- https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg01hp3_battery.md

https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg01hp3_bms.md
https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg01hp3_ups.md
https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_settings.md

bsafh, 2024-07-03:
Actually I found the register list

[MODBUSRTU三相高压储能通信规约V104（高压）(3)(1)1111_SG01-HP3-AM2.pdf](https://github.com/user-attachments/files/16090082/MODBUSRTU.V104.3.1.1111_SG01-HP3-AM2.pdf)

davidrapan, 2024-07-03:
Great, I'll try to integrate it as soon as I can but we still need to find out if there is even any communication happening (cause of that V5 or V4 matter).

davidrapan, 2024-07-04:
@bsafh I added file for your device in: [b53e902](https://github.com/davidrapan/ha-solarman/commit/b53e9025d5346c7d33a79828e0b0ddc0b2136358)

bsafh, 2024-07-04:
tested it, using "main" in the version dialog (HACS download): protocol 5 works!

But ... config flow is ... not commentet, so I have no idea what to fill in:

[image]

bsafh, 2024-07-04:
I managed to get around this by just copying the file deye_sg01hp3.yaml over custom_parameters.yaml (which was alread configured from the now overwritten config flow from @StephanJoubert 's version.

It does recognize the inverter :-)
And it does read the new definitions.

But it does not contain any "Time of use" information, neither the switcg for that nor the timetable itself. And there are no switches (e..g switch priority_load on and off - oops, priority_load is not even there?)

[image]
[image]
[image]
[image]
[image]

davidrapan, 2024-07-04:
Time of Use is not read as far as I know in any of the inverter definition files yet.

Instead of reconfiguring, try remove and then add device again. I'll look into why configuring flow is missing the config values description.

But I'm glad that it at least tries to read any values, that's a good sign.

bsafh, 2024-07-04:
> So I advise to you really re add the device.

how do I get rid of the old devices?

For the switches I have found this:

[deye-modbus.zip](https://github.com/user-attachments/files/16099985/deye-modbus.zip)

davidrapan, 2024-07-04:
It's really not about the registry addresses but about implementation in the integration. There are not entities for it. I do want to eventually add support for as much things as possible but my focus was currently on the connection stability.

Anyway for removing the device:
[image]

Mights001, 2024-07-05:
Hi i use this integration for a HV 50KW Deye too, works very well. Some Entitys you have to set a other Scale is in the Documentation. I will attached this new Version what i get from Deye since 2 Weeks. 
[【b】30240301001287-使用说明书-储能三相-sun-(25-50)k-sg01hp3-eu-bm4-德业英文-封面彩印-封面封底157g-内页70g铜版纸-142.5x210mm-ver2.2.pdf](https://github.com/user-attachments/files/16107073/b.30240301001287-.-.-sun-.25-50.k-sg01hp3-eu-bm4-.-.-.157g-.70g.-142.5x210mm-ver2.2.pdf)

davidrapan, 2024-07-05:
Hi there, thanks for your comment. I was trying to adjust the scaling according to the previus doc but is possible I missed some of them so if that's the case let me know which one and I'll correct them. 😉

Mights001, 2024-07-06:
for Example 
- Battery Voltage LV 0.01 - Battery Voltage  HV 0.1

You can find it in the Manual on the 5t Row
[image]

Also what i add to your Config is the Gen Port, because a many useres from my side need this for the Micro Grid Input. 

[image]

davidrapan, 2024-07-07:
Hi, thanks for your comment, I'll look into that. 😉

Have fun!

davidrapan, 2024-07-14:
Hi guys, would be some of you willing to track for me different numeric states for "* Device Relay Status" (register: 0x0228) entity?

What i saw so far was 6 and 15, any idea what they could mean? Cause they do not match states from my LV device and not even match the modbus doc for HV device...

Would be highly appreciated as this register is used for detecting Power Outages.

Thanks and have fun!

davidrapan, 2024-07-14:
> for Example
> 
> * Battery Voltage LV 0.01 - Battery Voltage  HV 0.1
> 
> You can find it in the Manual on the 5t Row ![grafik](https://private-user-images.githubusercontent.com/35522918/346296678-a9cb909a-be1b-4b66-93a1-466f6fbae162.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA5ODkwMTIsIm5iZiI6MTcyMDk4ODcxMiwicGF0aCI6Ii8zNTUyMjkxOC8zNDYyOTY2NzgtYTljYjkwOWEtYmUxYi00YjY2LTkzYTEtNDY2ZjZmYmFlMTYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE0VDIwMjUxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFjODZmNzlhNWZkZjFlZjBlNmRiMDNlMjAxMzYwMmFmZDIwZjY0YjY3YWYwNTM0YmMzMWE2NDJkZWQxYzc5YTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.EynW4VPWqgvCxKXjqOZySNkNf4Dk_s0lbYl5k5-eXb0)
> 
> Also what i add to your Config is the Gen Port, because a many useres from my side need this for the Micro Grid Input.
> 
> ![grafik](https://private-user-images.githubusercontent.com/35522918/346296879-22f418b5-43eb-4614-9235-a94e72c87b9c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA5ODkwMTIsIm5iZiI6MTcyMDk4ODcxMiwicGF0aCI6Ii8zNTUyMjkxOC8zNDYyOTY4NzktMjJmNDE4YjUtNDNlYi00NjE0LTkyMzUtYTk0ZTcyYzg3YjljLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE0VDIwMjUxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU1ODY2MmU2MjE4NWRhY2IyNDZjY2UyNjI4OThlYWNhM2VjOTdlNTU2ODlkYWI2OWE5NWYxOTQ0Mjk3MGRiM2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zdeVMrLQVoGpcMSVGOMVNKtHaAHj0ecfTjPARY37GJ8)

I added entities for SmartLoad/Generator/Microinverter power and voltage in [feat: Add SmartLoad/Gen voltage and power entities](https://github.com/davidrapan/ha-solarman/commit/f1b80fae5b805258a4516fbffe0521719fe8b46c)

_Naming of the entities are still subjects to change as those entities could have meaning in three different contexts..._

davidrapan, 2024-07-21:
All the necessary configurations is now possible using this integration (for SG04LP3 and SG01HP3).

But there is still plenty (not so crucial) to implement. So if let me know guys if forget any important ones.

And I have to say that working with the solarman HW is pain in the ass. I proly never saw something implemented so poorly... That being said setting of those configurables sometimes don't work very smoothly. I will try to improve on that in the future. 😉

It will be all part of next release which is scheduled on: 24.07.24

Cheers!

closed 2024-07-25 by davidrapan