davidrapan/ha-solarman #9: Settings Parameters

Mights001, 2024-07-05:
Hi, in the YAML File from this integreation are also Settings Parameters, but if i add the Inverter i get only the Sensor Entitys. 

[image]

Is there any hidden option ?

davidrapan, 2024-07-05:
Hi there, some of them are as an attributes attached to the Battery sensor. Click on the sensor, and at the bottom of the detail view click on attributes.

Have fun.

Mights001, 2024-07-06:
Hi, so i did a lot of them with Esp Home and i love it the speed of your Integration. In any case i have time to can help you with the register let me now. 
[image]
[image]

davidrapan, 2024-07-07:
Hi, that would be really great! I'm currently working on implementing configuration of the inverter so anyone willing to test the functionality in the next release will be highly appreciated!

Thanks and have fun!

Summer-of69, 2024-07-08:
Hi David!
Compliments for the great work. 👍I've been using your integration for about a week now. Works perfectly. Also, I use an ESP Home variant and can also configure the Deye with it. I look forward to this expansion in your solution!

davidrapan, 2024-07-10:
Hi!

Thank you for your kind words! And sure, I will let you guys know in this issue tracker when there are any updates in this regard!

Thanks again and have fun!

davidrapan, 2024-07-18:
Battery parameter settings were added in [feat: Battery parameters made configurable](https://github.com/davidrapan/ha-solarman/commit/6d3367d7260bfdb4f85a592d2afe596852b766fb)

davidrapan, 2024-07-20:
Can please any of you guyes show me how are those sensors:
[image]

Presented in the device detail page? 😆 

Cheers!

Mights001, 2024-07-20:
Hi, i just make this with Helpers and Nodered behind, in the detail View it looks so 
[image]

And in Nodered i manipulate the Helpers 
[image]

davidrapan, 2024-07-20:
Ou my, that's total jungle for me! 😆 But thanks!

Anyway when comes to an update of imlementing the controls within integration:
From Work Mode are left to do: Times and Grid/Gen charging.

General question:
That charging from grid and gen is made as an selectable enum on the inverter side. Is it a problem that it won't be a simple switch for both Grid/Gen separatedly?

davidrapan, 2024-07-21:
All the necessary configurations is now possible using this integration (for SG04LP3 and SG01HP3).

But there is still plenty (not so crucial) to implement. So if let me know guys if forget any important ones.

And I have to say that working with the solarman HW is pain in the ass. I proly never saw something implemented so poorly... That being said setting of those configurables sometimes don't work very smoothly. I will try to improve on that in the future. 😉

Mights001, 2024-07-24:
Hi, where can i find this controlls? I saw this in the yml from you but under Device i can see only the sensors.

davidrapan, 2024-07-24:
You can't see the controls under device detail?

Mights001, 2024-07-24:
No only the Sensors 
[image]

davidrapan, 2024-07-24:
What device are you using?
[image]

davidrapan, 2024-07-24:
Ah you have to update to latest main. I can see that you are proly using latest realse version. You either have to update manually or wait cause another release is scheduled for today! 😉

Mights001, 2024-07-24:
Hi, yes i made the update since 2 hours 

[image]

I use the Battery Safety Controller and sometimes is a Alarm coming, we don't know at the moment what this is. But the System is working perfect :) 
https://github.com/shining-man/bsc_fw

davidrapan, 2024-07-24:
I'm gonna release v24.07.24 today so wait a sec for that. (HACS only downloads latest release which is from a 10 days ago 😉 )

closed 2024-07-25 by davidrapan

Mights001, 2024-07-26:
Hi, is still the same i have only the Sensors
[image]

I try also to delet all and make it new, but nothing is changing

davidrapan, 2024-07-26:
What device? What inverter profile?

davidrapan, 2024-07-26:
If you are on the latest version and still don't see any new entities you are probably having different device than (SG04LP3 or SG01HP3) or using your custom inverter definitions profile.

Mights001, 2024-07-30:
Sorry for the late reply i use your profile 
[image]

I also try to delet all Integration and Settings but i always get only the Sensors.

davidrapan, 2024-07-30:
What version are you running?

Mights001, 2024-07-30:
The latest one 
[image]

Mights001, 2024-07-30:
I think, there is something not delete from the old Solarman integration, because if i download the new one from you and i add a new device the always knows the parameters Dongel ID and IP Address, do you know where this is stored? i search in old the folders from HA but cant find it

davidrapan, 2024-07-30:
First delete and re-add your device and use rather that original deye_sg04lp3 cause that custom one is now gone.

Then try to turn on that "Show beta versions" and then select the latest one a try download it...

Restart.

davidrapan, 2024-08-05:
Hey @Mights001 were you successful?

Mights001, 2024-08-05:
Hi, no i try all what i can do, but still the same thing only sensors are there

davidrapan, 2024-08-05:
Did you update to the latest 24.08.04?

Mights001, 2024-08-06:
yes but nothing change

davidrapan, 2024-08-06:
What profile are you using? @Mights001 

Show me your HA, installed integration, your device, it's detail page, etc.

davidrapan, 2024-08-07:
@Mights001 well.. let me know when you will time to look into it. :)

Mights001, 2024-09-09:
Hi, I've been busy for a month and haven't had time for this. I just tried it again with the latest version and look, it works now. 
;-) 
THX

davidrapan, 2024-09-09:
I'm glad to hear that! 😉