davidrapan/ha-solarman #10: [request] please add sensor: Solarman - Daily Total Consumption to match the value in energy dashboard and Solarman App

PV-Joe, 2024-07-07:
Hi David,

could you please add a sensor that shows exactly the same value for the “Daily total consumption” as the value in the energy dashboard?

The KWh value in the official Solarman App, Deye App and Energy Ease App for mobile devices show the same value for Daily total consumption as the Energy dashboard in Home Assistant.

The already available “Daily Load Consumption” doesn’t match wit the “Daily total consumption”. This value is slightly lower.

Adding this sensor would be highly appreciated. Alternatively could you make it available as a template sensor? I have no clue how generate the code for it.

Joe

PV-Joe, 2024-07-07:
Hi again,

these values confuse me:

Daily consumption from Energy Ease App:
[image]

Daily consumption from HA energy dashnoard:
[image]

Daily load consumption from Solarman Modbus:
[image]

I thought that the load consumption with the battery discharge added leads to the right value, but it doesn’t:
[image]

Hope you know what I mean now.
I want a sensor that is exactly the same.

if it is not possible because of your discription above, never mind.

Anyway, thx for helping and trying to pimp the integration to max it out.

Joe

davidrapan, 2024-07-10:
To be clear, the integration don't make up any values, nor it calculates them in this particular example, it takes the data from the inverter, and presents them for you. If there are any errors in that regard it has to be due to some misconfiguration.

That being said to debug this issue... Could you please provide again the same screenshots from the same time, obviously and also add data from inverter display itself? From the screen where is visible Today Load Consumption (It should be under house icon).

Have fun.

PV-Joe, 2024-07-10:
Hi David,

Thy for explaining.

Closed 
[image]

davidrapan, 2024-07-10:
I think I finally figured out what this was all about. So the mentioned sensor 'X Load Consumption' only shows the load consumption of the house. But what Energy dashboard or Solarman App shows you is load consumption of the house + what inverter consumed + all the other potential losses! In the yaml for SG04LP3 are additional sensors which also provides information about those losses! 😉

Have fun and stay sharp!

closed 2024-07-10 by davidrapan

PV-Joe, 2024-07-10:
Yes I think you probably got it.
I never understood why there is no sensor that shows the same value as the one in the energy dashboard. I understand that your integration is just reading modbus registers and can‘t add calculated values.

Thanks for picking up my comment for another explanation. 😄

You are IT Kram diskutieren über übers Hecken von dem Wechselrichter unten im Keller, der baut

davidrapan, 2024-07-10:
The Energy dashboard just sums up Exported/Bought energy and Produced energy and then shows you the Consumed amount. It does not even look at Load Consumption because it does not have to. 😉 

But I also wanted to see this losses as an sensor so what I did is:
Total losses = Total Energy Exported/Bought + Total Produced + Total Battery Discharge - Total Energy Imported/Sold - Total Battery Discharge - Total Load Consumption

And the same for daily/today:
Daily losses = Daily Energy Exported/Bought + Daily Produced + Daily Battery Discharge - Daily Energy Imported/Sold - Daily Battery Discharge - Daily Load Consumption

This will give you all the energy consumed by the inverter itself + also all the losses by AC/DC conversions (Battery charging/discharging).

What I don't understand is why those are not provided by the inverter itself. 😄

_Or maybe I do understand, they probably don't want to directly show how inefficient the inverter is. 😄_

You welcome, I'm always trying to be as helpful as possible. 😉

PV-Joe, 2024-07-20:
Meanwhile I firured out how to match the values for "Daily Consumption" from Solarman APP with Solarman for Home Assistant.

To add a newly created template sensor, just copy this code into your configuration.yaml:

```
  # Sensor template for:
  # "Solarman Daily Energy Consumption" to match the value from Solarman APP
  # To install: Add lines to your "configuration.yaml" and restart
  # The sensor names match for: DEYE Hybrid Inverter "sg04lp3.yaml"
  # Code by PV-Joe (2024)
  - platform: template
    sensors:
      daily_energy_consumption:
        friendly_name: "Solarman Daily Energy Consumption"
        unique_id: solarman_daily_energy_used
        unit_of_measurement: 'kWh'
        device_class: "power"
        value_template: >
          {{
            (states('sensor.solarman_daily_production')|float(0) +
            states('sensor.solarman_daily_energy_bought')|float(0) +
            states('sensor.solarman_daily_battery_discharge')|float(0) -
            states('sensor.solarman_daily_energy_sold')|float(0) -
            states('sensor.solarman_daily_battery_charge')|float(0))
            | round(1)
          }}
```

davidrapan, 2024-07-20:
Yes I calculate energy losses this way. You can even do this sensor from within inverter definition file in my integration I support simple math operation with registers. ;)

PV-Joe, 2024-07-20:
I like the way to have a sensor that adds up Load Consumption and Losses and in addition your seperated sensor for Losses values as well.

Your dashboard is brilliant. I love it!

**Maybe you could share the configuration and the yaml code for your dashboard  in your Wiki.
I could imagine more people are interested in the professional look and feel of these amazing gauges.**

Well done!

davidrapan, 2024-07-20:
> I like the way to have a sensor that adds up Load Consumption and Losses and in addition your seperated sensor for Losses values as well.

I understand.. cause you are also using InfluxDB and that means you don't have the value there... makes sense!

> Your dashboard is brilliant. I love it!
> 
> **Maybe you could share the configuration and the yaml code for your dashboard in your Wiki. I could imagine more people are interested in the professional look and feel of these amazing gauges.**
> 
> Well done!

Thank you! It took a lot of effort so I'm glad you like it.
Anyway I have it planned in milestones here: [Start Wiki](https://github.com/davidrapan/ha-solarman/milestone/2) 😉

Cheers!

davidrapan, 2024-07-20:
I extended the wiki with those sensors for you: [Today/Daily Consumption [kWh]](https://github.com/davidrapan/ha-solarman/wiki/Custom-sensors#today-consumption-kwh) and [Total Consumption [kWh]](https://github.com/davidrapan/ha-solarman/wiki/Custom-sensors#total-consumption-kwh)

davidrapan, 2024-07-20:
[Dashboards](https://github.com/davidrapan/ha-solarman/wiki/Dashboards) 😉