davidrapan/ha-solarman #11: insert own yaml fail

amiko24, 2024-07-07:
Hi David,
I have created my own yaml file and added it to the folder "inverter_definitions". I can't select my yaml fail in the configuration because the name is not displayed. At StephanJoubert all inverter yaml are listed in the fail const.py.  
In which file can I find the list of inverter yaml files?

amiko24, 2024-07-07:
[image]

PV-Joe, 2024-07-07:
I have tried to load an individual configuration without success. The file just doesn’t show up in the configuration selector.

I duplicated an existing file and renamed it for backup purposes and overwrote the original file with my custom configuration.

Maybe it helps as a workaround until we know more?

amiko24, 2024-07-07:
Good idea, thank you.

davidrapan, 2024-07-10:
Hi guys, from the screenshot provided it's obvious you are not using latest version. Please kindly update the latest code and try again.

Have fun!

closed 2024-07-12 by davidrapan