davidrapan/ha-solarman #12: Update Interval

Summer-of69, 2024-07-08:
Hi David!
Where can I change the update interval? Currently the data is changed every 5 seconds.

Thanks Barnie

davidrapan, 2024-07-10:
Hi Bernie, the update interval of the integration is somewhat hardcoded and due to many reasons is not really possible to make it any lower.

That being said if you want to fine control update intervals of specific sensors you can do it inside inverter definition yaml file by setting for example 'update_interval: 30' which translates to update every 30 seconds or 'realtime:' which translates to every 5 seconds which is integration minimum. **But you have to remove registry sets at the start of the yaml file (if it's present) which are used to determine requests which activates dynamic requests.**

You can see all that in practice for example in the 'deye_sg04lp3.yaml' file.

Have fun!

davidrapan, 2024-07-10:
The possibility of too often never even occurred to me.. AS there for sure are sensor which don't need to be requested often which are for example those 'Total' ones.. But for example sensors which shows for example PV Power I wanted to always have the most accurate value possible.

Thank you for your kind words! And as always, have fun!

closed 2024-07-12 by davidrapan

MarkoSchweizer, 2024-11-06:
Hi, 
I tried to change the interval to realtime, but it didnt worked out.
Here is what i did:

#
# Sofar G3 also HYD 5-20KTL-3PH
#
# Reference: https://github.com/user-attachments/files/17180102/SOFAR-G3.External.Modbus.Protocol-EN-V1.10.20220622.xlsx
#
# This works also for rebranded ZCS Azzurro 3-Phase inverters such as the 3PH HYD6000 ZSS, or single phase such as 1PH HY6000 ZSS HP
# Note that this won't work if your ZCS inverter is connected via Connext, you have to be using a Wi-Fi or Ethernet Kit such as ZSM-WIFI-USB.
#

default:
  #update_interval: 10
  realtime:
  digits: 6

parameters:
  - group: Inverter
    items:
      - name: "Inverter status"
        rule: 1
        registers: [0x0404]
        lookup:
          - key: 0
            value: "waiting"
          - key: 1
            value: "detection"
          - key: 2
            value: "grid-connected"
          - key: 3
            value: "emergency power supply"
          - key: 4
            value: "recoverable fault"
          - key: 5
            value: "permanent fault"
          - key: 6
            value: "upgrade"
          - key: 7
            value: "self-charging"
        icon: "mdi:wrench"
      - name: "Ambient temperature 1"
        class: "temperature"
        uom: "°C"
        rule: 2
        registers: [0x0418]
        icon: "mdi:thermometer"
      - name: "Ambient temperature 2"
        class: "temperature"
        uom: "°C"
        rule: 2
        registers: [0x0419]
        icon: "mdi:thermometer"
      - name: "Radiator temperature 1"
        class: "temperature"
        uom: "°C"
        rule: 2
        registers: [0x041A]
        icon: "mdi:thermometer"
      - name: "Radiator temperature 2"
        class: "temperature"
        uom: "°C"
        rule: 2
        registers: [0x041B]
        icon: "mdi:thermometer"
      - name: "Radiator temperature 3"
        class: "temperature"
        uom: "°C"
        rule: 2
[... 1304 lines cut ...]
            value: "ID192 PLC module heartbeat loss"
      - name: "Storage Control Mode"
        platform: select
        rule: 1
        registers: [0x1110]
        lookup:
          - key: 0
            value: "Self Use"
          - key: 1
            value: "Time of Use"
          - key: 2
            value: "Optimized Revenue"
          - key: 3
            value: "Passive"
          - key: 4
            value: "Peak Shaving"
          - key: 5
            value: "Off-Grid"
          - key: 6
            value: "Generator"
        icon: "mdi:power-settings"

What did i wrong?

Greets

davidrapan, 2024-11-06:
Do:
```
default:
  update_interval: 5
  digits: 6
```

MarkoSchweizer, 2024-11-06:
> Do:
> 
> ```
> default:
>   update_interval: 5
>   digits: 6
> ```

Works faster, thanks!!!

MarkoSchweizer, 2024-11-17:
Hi, can I update just certain register? I dont need all register fast, just 2-3 register.