davidrapan/ha-solarman #15: problems with 2 inverters, maybe not really a bug

Kedryn, 2024-07-10:
I have 12 inverters
all of them are ZCS, same device, same firmware version
all of them are accessible via thir web interface
i've configured all of them:
[image]

But 2 of them aren't working.

In the debug log i get :
```

2024-07-10 09:24:21.419 INFO (SyncWorker_4) [custom_components.solarman.solarman] Connecting to solarman data logger 10.1.80.109:8899
2024-07-10 09:24:22.933 WARNING (MainThread) [homeassistant.components.sensor] Updating solarman sensor took longer than the scheduled update interval 0:00:30
2024-07-10 09:24:23.027 WARNING (SyncWorker_36) [custom_components.solarman.solarman] Querying [1024 - 1067] failed with exception [Empty: ]
2024-07-10 09:24:23.028 INFO (SyncWorker_36) [custom_components.solarman.solarman] Disconnecting from solarman data logger 10.1.80.112:8899
2024-07-10 09:24:23.039 WARNING (SyncWorker_36) [custom_components.solarman.solarman] Querying [1024 - 1067] failed, [0] retry attempts left
2024-07-10 09:24:23.039 WARNING (SyncWorker_36) [custom_components.solarman.solarman] Querying registers [1024 - 1067] failed, aborting.
2024-07-10 09:24:23.501 WARNING (MainThread) [homeassistant.components.sensor] Updating esxi_stats sensor took longer than the scheduled update interval 0:00:15
2024-07-10 09:24:28.636 WARNING (SyncWorker_4) [custom_components.solarman.solarman] Querying [1154 - 1188] failed with exception [CRCError: CRC validation failed.]
2024-07-10 09:24:28.636 INFO (SyncWorker_4) [custom_components.solarman.solarman] Disconnecting from solarman data logger 10.1.80.109:8899
```

Any advice on what i can check to fix that?

davidrapan, 2024-07-10:
Uh! Anyway are you sure you are using my integration? Cause it looks like you do not!

Anyway enable debugging, restart the HA, record the logs for some time and post it here!

Have fun!

davidrapan, 2024-07-12:
Upgrade to this version of the integration and then try again. Then if the problem persists open new issue with debug log attached!

Thank you and as always.. have fun!

closed 2024-07-12 by davidrapan