davidrapan/ha-solarman #17: Could we silence errors when inverter is offline?

Floppe, 2024-07-11:
My inverter gets power from the sun, so at nights the inverter is offline and stick logger is unreachable.

I've lost the logs now when I've restarted several times, but the logs are quite noisy at nights. Perhaps enough with a warning message or remove stack trace and only show unreachable.

PV-Joe, 2024-07-11:
What inverter are we talking about?

Floppe, 2024-07-11:
It's a Sofar KTLX-G3

davidrapan, 2024-07-11:
Hi there, It's made for continuous connection.. So it goes crazy when it can't reach the inverter.

I will think about a way how it could be done without affecting current working state. But can't make any promises as I'm really busy lately.

I wish I could help you more.

Floppe, 2024-07-12:
Hi. I understand that. Here are some log snippets incase you do find some time.

```
2024-07-12 04:26:06.656 ERROR (MainThread) [custom_components.solarman.api] OSError: [Errno 113] Connect call failed ('192.168.10.159', 8899)
Traceback (most recent call last):
  File "/config/custom_components/solarman/api.py", line 47, in reconnect
    self.reader, self.writer = await asyncio.open_connection(self.address, self.port)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1120, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1102, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1005, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 641, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 681, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.10.159', 8899)
2024-07-12 04:26:06.658 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-07-12 04:26:06.658 ERROR (MainThread) [custom_components.solarman.api] [2358423359] Send/Receive error: [Errno 113] Host is unreachable
Traceback (most recent call last):
  File "/config/custom_components/solarman/api.py", line 69, in _send_receive_v5_frame
    await self.writer.drain()
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 380, in drain
    raise exc
  File "/config/custom_components/solarman/api.py", line 199, in async_get
    await self.async_read(code, params, start, end)
  File "/config/custom_components/solarman/api.py", line 133, in async_read
    await self._read_registers(code, params, start, end)
  File "/config/custom_components/solarman/api.py", line 125, in _read_registers
    response = await self.read_holding_registers(register_addr = start, quantity = length)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 283, in read_holding_registers
    modbus_values = await self._get_modbus_response(mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 246, in _get_modbus_response
    mb_response_frame = await self._send_receive_modbus_frame(mb_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 233, in _send_receive_modbus_frame
    v5_response_frame = await self._send_receive_v5_frame(v5_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 69, in _send_receive_v5_frame
    await self.writer.drain()
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 380, in drain
    raise exc
  File "/config/custom_components/solarman/api.py", line 199, in async_get
    await self.async_read(code, params, start, end)
  File "/config/custom_components/solarman/api.py", line 133, in async_read
    await self._read_registers(code, params, start, end)
  File "/config/custom_components/solarman/api.py", line 125, in _read_registers
    response = await self.read_holding_registers(register_addr = start, quantity = length)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 283, in read_holding_registers
    modbus_values = await self._get_modbus_response(mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 246, in _get_modbus_response
    mb_response_frame = await self._send_receive_modbus_frame(mb_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 233, in _send_receive_modbus_frame
    v5_response_frame = await self._send_receive_v5_frame(v5_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 69, in _send_receive_v5_frame
    await self.writer.drain()
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 380, in drain
[... 35 lines cut ...]
OSError: [Errno 113] Host is unreachable
2024-07-12 04:26:06.662 WARNING (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) failed. #0 [OSError: [Errno 113] Host is unreachable]
2024-07-12 04:26:10.663 ERROR (MainThread) [custom_components.solarman.coordinator] Unexpected error fetching Solarman data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh
[... cut ...]
  File "/config/custom_components/solarman/api.py", line 173, in async_get_failed
    await self.async_disconnect()
  File "/config/custom_components/solarman/api.py", line 113, in async_disconnect
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 364, in wait_closed
    await self._protocol._get_close_waiter(self)
OSError: [Errno 113] Host is unreachable
....
....snip....
....
2024-07-12 04:26:18.614 ERROR (MainThread) [custom_components.solarman.api] OSError: [Errno 113] Connect call failed ('192.168.10.159', 8899)

```

davidrapan, 2024-07-15:
Hi there, I added in the [fix: Try for extended handling for EHOSTUNREACH](https://github.com/davidrapan/ha-solarman/commit/47b6d34dc69c8517f7864add46a1a3fc4dcc5648) some extended error handling to try to debug our way into some solution. Could you please try it? Enable debug logging and post the logs here?

_You will have to manually overwrite the code from the repo in your HA as this is not as part of latest release._

Thanks and have fun!

Floppe, 2024-07-17:
I cloned the repo manually and enabled extensive debug logging. Lets wait and see.

Floppe, 2024-07-18:
Here is a snippet from when the inverter did go offline. Hope it's enough.

```
2024-07-17 23:20:37.980 DEBUG (MainThread) [custom_components.solarman.parser] Default update_interval: 60, min_span: 25, digits: 6
2024-07-17 23:20:37.980 DEBUG (MainThread) [custom_components.solarman.api] Scheduling 4 query requests. #-5
2024-07-17 23:20:37.980 DEBUG (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) ...
2024-07-17 23:20:37.980 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-17 23:20:37.980 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] SENT: a5 17 00 10 45 1d 00 3f af 92 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 2
b 45 24 37 15
2024-07-17 23:20:51.981 WARNING (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) failed. #-5 [TimeoutError]
2024-07-17 23:20:54.983 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 4
2024-07-17 23:20:54.983 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-17 23:20:54.983 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] SENT: a5 17 00 10 45 1e 00 3f af 92 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 2
b 45 24 38 15
2024-07-17 23:21:07.980 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Solarman data in 30.000 seconds (success: False)
2024-07-17 23:21:12.968 DEBUG (MainThread) [custom_components.solarman.parser] Default update_interval: 60, min_span: 25, digits: 6
2024-07-17 23:21:12.968 DEBUG (MainThread) [custom_components.solarman.api] Scheduling 4 query requests. #-5
2024-07-17 23:21:12.968 DEBUG (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) ...
2024-07-17 23:21:12.968 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-17 23:21:12.968 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] SENT: a5 17 00 10 45 1f 00 3f af 92 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 2
b 45 24 39 15
2024-07-17 23:21:26.971 WARNING (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) failed. #-5 [TimeoutError]
2024-07-17 23:21:29.971 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 4
2024-07-17 23:21:29.971 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: True, cancelled: False.
2024-07-17 23:21:29.972 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] SENT: a5 17 00 10 45 20 00 3f af 92 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 2
b 45 24 3a 15
2024-07-17 23:21:29.972 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] EHOSTUNREACH error: [Errno 113] Host is unreachable
2024-07-17 23:21:29.972 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] Send/Receive error: [Errno 113] Host is unreachable
2024-07-17 23:21:29.972 WARNING (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) failed. #-5 [TimeoutError]
2024-07-17 23:21:32.973 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 3
2024-07-17 23:21:32.973 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: True, cancelled: False.
2024-07-17 23:21:32.973 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] SENT: a5 17 00 10 45 21 00 3f af 92 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 2
b 45 24 3b 15
2024-07-17 23:21:32.973 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] EHOSTUNREACH error: [Errno 113] Host is unreachable
2024-07-17 23:21:32.973 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] Send/Receive error: [Errno 113] Host is unreachable
2024-07-17 23:21:32.973 WARNING (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) failed. #-5 [TimeoutError]
2024-07-17 23:21:35.974 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 2
2024-07-17 23:21:35.974 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: True, cancelled: False.
2024-07-17 23:21:35.974 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] SENT: a5 17 00 10 45 22 00 3f af 92 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 2
b 45 24 3c 15
2024-07-17 23:21:35.974 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] EHOSTUNREACH error: [Errno 113] Host is unreachable
2024-07-17 23:21:35.974 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] Send/Receive error: [Errno 113] Host is unreachable
2024-07-17 23:21:35.974 WARNING (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) failed. #-5 [TimeoutError]
2024-07-17 23:21:38.975 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 1
2024-07-17 23:21:38.975 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: True, cancelled: False.
2024-07-17 23:21:38.975 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] SENT: a5 17 00 10 45 23 00 3f af 92 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 2
b 45 24 3d 15
2024-07-17 23:21:38.975 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] EHOSTUNREACH error: [Errno 113] Host is unreachable
2024-07-17 23:21:38.975 DEBUG (MainThread) [custom_components.solarman.api] [2358423359] Send/Receive error: [Errno 113] Host is unreachable
[... cut ...]
2024-07-17 23:21:42.075 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Solarman data in 29.107 seconds (success: False)
2024-07-17 23:21:46.857 DEBUG (MainThread) [custom_components.solarman.parser] Default update_interval: 60, min_span: 25, digits: 6
2024-07-17 23:21:46.857 DEBUG (MainThread) [custom_components.solarman.api] Scheduling 4 query requests. #-5
2024-07-17 23:21:46.857 DEBUG (MainThread) [custom_components.solarman.api] Querying (1028 - 1070) ...
2024-07-17 23:21:46.857 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: True, cancelled: False.

davidrapan, 2024-07-18:
Great! So it looks like I was able to successfully catch and filter out the Errno 113.

So I guess we could make them completely silent...

Thanks for your help and I let you know when it will be ready. ;)

davidrapan, 2024-07-18:
Hi @Floppe so I just committed prob final changes regarding "our issue" so could you maybe update the integration and go through the debug logging with me so I can check the flow?

This latest version should quiet the logging! (With debug disabled of course)

If there won't be any other issues consider this matter as done!

Have fun!

gedger, 2024-07-20:
I have been watching this issue with interest as I have the same problem with a Solis Inverter. My current solution is to stop the integration polling and instead do it from an automation that starts after sunrise and stops at sunset. Would an improved solution be to have an option in the configuration to "only poll during daylight"? As far as I'm aware all solar inverters shut down at night unless they are a hybrid version with an attached battery.

Floppe, 2024-07-20:
Tested and it does not log, so that's good. But it did not connect after sunrise. Could it be because I restarted after sunset. 🤔

All sensors are now unavailable.

davidrapan, 2024-07-20:
> I have been watching this issue with interest as I have the same problem with a Solis Inverter. My current solution is to stop the integration polling and instead do it from an automation that [...]

That is really good idea! I look into it!

davidrapan, 2024-07-20:
> Tested and it does not log, so that's good. But it did not connect after sunrise. Could it be because I restarted after sunset. 🤔
> 
> All sensors are now unavailable.

And what if you enable debug log? And let it run "over night" how does the log look like in the morning?

szyx22, 2024-07-20:
There is a big problem with this addon,  my HA cpu usage normaly is under 1%,  cpu usage growing up to 10%  from sunset to midnight  (21:00 to 00:00 3 hours).  Debug log size about 490 MB and a milions of lines...
[image]

davidrapan, 2024-07-21:
Okay but what it logs though? What does the log look like? There is a possibility that it does not behave correctly when is the inverter offline. Are you turning of the inverter?

Floppe, 2024-07-21:
> > Tested and it does not log, so that's good. But it did not connect after sunrise. Could it be because I restarted after sunset. 🤔
> > 
> > All sensors are now unavailable.
> 
> And what if you enable debug log? And let it run "over night" how does the log look like in the morning?

Enabled debug and let it run over night. This time it worked/started after the inverter came online. The only difference now is that debugging is enabled and I restarted while the inverter was online.

szyx22, 2024-07-21:
No I am not turning off my inverter, it do automaticly turn off after sunset like a lot of other inverter models.  I think addon should be prepared for that situation. "Turn off addon after sunset" it is not resolving issue.  Maybe ping command to check inverter availability? . Error logs are the same as up

davidrapan, 2024-07-21:
> Enabled debug and let it run over night. This time it worked/started after the inverter came online. The only difference now is that debugging is enabled and I restarted while the inverter was online.

Try it one more time then. But without touching the inverter if possible...

Debug enabled/disable should really don't have any impact on the inner workings it just simple show some more massages in the flow along the way I don't do there any flow changes based on debugging...

davidrapan, 2024-07-21:
> No I am not turning off my inverter, it do automaticly turn off after sunset like a lot of other inverter models. I think addon should be prepared for that situation. "Turn off addon after sunset" [...]

I was asking in general if you inverter runs over night or not.

So it does not okay.

My inverter never goes offline, cause I'm charging batteries in off-peak hours and so. So this integration is mainly written to work in that scenario.

You said that your log increased in size during night and now are you saying that it has the same messages as during the day? I don't follow...

If you want to resolve any issues you have you need to be more specific (provide logs, or better provide debug logs). Just by simply stating: "It does not work, fix it." Does not help at all and my hands are tied.

Adapting flow based on ping would be really bad Idea cause some are connecting to the inverters trough the WAN and there is not guaranteed that ping would be allowed.

_Btw I don't think the increased CPU load is not caused by this integration. When i turn off the inverter. It does not happen. But I can be mistaken. That's why is crucial that you provide more info._ :-)

Cheers!

Floppe, 2024-07-21:
> > > > Tested and it does not log, so that's good. But it did not connect after sunrise. Could it be because I restarted after sunset. 🤔
> > > > All sensors are now unavailable.
> > > 
> > > 
> > > And what if you enable debug log? And let it run "over night" how does the log look like in the morning?
> > 
> > Enabled debug and let it run over night. This time it worked/started after the inverter came online. The only difference now is that debugging is enabled and I restarted while the inverter was online.
> 
> Try it one more time then. But without touching the inverter if possible...
> 
> Debug enabled/disable should really don't have any impact on the inner workings it just simply show some more massages in the flow along the way I don't do there any flow changes based on debugging...

If I recall correctly this happened once before also with the original version. So I guess it's not related to this issue that silences the logging.

davidrapan, 2024-07-21:
Yeah sure I was just wondering if this changes did not somehow broke the "coming up" after night off. Cause what could happen is that logging of the messages can slow down that reconnecting intervals and thus without logging them the integration could just do it more frequently. But that's just shooting in the dark again.

I'll try playing with turning off the inverter tonight and I'll see what comes from it.

szyx22, 2024-07-22:
Yes good.  So installed this addon again.  First night was everything good with cpu usage. Second night, today, cpu again growing up, few minutes log is huge.   Graph cpu usage with this addon and next few minutes with uninstaled addon.
[image]
` File "/config/custom_components/solarman/api.py", line 64, in _send_receive_v5_frame
    await self.writer.drain()
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 380, in drain
    raise exc
  File "/config/custom_components/solarman/api.py", line 181, in async_get
    await self.async_read(params, code, start, end)
  File "/config/custom_components/solarman/api.py", line 116, in async_read
    response = await self.read_holding_registers(register_addr = start, quantity = length)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 283, in read_holding_registers
    modbus_values = await self._get_modbus_response(mb_request_frame)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 246, in _get_modbus_response
    mb_response_frame = await self._send_receive_modbus_frame(mb_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 233, in _send_receive_modbus_frame
    v5_response_frame = await self._send_receive_v5_frame(v5_request_frame)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 64, in _send_receive_v5_frame
    await self.writer.drain()
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 380, in drain
    raise exc
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 312, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/coordinator.py", line 33, in _async_update_data
    return await self.inverter.async_get(self._accounting())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 207, in async_get
    await self.async_get_failed(f"Querying {self.serial} at {self.address}:{self.port} failed. [{format_exception(e)}]")
  File "/config/custom_components/solarman/api.py", line 155, in async_get_failed
    await self.async_disconnect()
  File "/config/custom_components/solarman/api.py", line 107, in async_disconnect
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 364, in wait_closed
    await self._protocol._get_close_waiter(self)
OSError: [Errno 113] Host is unreachable
2024-07-22 22:15:03.766 WARNING (MainThread) [custom_components.solarman.api] Querying (0 - 39) failed. #0 [OSError: [Errno 113] Host is unreachable]`

davidrapan, 2024-07-22:
I will really need whole log with debug enabled (It can even be just enabled when this happens for few minutes not whole time) cause I'm really not able to replicate it in my setup.

_Don't forget to obfuscate serial number._

Cheers!

Floppe, 2024-07-23:
For my part it has worked really great now. It has connected again two mornings in a row. So I still have no debug logs when it fails in the morning.

gedger, 2024-07-23:
The more I think about this the more I believe this is not the way to go. Suppressing warnings/errors is never a good idea as when an inverter is off-line with a genuine problem then it will be masked. There is no way of knowing whether an inverter is down because of a fault or because it's turned off. Either way a fault log should be raised.

Can I suggest a much easier and cleaner way to achieve this and that is to provide a control input to the integration. This is a simple run/stop control that can then be used by a HA automation to stop the inverter at sunset and then start at sunrise. An automation to trigger and sunrise and sunset is extremely simple to do.

davidrapan, 2024-07-23:
> The more I think about this the more I believe this is not the way to go. Suppressing warnings/errors is never a good idea as when an inverter is off-line with a genuine problem then it will be masked. There is no way of knowing whether an inverter is down because of a fault or because it's turned off. Either way a fault log should be raised.
> 
> Can I suggest a much easier and cleaner way to achieve this and that is to provide a control input to the integration. This is a simple run/stop control that can then be used by a HA automation to stop the inverter at sunset and then start at sunrise. An automation to trigger at sunrise and sunset is extremely simple to do.

Yeah the thing is I don't know about simple way how to accomplish this.. Even HA itself does not have feature like this. You just simply can't turn integration on/off on demand...

davidrapan, 2024-07-23:
And I'm not really hiding it. I'm just masking it as a TimeoutError which should in theory raise but not so much.

davidrapan, 2024-07-23:
> For my part it has worked really great now. It has connected again two mornings in a row. So I still have no debug logs when it fails in the morning.

No worries, just post it here when it happens. ;)

gedger, 2024-07-23:
> Yeah the thing is I don't know about simple way how to accomplish this.. Even HA itself does not have feature like this. You just simply can't turn integration on/off on demand...

Fair do's. The Home Assistant recorder allows you enable disable recording via a service call. I also have an integration that allows you to change the polling interval from a service call, but have no idea how. If I find anything that looks useful I'll share the code.

I could also go back to my previous way of working with Stephans integration, we'll see how it goes.

https://www.home-assistant.io/common-tasks/general/  

section: Defining a custom polling interval

davidrapan, 2024-07-23:
> Fair do's. The Home Assistant recorder allows you enable disable recording via a service call. I also have an integration that allows you to change the polling interval from a service call, but have [...]

Yes I know about this one but if I understand it correctly this just tweaks it from HA side into DataUpdateCoordinator.data which is filled by polling interval defined in the integration.

So in short It doesn't change how often it tries to connect to the inverter.

But maybe I'm wrong...

Floppe, 2024-07-23:
> > For my part it has worked really great now. It has connected again two mornings in a row. So I still have no debug logs when it fails in the morning.
> 
> No worries, just post it here when it happens. ;)

One thing though. The "Connection status" sensor says it is connected 3 days now, which is not true!

davidrapan, 2024-07-23:
> > > For my part it has worked really great now. It has connected again two mornings in a row. So I still have no debug logs when it fails in the morning.
> > 
> > 
> > No worries, just post it here when it happens. ;)
> 
> One thing though. The "Connection status" sensor says it is connected 3 days now, which is not true!
> 
> And all other sensors freezes, but should really be unavailable.

Hmm that's bad! Could you maybe post log from a time when inverter goes offline? (a debug one)

A fifth TimeoutError should switch the integration into disconnected state. If that does not happen that's a bug then.

_Don't forget to obfuscate!_

gedger, 2024-07-23:
When the inverter is offline my logs are filled continuously with warning messages.

[home-assistant_solarman_2024-07-23T21-03-45.045Z.log](https://github.com/user-attachments/files/16354304/home-assistant_solarman_2024-07-23T21-03-45.045Z.log)

davidrapan, 2024-07-23:
And what about connection state?

gedger, 2024-07-24:
I'll grab a log when it goes offline tonight.

gedger, 2024-07-24:
Here we go, rather a lot of logging....

[home-assistant_2024-07-24T21-03-15.464Z.log](https://github.com/user-attachments/files/16367504/home-assistant_2024-07-24T21-03-15.464Z.log)

davidrapan, 2024-07-24:
Mh, I'm really not sure what can be done with this one as it originates from [asyncio] 😞

gedger, 2024-07-25:
It also failed to make the connection when the inverter came back on line. I had to reload the config to get it to connect. Overnight there were quite a few logs.....

This error originated from a custom integration.
Logger: asyncio
Source: custom_components/solarman/api.py:63
integration: Solarman (documentation, issues)
First occurred: 24 July 2024 at 21:34:30 (4911 occurrences)
Last logged: 07:07:20
socket.send() raised exception.

gedger, 2024-07-25:
A possible temporary work around for this is to install the Frenck integration Spook. This provides several useful advanced service calls including the ability to enable/disable the polling for an integration. I'm testing this and will report back. I'll also try and increase my knowledge of the HA integrations to see if/how this functionality could be built into the integration.

Note: the config_entry_id for Solarman is found for your system when you build the automation using the GUI so don't copy this code verbatim.
```
alias: PV - Control Polling
description: Enables inverter polling after sunrise
trigger:
  - platform: sun
    event: sunrise
  - platform: sun
    event: sunset
  - platform: homeassistant
    event: start
condition: []
action:
  - if:
      - condition: sun
        after: sunrise
    then:
      - service: homeassistant.enable_polling
        metadata: {}
        data:
          config_entry_id: 8e29598c8e01390f2cf499a1eca7443a
    else:
      - service: homeassistant.disable_polling
        metadata: {}
        data:
          config_entry_id: 8e29598c8e01390f2cf499a1eca7443a
mode: single
```

davidrapan, 2024-07-25:
It just baffles me how on four installations it behaves in four different ways...

gedger, 2024-07-25:
I can only assume that each inverter goes offline in a different way? I know my Solis keeps working for a good while once generation has stopped by using grid power before it finally decides to go to sleep. It may be that in that state you can make a connection but it doesn’t respond to modbus/solarman?

gedger, 2024-07-25:
Not sure if this is worth a look?
https://github.com/StephanJoubert/home_assistant_solarman/pull/535

davidrapan, 2024-07-25:
That's basically what I did. Logging suppression. But Stephan is using synchronous flow so very different approach and sadly it won't help in any way in our asynchronous implementation.

davidrapan, 2024-07-26:
> It also failed to make the connection when the inverter came back on line. I had to reload the config to get it to connect. Overnight there were quite a few logs.....
>
> This error originated from a [...]

Does it also stop writing into logs then? Cause it would not make any sense otherwise...

gedger, 2024-07-26:
No, still writing into the logs but didn't pick up that the inverter was on-line 😕 

To be honest there's been several changes to my system over the last few days so I need to let it run for a few days and get a true picture of what's happening. The inverter going off-line also generates a problem if HA is restarted when it's dark as the integration can't initialise and fails, no idea whether it will recover from this when the inverter comes back on-line? It's hard to test when you only have one shutdown event per day...

davidrapan, 2024-07-26:
> It's hard to test when you only have one shutdown event per day...

Yeah exactly. It's the same with trying to guess how to mitigate this by shooting in the dark as I really don't know how to test this scenarios... My inverter runs 24/7 and every try to turn it off or even just interfere with the connection behaves very differently than what you guys describing :(

One of my latest attempts to help you guys out was to add clearing of the variables handling the reading and writing into the socket cause I thought that maybe because they are still holding an instances after disconnect flow and consecutive try to connect  fails then maybe the socket.send() exception was cause of that... But it is yet another shot in the dark...

gedger, 2024-07-27:
My automation failed to work. It successfully turned off polling and no errors reported however, when polling was enabled in the morning it failed to get a data although did seem to start trying to reconnect. Reloading the integration immediately connected and restored the data flow...

Polling was enabled at 5:07 and debug turned on for a bit from 7.29....

[home-assistant_solarman_2024-07-27T06-30-42.075Z.log](https://github.com/user-attachments/files/16398483/home-assistant_solarman_2024-07-27T06-30-42.075Z.log)

PS. I'm also debugging another integration so ignore the other errors...

davidrapan, 2024-07-27:
Okay now there is obvious it's not trying to connect back again after disconnect and sure it can't work like that...

BTW nice pic

gedger, 2024-07-27:
I have updated to latest version and the Update_Interval is now correct at around 30 secs before it recording 5-6 secs. I'll capture more log tomorrow morning if it has the same issue as I won't be dashing out....Will report back.

gedger, 2024-07-28:
Whoops seems to be a coding issue.  You can replicate by going to system options and disable polling. Needed a reboot to recover.

[home-assistant_solarman_2024-07-28T06-00-49.451Z.log](https://github.com/user-attachments/files/16402418/home-assistant_solarman_2024-07-28T06-00-49.451Z.log)

davidrapan, 2024-07-28:
Ou it looks like I forgot to commit some changes, sorry bout that: [321e9e4](https://github.com/davidrapan/ha-solarman/commit/321e9e436801111c0acad5dab0b4070fdd9c1e9b)

gedger, 2024-07-28:
It's reassuring to be human and forget things 😆 

All working will report back in 24hr....

Cheers

Rich

gedger, 2024-07-29:
It worked!! So using the automation provided earlier the inverter polling was successfully stopped at sunset and started at sunrise with no errors. Interestingly the connection sensor went off line very briefly when the polling was disabled but then reconnected. I'll run the debug log around sunset time tonight so we can see what is happening.

gedger, 2024-07-30:
Updated to 24.07.28 and worked fine again overnight with the Polling Enable/Disable. I'll disable to the automation tonight and see how it behaves going off-line without changing the polling state. Interestingly after disabling the polling the logs indicate that the integration still reconnects to the inverter and then goes off-line but the Connection State sensor still says connected. Probably now a low priority but here are the debug logs around the events.

[home-assistant_solarman_2024-07-30T05-57-19.308Z.log](https://github.com/user-attachments/files/16423746/home-assistant_solarman_2024-07-30T05-57-19.308Z.log)

davidrapan, 2024-07-30:
Yeah it proly won't change state on its own cause it's stateless as it needs to fail to connect first and then it will change connection state.

gedger, 2024-07-31:
Here's the log from last night without changing the enable_polling state via an automation so the integration was just free running. On the positive side it kept working and recovered in the morning although it did raise a few log messages which I have attached. So I will go back to using the Spook integration to disable polling and have a think about how that stop polling functionality could be added to this integration.

[home-assistant_2024-07-31T07-29-44.541Z.log](https://github.com/user-attachments/files/16438009/home-assistant_2024-07-31T07-29-44.541Z.log)

davidrapan, 2024-07-31:
@Floppe can I have little OF question? What definition profile file are you using with your device? (For Supported inverters in Wiki)

And also you @gedger, what combination of device model (full model name/number please) and definition file are you running?

Floppe, 2024-07-31:
> @Floppe can I have little OT question? What definition profile file are you using with your device? (For Supported inverters in Wiki)
> 
> And also you @gedger, what combination of device model (full model name/number please) and definition file are you running?
> 
> When comes to the feature of enable/disable of polling directly in the integration I'm all for it if we find an elegant solution, but I would also find a nice and workable solution even without it... I think we are getting there.

Hello. I'm using this one.
https://gist.github.com/Floppe/83365483954d8bcee8d8854eeb3fa818

davidrapan, 2024-07-31:
Is it based on any particular one which is in this repo or it's totally different?

gedger, 2024-08-01:
Sure, the inverter is a Solis-1P8K-5G Inverter. I now have the official Modbus spec. from Solis but it's generic and not all Modbus registers work on all Inverters which means you still have to try and see if it works.
https://www.solisinverters.com/global/1p_inverter4/34588.html

The repro config file for my inverter solis_1p8k-5g.yaml currently has an issue in that the parameters: section has been offset one to the right so doesn't work plus there is an incorrect section which I've removed.  

[solis_1p8k-5g.yaml.zip](https://github.com/user-attachments/files/16451898/solis_1p8k-5g.yaml.zip)

I haven't submitted an update yet as there is still some odd timing issues I'm trying to get to the root cause of. I'll raise an another issue rather than pollute this thread....

I did have some odd messages this morning when polling was enabled.

[home-assistant_2024-08-01T07-18-23.697Z.log](https://github.com/user-attachments/files/16451939/home-assistant_2024-08-01T07-18-23.697Z.log)

[24 more replies]

closed 2024-08-17 by davidrapan