davidrapan/ha-solarman #19: Question: does this addon work with the SUN-M160G4-EU-Q0?

Bhoft, 2024-07-13:
Hi I am planing to buy this SUN-M160G4-EU-Q0 inverter and I am not sure if this addon supports this device.

Is there a "supported" or "checked" device list? 

Do i need an additional Hardware device (Solarman Stick Logger) to get the information from my inverter to my home assitant?
Or is there an ESP Home equivalent of this hardware (similar to the DTU for Hoymiles) ? (I am asking because if found the following in another issue)

    Compliments for the great work. 👍I've been using your integration for about a week now. 
    Works perfectly. Also, I use an ESP Home variant and can also configure the Deye with it.
    I look forward to this expansion in your solution!

_Originally posted by @Summer-of69 in https://github.com/davidrapan/ha-solarman/issues/9#issuecomment-2213952976_

davidrapan, 2024-07-13:
Hi there!

> Is there a "supported" or "checked" device list?
Not yet. But list from the original integration should be valid.

But yes this integration is made to work with Solarman Stick Logger so you need that piece of a HW.

As for the support of mentioned device profile 'deye_4mppt.yaml' is probably the one to go...

And when comes to the ESP I'm not really that much familiar to be giving advices.. :/

Adminius, 2024-07-30:
Hey @davidrapan I bought M160G4, my problem ist now, that it's serial number ends wirh letter "B" and I can't put letter in to serial number field

davidrapan, 2024-07-30:
Hi @Adminius, you need serial number from the Solarman Stick Logger and they should only contain numbers.

Adminius, 2024-07-30:
yeah...
[image]

this inverter, like other deye micro inverters, has integrated logger.

I tried this integration also with my Deye SUN12K and other micro inverters that worked with "original" Solarman integration and it doen't work at all... (sticks at intiaslisation). May I should go back to "original" Sorarman integration... sadly :/

davidrapan, 2024-07-30:
Solarman serial, its little bit lower on the screenshot you sent here.

Adminius, 2024-07-30:
Now I used the Solarman Serial
[image]

davidrapan, 2024-07-30:
**It is possible that for string inverters you gonna need beta pre-release of this integration. which you can enable using Redownload under 3 dots button menu in top right corner on the HACS page. Then just select the latest version (24.07.29) and click DOWNLOAD.**

You are using 24.07.24

davidrapan, 2024-07-30:
Also always enable debug on the same page you just showed me the screenshot from and send logs which it produces without it I really can't help you much. Logs are crucial.

Adminius, 2024-07-30:
[sm_debug_M160G4.txt](https://github.com/user-attachments/files/16429531/sm_debug_M160G4.txt)
Here is debug from one M160G4 only

davidrapan, 2024-07-30:
So it is working great but it gets some incorrect data for Total Production:
```
ValueError: Invalidate complete dataset (Total Production ~ 0.0)
```

Adminius, 2024-07-30:
Oha, it is comletely new inverter, yesterday connected to PV panels (WiFi Setup and so on), but not to the grid yet, that's why 0 kWh production

davidrapan, 2024-07-30:
I'm gonna remove it from the repository too as it really does not make any sense for it being there. Cause as you can see it causes troubles for any new inverter, lol.

davidrapan, 2024-07-30:
[2a10012](https://github.com/davidrapan/ha-solarman/commit/2a10012e5516655f3066aa1eb4e1fbdaa6625fa5)

Adminius, 2024-07-30:
Remove doen't works, it came again, lol
but 0.0 works now

Thanks!

davidrapan, 2024-07-30:
Great!

davidrapan, 2024-07-30:
[Supported Inverters](https://github.com/davidrapan/ha-solarman/wiki/Supported-Inverters)

closed 2024-07-30 by davidrapan