davidrapan/ha-solarman #21: Support for Ethernet Logger LSE-3

oetiker, 2024-07-18:
I am using a Sofar Hybrid Box (SOFAR HYD 20KTL-3PH) with a LSE-3 logger and get no responses when trying to use the thing with the ha-solarman code. Neither with the code here nor with the original from StephanJoubert.

Reading through @StephanJouberts repo I found [this issue](https://github.com/StephanJoubert/home_assistant_solarman/issues/247#issuecomment-1590665761) which seems to indicate that the LSE-3 device needs different encapsulation ...

Anyone working on this ? Or is there some secret to get my setup to communicate with ha-solarman which I just neglected until now?

davidrapan, 2024-07-18:
Hi @oetiker!

I can try to implement suggested changes but as I don't have the device it will be like shooting in the dark.

Are you willing to try it afterwards and solve potential bugs and so?

Have fun!

oetiker, 2024-07-18:
> Hi @oetiker!
> 
> I can try to implement suggested changes but as I don't have the device it will be like shooting in the dark.
> 
> Are you willing to try it afterwards and help solve potential bugs and so?
> 
> Have fun!

Hi David

yes sure, I would be totally willing to test that :)

cheers
tobi

davidrapan, 2024-07-18:
Ok, I'll let you know whenever there is anything to try out.

_But as the code of those suggested changes is apart from the pysolarmanv5 it won't be as straight forward implementation as I thought initially so I can't make promises it will work as great as when it goes through the mentioned python library. But I'll do what I can to make it work._ 😉 

Have fun!

davidrapan, 2024-08-03:
Does the Ethernet Stick Logger also responds to the discovery packet? Probably not right?

Vollpanto, 2024-08-03:
I have both sticks. The Ethernet Adapter also response to the discovery packet.

Vollpanto, 2024-08-03:
Here is a log:
2024-08-03 21:55:14.025 WARNING (MainThread) [custom_components.solarman.api] Querying (1028 - 1067) failed. #0 [TimeoutError]
2024-08-03 21:55:15.026 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 4
2024-08-03 21:55:15.026 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-08-03 21:55:15.028 DEBUG (MainThread) [custom_components.solarman.api] [2109340331] Successful reconnect
2024-08-03 21:55:15.028 DEBUG (MainThread) [custom_components.solarman.api] [2109340331] SENT: a5 17 00 10 45 b2 00 ab fa b9 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 59 15
2024-08-03 21:55:15.563 DEBUG (MainThread) [custom_components.solarman.api] [2109340331] V5_SEQ_NO_MISMATCH: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2e 00 00 00 2a 00 00 00 00 00 00 00 00 00 00 00 2b 00 00 00 00 00 00 00 00 00 00 05 23 00 05 e4 ef 00 08 87 bc 01 97
2024-08-03 21:55:25.020 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 30.001 seconds (success: False)
2024-08-03 21:55:25.020 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-08-03 21:55:25.020 INFO (MainThread) [custom_components.solarman.api] Disconnecting from 192.168.1.195:8899

lcavalli, 2024-08-03:
Indeed it seems so. I currently use an integration based on pure modbus protocol, but I'm testing your integration the config flow correctly present the logger IP address, port and also serial number.

davidrapan, 2024-08-03:
Okay, thanks guys. I'm currently investigating possible approaches to the implementation.

davidrapan, 2024-08-07:
Guys, can anyone test [feat: Experimental testing of ethernet stick](https://github.com/davidrapan/ha-solarman/commit/baea9f5e1b65980a0cdcd206066bc6bd9312120e) and show me the debug log? Thanks!

_For enabling this there is new configuration toggle which must be in on position._

lcavalli, 2024-08-07:
@davidrapan, It is not lack of interest, I'm currently not at home. I will be able to test it when I will come back! Thank you for spending time on this issue.

davidrapan, 2024-08-07:
And then for anyone who is going to try it there is this row in the api.py file:
```
modbus_frame = v5_frame[10:]
```

Try to tweak number 10. It says how much it should cut from the response from the start and treat the rest as an modbus message. So try then for example maybe also 20 and 2.

oetiker, 2024-08-08:
@davidrapan I will be glad to test, but I have not read the code to deeply and I am involved in way to many projects atm ... 

* how do I get the current version of the code ? can I just update ?

* do I just toggle the new config setting in the config screen ?

davidrapan, 2024-08-08:
Hi @oetiker,

> how do I get the current version of the code ? can I just update ?

You would have to manually download the code in zip and replace it in the HA as this changes are not part of any release.

> do I just toggle the new config setting in the config screen ?

Yes.

lcavalli, 2024-08-08:
>> how do I get the current version of the code ? can I just update ?

> You would have to manually download the code in zip and replace it in the HA as this changes are not part of any release.

You can also select _master_ in version combo box in HACS. Download again if already installed.

davidrapan, 2024-08-08:
> You can also select master in version combo box in HACS. Download again if already installed.

I know that you "can" do that but it won't switch to it. It reselects automatically to the release version and thus it never worked for me though.

davidrapan, 2024-08-09:
It's part of the 24.08.08 pre-release.

davidrapan, 2024-08-13:
I was hoping someone would at least give it a try before the planned 08/14/24 release, but I guess I'm out of luck, huh? 😆

davidrapan, 2024-08-13:
@lcavalli, I know you said that! 😆 I did not mean to pressure anyone with that! 😆

proggaras, 2024-08-18:
Great work. 
Will this also work, when I connect directly to my Deye inverter over RS485? 

Detail:
HomeAssistant connects over a "Waveshare  RS485 to Ethernet Converter" to my Deye SUN-25K-SG01HP3-EU-AM
[image]

davidrapan, 2024-08-18:
Hi @proggaras, this is what https://github.com/kellerza/sunsynk is for, I guess. 😉

This integration specializes to working using Solarman protocol only.

oetiker, 2024-08-19:
@davidrapan yey, I finally got to test ... unfortunately it does not work yet,. but there is debug data:

```
2024-08-19 09:15:48.279 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'data': {'name': 'hwr', 'inverter_discovery': True, 'inverter_host': '192.168.0.19', 'inverter_serial': 2108634189, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': True, 'lookup_file': 'sofar_g3hyd.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J5MQQ8GDVHPMDAB1A3PFHKXQ', 'minor_version': 1, 'options': {'name': 'hwr', 'inverter_discovery': False, 'inverter_host': '192.168.0.19', 'inverter_serial': 2108634189, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': True, 'lookup_file': 'sofar_g3hyd.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'hwr', 'unique_id': None, 'version': 1})
2024-08-19 09:15:48.406 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 3, min_span: 25, digits: 6
2024-08-19 09:15:48.406 DEBUG (MainThread) [custom_components.solarman.api] {'identifiers': {('solarman', 2108634189)}, 'name': 'hwr', 'manufacturer': 'Sofar', 'model': 'G3HYD', 'serial_number': 2108634189}
2024-08-19 09:15:48.406 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-08-19 09:15:48.406 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Scheduling 8 query requests. #0
2024-08-19 09:15:48.406 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Querying (1028 - 1067) ...
2024-08-19 09:15:48.406 INFO (MainThread) [custom_components.solarman.api] [2108634189] Connecting to 192.168.0.19:8899
2024-08-19 09:15:48.410 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] SENT: a5 17 00 10 45 ae 00 4d 34 af 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 27 15
2024-08-19 09:15:48.896 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] RECD: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 00 00 00 27 00 00 00 00 00 00 00 00 00 00 00 25 00 00 00 00 00 00 00 00 00 00 02 2b 00 03 d9 6b 00 03 f2 f5 01 a9
2024-08-19 09:15:48.897 WARNING (MainThread) [custom_components.solarman.api] [2108634189] Querying (1028 - 1067) failed. #0 [CRCError: CRC validation failed.]
2024-08-19 09:15:49.098 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Querying attempts left: 4
2024-08-19 09:15:49.103 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Successful reconnect
2024-08-19 09:15:49.103 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] SENT: a5 17 00 10 45 af 00 4d 34 af 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 28 15
2024-08-19 09:15:49.606 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] RECD: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 00 00 00 27 00 00 00 00 00 00 00 00 00 00 00 25 00 00 00 00 00 00 00 00 00 00 02 2b 00 03 d9 6b 00 03 f2 f5 01 a9
2024-08-19 09:15:49.607 WARNING (MainThread) [custom_components.solarman.api] [2108634189] Querying (1028 - 1067) failed. #0 [CRCError: CRC validation failed.]
2024-08-19 09:15:50.008 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Querying attempts left: 3
2024-08-19 09:15:50.012 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Successful reconnect
2024-08-19 09:15:50.012 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] SENT: a5 17 00 10 45 b0 00 4d 34 af 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 29 15
2024-08-19 09:15:50.522 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] RECD: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 00 00 00 27 00 00 00 00 00 00 00 00 00 00 00 25 00 00 00 00 00 00 00 00 00 00 02 2b 00 03 d9 6b 00 03 f2 f5 01 a9
2024-08-19 09:15:50.523 WARNING (MainThread) [custom_components.solarman.api] [2108634189] Querying (1028 - 1067) failed. #0 [CRCError: CRC validation failed.]
2024-08-19 09:15:51.125 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Querying attempts left: 2
2024-08-19 09:15:51.128 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Successful reconnect
2024-08-19 09:15:51.128 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] SENT: a5 17 00 10 45 b1 00 4d 34 af 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 2a 15
2024-08-19 09:15:51.635 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] RECD: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33 00 00 00 27 00 00 00 00 00 00 00 00 00 00 00 26 00 00 00 00 00 00 00 00 00 00 02 2b 00 03 d9 6b 00 03 f2 f5 01 a9
2024-08-19 09:15:51.636 WARNING (MainThread) [custom_components.solarman.api] [2108634189] Querying (1028 - 1067) failed. #0 [CRCError: CRC validation failed.]
[... cut ...]
2024-08-19 09:15:53.916 DEBUG (MainThread) [custom_components.solarman.api] [2108634189] Request failed. [Previous State: Disconnected (-1)]
2024-08-19 09:15:53.916 INFO (MainThread) [custom_components.solarman.api] [2108634189] Disconnecting from 192.168.0.19:8899
2024-08-19 09:15:53.917 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching hwr data in 5.511 seconds (success: False)
2024-08-19 09:15:53.917 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-08-19 09:15:53.918 INFO (MainThread) [custom_components.solarman.api] [2108634189] Disconnecting from 192.168.0.19:8899
```

davidrapan, 2024-08-19:
> And then for anyone who is going to try it there is this row in the api.py file (on line 34):
> 
> ```
> modbus_frame = v5_frame[10:]
> ```
> 
> Try to tweak number 10. It says how much it should cut from the response from the start and treat the rest as an modbus frame. So try for example maybe also 20 and 2...
> 
> _But it really depends on what the code is gonna do._

This ^^ 😉

davidrapan, 2024-08-19:
Hey @githubDante, can you take a look at our experiment? Did I cut from the solarman frame too much or not enough?

lcavalli, 2024-08-19:
Back at home.
I tried the new feature and as already reported it doesn't still work.
The reported error is the same as already logged above (crc error).
To further analyse the issue I installed tcpdump on Home Assistant Operating System and dumped a session from a different integration based on bare modbus (I would like to replace due to some inconsistencies in reported values). This is the starting working request to read the serial number on ZCS Azzurro  inverter (addess 0x0445, length 7).
Request:

[image]

Reply (obfuscated):

[image]

CRC seems not to be used as part of the protocol, since it is modbus TCP, not RTU.

davidrapan, 2024-08-19:
Oh I didn't know that tcp does not have the crc, my bad.

davidrapan, 2024-08-19:
So the cutting part will be:
```
modbus_frame = v5_frame[7:]
```
+ crc

davidrapan, 2024-08-19:
Guys try it from this commit: [feat: Experimental testing of ethernet stick](https://github.com/davidrapan/ha-solarman/commit/5d44f3cfe0bc56e84997ffefab60c040b3aa7912)

githubDante, 2024-08-19:
Hi,

@lcavalli do you have the pcap  file from tcpdump ?

@davidrapan if the ethernet logger operates in MODBUS TCP mode you can use a shortcut  by passing a prepared request directly to the `_send_receive_v5_frame` function and await the result from it. The following is a simple example

```python
from umodbus.client.tcp import read_holding_registers, read_input_registers, read_discrete_inputs, etc.   # you can map these to function codes should you wish 
...
tcp_req = read_holding_registers(<slave id>, <starting address>, <quantity>)  # choose the appropriate function 
# feed the tcp_req directly to _send_receive_v5_frame and process the result as follows ...
resp_adu = await super()._send_receive_v5_frame(tcp_req)
inverter_data = parse_response_adu(resp_adu, tcp_req)
# no other intervention should be needed the data can be returned directly 
```

* An example with data provided by @lcavalli (with slightly modified Transaction identifier)

```python
>>> req = read_holding_registers(1, 0x0445, 7)
>>> req
b',\x9a\x00\x00\x00\x06\x01\x03\x04E\x00\x07'
>>> req.hex()
'2c9a00000006010304450007'
>>> resp_b = bytes.fromhex('2c 9a 00 00 00 11 01 03 0e 5a 4d 32 45 53 30 30 30 30 30 30 30 30 30')
>>> resp = parse_response_adu(resp_b, req)
>>> resp
[23117, 12869, 21296, 12336, 12336, 12336, 12336]
>>> b''.join([struct.pack('>H', x) for x in resp])
b'ZM2ES000000000'
```

lcavalli, 2024-08-19:
It works now! Experienced a short disconnection but sensors are now available. Thanks!

@githubDante yes, I have a pcap dump but it contains sensitive data, I suppose, like the serial number of the inverter used for solarman cloud registration.

davidrapan, 2024-08-19:
> It works now! Experienced a short disconnection but sensors are now available. Thanks!

😆 😆 😆

I'm literally laughing my ass off! 😆 But I'm glad it worked!

@githubDante Turns out umodbus.client.tcp is not really needed and pysolarmanv5 is capable of reading the data + with very little tweak. 🎉

davidrapan, 2024-08-19:
I wonder if is the start of the reply `a5 17 00 10 45 53` the same every single time?

lcavalli, 2024-08-19:
Here is a short debug log.
[solarman_log.txt](https://github.com/user-attachments/files/16666771/solarman_log.txt)

davidrapan, 2024-08-19:
Great! Can you also test it with changes made here?: [feat: Add some constraints for ethernet sticks](https://github.com/davidrapan/ha-solarman/commit/db9714c784eb70491b01289f971816b410d5c609)

PV-Joe, 2024-08-20:
I was watching the development and think you are doing a great job here. Glad to see that Solarman is getting some love and is being continued. ☀️

lcavalli, 2024-08-20:
> Great! Can you also test it with changes made here?: [feat: Add some constraints for ethernet sticks](https://github.com/davidrapan/ha-solarman/commit/db9714c784eb70491b01289f971816b410d5c609)
> 
> _Can you guys believe it that adding support for those sticks only needed around 16 rows of code?_ 😆

I would like to test your last commit but after a series of periodic disconnection this night the inverted become offline and I cannot connect anymore.

[image]

The inverter is offline also on the official app.

Is is possible that the query rate was too high? Will try to reset the unit.

davidrapan, 2024-08-20:
It should not be possible to shut it down like that though. But you can increase it in the definition profile with update_interval.

oetiker, 2024-08-20:
I copied with the api.py from main and yey it does work ... it does seem that not all queries succeed though ... I guess some timeouts are not yet as they should be

davidrapan, 2024-08-20:
Well for me to even have a chance to do something about it I'm gonna need to see the debug logs, haha. 😆 

And you tried it also with the last commit?

oetiker, 2024-08-20:
> Well for me to even have a chance to do something about it I'm gonna need to see the debug logs, haha. 😆
> 
> And you tried it also with the last commit?

yes with the version from 9h ago ... will investigate the timeouts

davidrapan, 2024-08-20:
What profiles are you using guys anyway?

lcavalli, 2024-08-20:
> It should not be possible to shut it down like that though. But you can increase it in the definition profile with update_interval.
> 
> Please could you try to catch in debug log why it disconnects periodically? 😉

Inverter still offline with the integration disabled. As soon as it will be online again I will try to investigate the disconnection issue. Good to know that the rate can be configured in definition file.

oetiker, 2024-08-20:
sofar_tlx-g3.yaml and sofar_g3hyd.yaml

the connectivity problems seem to have 'system' ... 
[image]

davidrapan, 2024-08-20:
Enable debug logging and try to catch that disconnect. 😉

lcavalli, 2024-08-20:
> What profiles are you using guys anyway?

I'm using this profile, although my inverter is single phase only: zcs_azzurro-hyd-zss-hp.yaml

BTW, the logger is still unable to communicate with the inverter (LAN led is on, COM led is off, SER led is blinking@1hz). Now I disconnected the logger from the inverter, will reconnect tomorrow.

davidrapan, 2024-08-20:
> BTW, the logger is still unable to communicate with the inverter (LAN led is on, COM led is off, SER led is blinking@1hz). Now I disconnected the logger from the inverter, will reconnect tomorrow.

I really hope everything will be alright in there!

What 'bout you @oetiker, everything good?

davidrapan, 2024-08-20:
Can you guys also try this version?: [feat: Ethernet Logger through umodbus.client.tcp](https://github.com/davidrapan/ha-solarman/commit/c9ab276599e5b761a981568a1b9d62b12583736d)

It's according @githubDante suggestions.

lcavalli, 2024-08-21:
Hi, inverter is back online.
I'm checking the disconnection problem, this is the log portion of interest:

```
2024-08-21 15:32:03.714 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Scheduling 7 query requests. #240
2024-08-21 15:32:03.715 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1028 - 1067) ...
2024-08-21 15:32:03.715 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: a5 17 00 10 45 f6 00 fa fc 73 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 04 00 28 05 25 a8 15
2024-08-21 15:32:04.299 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] RECD: a5 17 00 10 45 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 3b 00 00 00 3b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 da 00 04 bb 92 00 08 78 23 01 d1
2024-08-21 15:32:04.300 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying succeeded.
2024-08-21 15:32:04.300 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1156 - 1165) ...
2024-08-21 15:32:04.300 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: a5 17 00 10 45 f7 00 fa fc 73 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 84 00 0a 84 d4 39 15
2024-08-21 15:32:04.834 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] RECD: a5 17 00 10 45 17 01 03 14 13 89 01 83 00 00 00 00 00 86 00 00 00 00 00 00 00 00 09 03
2024-08-21 15:32:04.834 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying succeeded.
2024-08-21 15:32:04.834 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1199 - 1199) ...
2024-08-21 15:32:04.834 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: a5 17 00 10 45 f8 00 fa fc 73 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 04 af 00 01 b5 1b d4 15
2024-08-21 15:32:05.349 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] RECD: a5 17 00 10 45 05 01 03 02 00 fc
2024-08-21 15:32:05.349 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying succeeded.
2024-08-21 15:32:05.350 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1284 - 1284) ...
2024-08-21 15:32:05.350 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: a5 17 00 10 45 f9 00 fa fc 73 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 05 04 00 01 c5 07 27 15
2024-08-21 15:32:05.893 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] RECD: a5 17 00 10 45 05 01 03 02 00 03
2024-08-21 15:32:05.894 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying succeeded.
2024-08-21 15:32:05.894 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1412 - 1417) ...
2024-08-21 15:32:05.894 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: a5 17 00 10 45 fa 00 fa fc 73 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 05 84 00 06 85 2d 93 15
2024-08-21 15:32:06.529 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] RECD: a5 17 00 10 45 0f 01 03 0c 09 4e 03 53 00 ca 09 cb 03 11 00 c5
2024-08-21 15:32:06.530 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying succeeded.
2024-08-21 15:32:06.530 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1540 - 1546) ...
2024-08-21 15:32:06.530 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: a5 17 00 10 45 fb 00 fa fc 73 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 06 04 00 07 45 41 ea 15
2024-08-21 15:32:07.054 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] RECD: a5 17 00 10 45 11 01 03 0e 02 21 00 00 00 00 00 25 00 64 00 63 00 00
2024-08-21 15:32:07.055 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying succeeded.
2024-08-21 15:32:07.055 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1667 - 1691) ...
2024-08-21 15:32:07.055 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: a5 17 00 10 45 fc 00 fa fc 73 7d 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 06 83 00 19 75 60 cb 15
2024-08-21 15:32:07.619 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] RECD: a5 17 00 10 45 35 01 03 32 ff ff 00 00 0a 76 00 01 1a 81 00 00 07 d2 00 01 18 94 00 00 01 20 00 00 a3 8a 00 00 01 bc 00 00 92 7e 00 00 01 c0 00 00 2e f1 00 00 00 3c 00 00 2a 8a
2024-08-21 15:32:07.620 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying succeeded.
2024-08-21 15:32:07.620 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Returning 47 new values to the Coordinator. [Previous State: Connected (1)]
2024-08-21 15:32:07.621 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching ZCS data in 3.907 seconds (success: True)
2024-08-21 15:32:12.713 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Scheduling 0 query requests. #245
2024-08-21 15:32:12.713 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching ZCS data in 0.001 seconds (success: True)
2024-08-21 15:32:17.714 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Scheduling 0 query requests. #250
2024-08-21 15:32:17.714 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching ZCS data in 0.001 seconds (success: True)
[... cut ...]
```

I modified the inverter profile to query every 20 seconds. At the beginning everything is working, until 15:32:33.815, when the data logger stops answering. After 5 attempts the integration aborts and close the connection. After several retries the data logger restart its normal behavior and replies are correctly processed. I don't know if it is a sort of protection of the data logger with respect the query rate (now 20 seconds, but 7 independent queries are performed together in sequence).

davidrapan, 2024-08-21:
Yeah well it could be really anything... but as I already stated I would guess that the ethernet version could be little bit more resilient that the wireless. :-/

Anyway try it with the changes I was talking one post above. It could change this to the better. (If it will work though as I can't be sure if I did not forget anything I'm just blind guessing cause I don't have the device) 😉

lcavalli, 2024-08-21:
> Anyway try it with the changes I was talking one post above. It could change this to the better. (If it will work though as I can't be sure if I did not forget anything I'm just blind guessing cause I don't have the device) 😉

Could you please make a pre-release? The update is easier with HACS 2.0 (cannot download master anymore).

davidrapan, 2024-08-21:
Haha the changes are in different branch.. I mean you can just swap the content of the api.py. I also gave you there link to the file.

I mean I could try to do that pre-releas but I think ^^ will be bit faster. :-P

lcavalli, 2024-08-21:
Yep, will do manually, no problem. _dev_ branch?

davidrapan, 2024-08-21:
https://github.com/davidrapan/ha-solarman/blob/dev/custom_components/solarman/api.py

lcavalli, 2024-08-21:
Not working... Integration cannot even start.

```
2024-08-21 18:01:06.719 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1028 - 1067 | 0x0404 - 0x042B # 040) failed. [TimeoutError]
2024-08-21 18:01:06.920 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1028 - 1067 | 0x0404 - 0x042B # 040) attempts left: 4
2024-08-21 18:01:06.923 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Successful reconnect
2024-08-21 18:01:06.924 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: e7 5a 00 00 00 06 01 03 04 04 00 28
2024-08-21 18:01:08.982 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] V5_MISMATCH: e7 5a 00 00 00 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 37 00 00 00 36 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 6f 00 04 bc 27 00 08 78 b8 01 d1
2024-08-21 18:01:17.714 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Request failed. [Previous State: Disconnected (-1)]
2024-08-21 18:01:17.715 INFO (MainThread) [custom_components.solarman.api] [2104753402] Disconnecting from 192.168.1.178:8899
2024-08-21 18:01:17.717 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-08-21 18:01:17.717 INFO (MainThread) [custom_components.solarman.api] [2104753402] Disconnecting from 192.168.1.178:8899
2024-08-21 18:01:23.015 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-08-21T13:27:04.824613+00:00', 'data': {'name': 'ZCS', 'inverter_discovery': True, 'inverter_host': '192.168.xxx.xxx', 'inverter_serial': xxxxxxxxxx, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': True, 'lookup_file': 'zcs_azzurro-hyd-zss-hp.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxx', 'minor_version': 1, 'modified_at': '2024-08-21T13:27:04.824617+00:00', 'options': {'name': 'ZCS', 'inverter_discovery': True, 'inverter_host': '192.168.xxx.xxx', 'inverter_serial': xxxxxxxxxxx, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': True, 'lookup_file': 'zcs_azzurro-hyd-zss-hp.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'ZCS', 'unique_id': None, 'version': 1})
2024-08-21 18:01:23.016 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-08-21 18:01:23.017 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.xxx.xxx, xxxxxxxxxxxxxxxxxx, xxxxxxxxxxxxx]
2024-08-21 18:01:24.090 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 20, code: 3, min_span: 25, digits: 6
2024-08-21 18:01:24.090 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', '34:ea:e7:f6:55:6b')}, 'identifiers': {('solarman', 2104753402)}, 'name': 'ZCS', 'manufacturer': 'Zcs', 'model': 'AZZURRO-HYD-ZSS-HP', 'serial_number': xxxxxxxxxxx}
2024-08-21 18:01:24.090 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-08-21 18:01:24.090 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Scheduling 7 query requests. #0
2024-08-21 18:01:24.091 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] Querying (1028 - 1067 | 0x0404 - 0x042B # 040) ...
2024-08-21 18:01:24.091 INFO (MainThread) [custom_components.solarman.api] [2104753402] Connecting to 192.168.1.178:8899
2024-08-21 18:01:24.092 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] SENT: ee 51 00 00 00 06 01 03 04 04 00 28
2024-08-21 18:01:24.418 DEBUG (MainThread) [custom_components.solarman.api] [2104753402] V5_MISMATCH: ee 51 00 00 00 53 01 03 50 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 37 00 00 00 36 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 6f 00 04 bc 27 00 08 78 b8 01 d1
```

davidrapan, 2024-08-21:
Of course I forgot something:
```
    def _received_frame_is_valid(self, frame):
        if not self._passthrough:
            return super()._received_frame_is_valid(frame)
        return True
```
Can you add this method to the `PySolarmanV5AsyncWrapper` class?

lcavalli, 2024-08-21:
No worries, can do by myself...

lcavalli, 2024-08-21:
It is working now. Lets see the disconnection issue..

lcavalli, 2024-08-21:
Running for more than 15 minutes without disconnections!

davidrapan, 2024-08-21:
Excellent!

lcavalli, 2024-08-21:
Logs dumped, everything is working without communication errors anymore. Seems stable so far. I will try to revert back to original 10 seconds query cycle.

lcavalli, 2024-08-21:
Working well event at 10 seconds query cycle.

[22 more replies]

closed 2024-08-29 by davidrapan