davidrapan/ha-solarman #22: Not working second inverter

ks0002544, 2024-07-18:
I need to control several inverters. But it does not work. Only one inverter works.
But Solarman integration can work with three inverters.

davidrapan, 2024-07-18:
Hi, what does it do then? Can you maybe provide some logs with debugging enabled?

ks0002544, 2024-07-18:
can you check your code: I use connection to inverter via port transfering (from WAN to another WAN). Not inside local net.
real Solarman supports it.

ks0002544, 2024-07-18:
here is log
[home-assistant_solarman_2024-07-18T19-44-23.015Z.log](https://github.com/user-attachments/files/16288410/home-assistant_solarman_2024-07-18T19-44-23.015Z.log)

davidrapan, 2024-07-18:
> can you check your code: I use connection to inverter via port transfering (from WAN to another WAN). Not inside local net. real Solarman supports it.

That should not be a problem,

> here is log [home-assistant_solarman_2024-07-18T19-44-23.015Z.log](https://github.com/user-attachments/files/16288410/home-assistant_solarman_2024-07-18T19-44-23.015Z.log)

I need you to also enable debug here:
[image]

But
`V5FrameError: V5 frame contains incorrect data logger serial number`
looks quite suspicious...

Cheers!

ks0002544, 2024-07-18:
it is very strange

ks0002544, 2024-07-18:
it is very strange
I have 2 servers Deb 12 with HA installed both the same FW
These servers are located in other WAN
One server can connect to three inverters using your integration

davidrapan, 2024-07-18:
That's strange indeed...

But I want to help you debug your issue so to better understand your "setup" I have few questions. Are both mentioned servers behind the same WAN? Inverters are also behind the same WAN? So you are connecting between two WAN's like this WAN <-> WAN? And are using port forwarding on the routers? Or you are using tunneled connections?

_If it's the former I have to express some concerns due to security reasons. Exposing Solarman ports on the WAN is really not great Idea as the protocol don't understand what security is. So I hope you have at least some firewall rules in place which limits access to those exposed ports only from your other WAN where the servers sits._

ks0002544, 2024-07-18:
Yes I use port transfering on my routers.
I have
Wan1 - 2 x 3 phase inverters
Wan2 - server1 
Wan3 - server2 and 1x 1 phase inverter

davidrapan, 2024-07-18:
And which way it works and which on do not? Which server has mentioned issues?

ks0002544, 2024-07-18:
Server1 can connect to all three inverters
Server2 can connect only to 1 phase  inverter locally but no connection to 2x3 phase inverters

ks0002544, 2024-07-18:
But with real solarman integration it works

davidrapan, 2024-07-18:
Okay I have to say that at least this makes some sense...

Now we can say that the issue is probably with Wan1 configuration meaning maybe the router is blocking connection from Wan3 (server2). Could be some firewall rules which are allowing only IP from Wan2 and not from Wan3.

davidrapan, 2024-07-18:
> But with real solarman integration it works

What do you mean by real solarman integration? You mean the app or?

ks0002544, 2024-07-18:
I mean integration from Stephan

davidrapan, 2024-07-18:
> Server1 can connect to all three inverters Server2 can connect only to 1 phase inverter locally but no connection to 2x3 phase inverters

> I mean integration from Stephan

Please try to enable debug logging as I showed you on the picture before on server2. best will be to restart HA after you enable it and then let it run for few minutes and post the log here.

davidrapan, 2024-07-18:
> Yes I use port transfering on my routers. I have Wan1 - 2 x 3 phase inverters Wan2 - server1 Wan3 - server2 and 1x 1 phase inverter

Please also tell me your IP and S/N combination you are using in your setting in both servers (S/N only first few digits for example as 211XXXXXXX). So I can rule out some misconfiguration.

davidrapan, 2024-07-19:
I also found this regarding your issue [V5 frame contains incorrect data logger serial number](https://github.com/jmccrohan/pysolarmanv5/issues/14) and the problem in that case really was wrong serial number used. Probably serial number of the inverter instead of the one from Solarman Logging Stick. So I suggest you to try delete all inverters and add them again and double check you are putting correct serial numbers and ips.

**Keep in mind that I'm using Inverter autodiscovery so the serial number can be on your server prefilled!**

ks0002544, 2024-07-19:
so 
Wan1 - 2 x 3 phase inverters 192.168.1.95/274XXXXX and 192.168.1.91/279XXXXX
Wan2 - server1
Wan3 - server2 and 1x 1 phase inverter 192.168.1.24/290XXXXXX

Disabled discovery did not fix it

ks0002544, 2024-07-19:
this is the log
[home-assistant_solarman_2024-07-19_1.log](https://github.com/user-attachments/files/16302753/home-assistant_solarman_2024-07-19_1.log)

ks0002544, 2024-07-19:
I found on git
https://github.com/StephanJoubert/home_assistant_solarman/issues/419#issuecomment-1773924534

ks0002544, 2024-07-19:
log file 
[home-assistant_solarman_2024-07-19T18-53-43.199Z.log](https://github.com/user-attachments/files/16316147/home-assistant_solarman_2024-07-19T18-53-43.199Z.log)

davidrapan, 2024-07-19:
Great!

And I guess the IP you obfuscated (xxx.xxx.xxx.xxx) at first line with timestamp: 2024-07-19 21:39:32.364 is your WAN IP where the remote inverter sits right? Just to be clear.

So from the log it looks like it for some reason ignores the set IP and just connects to the local inverter even with discovery disabled so that is probably the issue. I'm gonna try to find out why it happens and I'll let you know!

Thanks for your cooperation!

ks0002544, 2024-07-19:
yes it is WAN where inverters is installed

davidrapan, 2024-07-19:
When did you updated the integration last time? And how did you do the update? Did you used HACS to do it or manually replaced the code? (Just making sure that we are both running the same code)

ks0002544, 2024-07-19:
I used both methods. But last time several times manually copied integration into folder solarman

davidrapan, 2024-07-19:
Yes manual copy is the proper way during "hot-fixing" like this cause HACS would download older version.

Could you try one more thing for me? The same thing as you did before add the device again (while having the debug enabled) but purposefully with wrong IP (some random local one for example)? And do it once with discovery enabled and once with discovery disabled.

Thanks again for your excessive help with this issue!

ks0002544, 2024-07-19:
[home-assistant_solarman_wrongIP_nonDiscovery.log](https://github.com/user-attachments/files/16316816/home-assistant_solarman_wrongIP_nonDiscovery.log)
[home-assistant_solarman_wrongIP_discovery.log](https://github.com/user-attachments/files/16316817/home-assistant_solarman_wrongIP_discovery.log)

davidrapan, 2024-07-19:
So I just committed [feat: Disable discovery for non private IPs](https://github.com/davidrapan/ha-solarman/commit/e6a548d11c1f9fbc63c4d495e2ebb691560ab59a) which should disable discovery for non private IPs by default. So update the code manually again and let me know if it worked.

And if won't I just get rid of it all together cause it already caused way more troubles than it was useful...

Cheers!

ks0002544, 2024-07-20:
[HA.log](https://github.com/user-attachments/files/16319033/HA.log)

ks0002544, 2024-07-20:
so, I noticed the following: I can connect to the 2 inverters from Wan2 to Wan1 because Wan2 has no any local inverter connected/discovered
otherwise no connection from Wan3 to Wan1 because solarman integration discovered one local inverter

Wan1 - 2 x 3 phase inverters 192.168.1.95/274XXXXX and 192.168.1.91/279XXXXX
Wan2 - server1
Wan3 - server2 and 1x 1 phase local inverter 192.168.1.24/290XXXXXX

davidrapan, 2024-07-20:
Ou sorry, I broke it:
```
2024-07-20 09:04:01.223 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Inverter for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 61, in async_setup_entry
    inverter = Inverter(inverter_host, inverter_serial, inverter_port, inverter_mb_slave_id, name, inverter_mac, lookup_path, lookup_file)
                                                                                                   ^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'inverter_mac' where it is not associated with a value
```

But it looks like it will work after I fix this. Gimme a sec! :)

davidrapan, 2024-07-20:
[fix: 'inverter_mac' not associated with a value](https://github.com/davidrapan/ha-solarman/commit/b2c9b8bb252deb7d367c09a307d57729225f9d47)

It should be all good now!

ks0002544, 2024-08-09:
So, 3 inverters are not working in one intergation but if I do separate integration for each inverter - it is works

davidrapan, 2024-08-10:
What do you mean by that? And always provide logs please.

davidrapan, 2024-09-18:
Closing because that shouldn't be an issue anymore.

closed 2024-09-18 by davidrapan