davidrapan/ha-solarman #23: [Deye Sun 8K Hybrid SG04LP3-EU] - Updating Solarman 1.4.3 from Stephen Joubert to HA-Solarman from David Rapan - Inverter Definition file

PV-Joe, 2024-07-21:
Hi David,

I gave it another ty to udate to your integration and I have the following problem:

**My hardware & software:**
- Deye Sun 8K Hybrid SG04LP3-EU
- LSW3 Stick Logger (SN 27xxxxxxxxxx) Firmware: LSW3_15_FFFF_1.0.91R
- Updating from Solarman 1.4.3 integration by Stephen Joubert with deyeSG04LP3.yaml selected

**After updating:**
- I deleted the configuration and added it again
- I selected "Solarman" as sensor prefix to match my old sensors
- I selected the deye_sg04lp3.yaml from your integration > many sensors aren't working anymore
- I selected my old deyeSG04LP3.yaml that was working before updating > Most of the sensors are working again and nearly everything is fine. The 5 seconds update interval was working fine.

**... but ...**

- The sensors that were newly added by your integration obviously don't work selecting my old YAML for configuration. 
- I gave up another ty to update to your configuration and restored the backup with Solarman 1.4.3 by Stephen.

I know that many users with my DEYE hardware configuration had problems with the latest Solarman 1.5.1 b Stephen Joubert are using the older 1.4.3 integration with success.

As I´m not a "Pro" user I really don't know why your deye_sg04lp3.yaml ist not working. I also tried the alternate deye_sun-12k-sg04lp3.yaml without success. I really don't know what is going on anymore and can't help myself to solve the problem. I guess that some modbus registers are different and my definition file hast to be combined with yours to make it work.
For now I#m back to "never change a running system" even though your inergration looks more tempting. I know you tried to gave so much support already but my understanding of what exactly I have to do do is limited. I wish there would be an update process for dummys like me ;-)

I will add my working deye_sg04lp3.yaml just in case others have the same problem and would like to try this inverter definition. Maybe it helps.

[deye_sg04lp3.yaml.txt](https://github.com/user-attachments/files/16323524/deye_sg04lp3.yaml.txt)

davidrapan, 2024-07-21:
Wo why you are simple not using this integration with just your definition file? There is nothing wrong with using it like that?

sofkaski, 2024-07-21:
I have the same setup. The inverter is hybrid Deye SUN-8K-SG04P3-EU with a logger with firmware version LSW3_15_FFFF_1.0.9E. (serial 27*). I have connected inverter also with Deye cloud.

I just day ago started using this instead of my own fork of latest Solarman by Stephan where I fixed some connection issues.  It was working fine, but there were issues that are fixed in this integration. So I decided to change to this.

I did it brutally by deleting the old integration and configuring a new with this one. I knew that I would loose some history data from the HA instance and would need to edit new sensor ids to cards in couple of dashboards (energy + some of my custom ones). I feared that it would be tedious, but actually it was not more than about 1 hour work. I have also [Spook](https://spook.boo/) installed and it reported nicely references to sensors that were not there any more.

I used unmodified `deye_sg04lp3.yaml` and so far everything seems to be ok. No unrecoverable communication errors etc. And, this integration has more data than the previous one that was using. One modification that I made (in my fork) is that I prolonged polling interval to 60 seconds that suits fine to my use cases and is a good compromise what it comes to amount of state data.

I haven't implemented any automations based on these measurements (yet). So, no comments on services.

I want to give kudos  to @davidrapan for his nice work!

PV-Joe, 2024-07-21:
Thanks for your kind words and your offer of personal support😄 Before giving up I will let you know and figure out more.
I‘ll probably give it another try. I was keen to get more sensors with your integration and you are right, I do panic to early. 😳
Next time I‘ll put up wit my inverter definition and that‘s it.
Love your support and many thanks David.

The Wiki has started great. I think this integration and the GitHub presentation has potential to make it the place to go for LSW3 Stick Logger users. 👍

davidrapan, 2024-07-21:
> I have the same setup. The inverter is hybrid Deye SUN-8K-SG04P3-EU with a logger with firmware version LSW3_15_FFFF_1.0.9E. (serial 27*). I have connected inverter also with Deye cloud.
>
> I just day [...]

That is really nice to hear. It's exactly why I do what I do. :-)

When comes to the losing history what does not have to be the case when you follow the steps here: [Sensor renaming](https://github.com/davidrapan/ha-solarman/wiki/Naming-Scheme-%E2%80%90-sensor-renaming)

Thanks again and have fun!

davidrapan, 2024-07-21:
> Thanks for your kind words and your offer of personal support😄 Before giving up I will let you know and figure out more. I‘ll probably give it another try. I was keen to get more sensors with your [...]

Look I get it I also hate when one loses history or who knows what other tricky situations can come up. That's why I tried to provide many ways how to deal with it. More so it would by a bad system which forces you resists any changes, etc. That's why I try to do things more free in that manner let's say. But here the thing is you don't even have to lose anything if you know how. I don't think that there are that many changes though. It's how much? around 7 sensors?

Anyway I'm always here ready to assist which anything.

Cheers!

sofkaski, 2024-07-22:
> And this is how you can adjust the update_interval globaly in the definition file:
> 
> ```
> default:
>   update_interval: 60
> ```
> 
> 60 is the default. But you will have to remove 'update_interval' from all other sensors as it has precedence over default value.
> 
Ah.. I noticed that default and was thinking that it was not working and so ended up modifying `const.py`. I did not realize that `update_interval` should have been removed in that case. Hmm.. I'll propably then make a custom definition file.

davidrapan, 2024-07-22:
> Ah.. I noticed that default and was thinking that it was not working and so ended up modifying `const.py`. I did not realize that `update_interval` should have been removed in that case. Hmm.. I'll [...]

It's somewhat not the best approach how it is done now. I know that. I was thinking maybe about adding another property something like 'update_interval_forced' or 'min_update_interval' for cases like yours so you doesn't have to edit whole profile file. But I wanted to think it through more thoroughly before any more changes.

Have fun!

PV-Joe, 2024-07-27:
Hi David,

just to let you know. I finally managed to upgrade to your Solarman integration and it works like a charm. Had to change some entity ID‘s from your naming back to my old ones and no values from the history were lost.

Totally happy now!
Many thanx for all your help and support. I really appreciated that.

You can close this now if you like.
Keep up the good work!

Joe

davidrapan, 2024-07-27:
> and no values from the history were lost.

Told ya! HA is quite clever in this regard!

But anyway I'm glad that you were able to go trough it successfully!

Have fun!

closed 2024-07-27 by davidrapan

PV-Joe, 2024-08-06:
Hi again @davidrapan,

sorry to bother you another time for the same problem!

**The state I'm in at the moment:**

1. You remember I changed my entityID`s from yours to my old ones from Stephen Joubert. Obviously that wasn't very intelligent as I have to do this every time I update to a new version from your integration.
2. As far as I understood, the better way is to change the entityID's to your new naming scheme before I update to the latest version to keep compatible with future updates.
3. I have read your Wiki from here: [https://github.com/davidrapan/ha-solarman/wiki/Naming-Scheme#sensor-renaming-w-history](url)
4. The idea is to change my current entityID's to match yours in advance and then update to the latest version. This should preserve all history data as far as I understand and saves me from renaming after each update. Am I right?

To make the process of renaming enityID's easier for me I would kindly ask you for a **total list of the current entityID namings**, so I can work my way through before updating to v24.08.04.
**I'm talking about the right entityID naming with the required underscores for the sensors like this for example: sensor.solarman_total_grid_power** in a complete list.

That list could even be used by other Users that update from individual naming schemes or Stephen Jouberts version and will be very helpful to get the system ready without any hassle and bothering you again.
Maybe the best way is to post the list into the Wiki for others to be prepared?

**Can you provide the total list of entityID's from your integration v24.08.04 and post it here please?**

As always,
Keep up with your excellent work and many thanks in advance!

davidrapan, 2024-08-06:
```
['sensor.inverter_diagnostics_test_sensor', 'select.inverter_total_energy_bought', 'sensor.inverter_connection_status', 'sensor.inverter_update_interval', 'sensor.inverter_device', 'sensor.inverter_smartload_state', 'sensor.inverter_battery_bms_alarm', 'sensor.inverter_device_state', 'sensor.inverter_today_battery_charge', 'sensor.inverter_today_battery_discharge', 'sensor.inverter_total_battery_charge', 'sensor.inverter_total_battery_discharge', 'sensor.inverter_today_battery_life_cycles', 'sensor.inverter_total_battery_life_cycles', 'sensor.inverter_today_energy_bought', 'sensor.inverter_today_energy_sold', 'sensor.inverter_total_energy_bought', 'sensor.inverter_total_energy_sold', 'sensor.inverter_today_load_consumption', 'sensor.inverter_total_load_consumption', 'sensor.inverter_today_losses', 'sensor.inverter_total_losses', 'sensor.inverter_today_production', 'sensor.inverter_total_production', 'sensor.inverter_temperature', 'sensor.inverter_device_power_state', 'sensor.inverter_device_relay_state', 'sensor.inverter_device_alert', 'sensor.inverter_device_fault', 'sensor.inverter_battery_temperature', 'sensor.inverter_battery_voltage', 'sensor.inverter_battery', 'sensor.inverter_battery_power', 'sensor.inverter_battery_power_inverted', 'sensor.inverter_battery_soh', 'sensor.inverter_battery_state', 'sensor.inverter_battery_current', 'sensor.inverter_grid_state', 'sensor.inverter_grid_l1_voltage', 'sensor.inverter_grid_l2_voltage', 'sensor.inverter_grid_l3_voltage', 'sensor.inverter_internal_ct1_power', 'sensor.inverter_internal_ct2_power', 'sensor.inverter_internal_ct3_power', 'sensor.inverter_internal_power', 'sensor.inverter_internal_ct1_current', 'sensor.inverter_internal_ct2_current', 'sensor.inverter_internal_ct3_current', 'sensor.inverter_external_ct1_current', 'sensor.inverter_external_ct2_current', 'sensor.inverter_external_ct3_current', 'sensor.inverter_external_ct1_power', 'sensor.inverter_external_ct2_power', 'sensor.inverter_external_ct3_power', 'sensor.inverter_external_power', 'sensor.inverter_grid_l1_power', 'sensor.inverter_grid_l2_power', 'sensor.inverter_grid_l3_power', 'sensor.inverter_grid_power', 'sensor.inverter_grid_power_inverted', 'sensor.inverter_inverter_l1_voltage', 'sensor.inverter_inverter_l2_voltage', 'sensor.inverter_inverter_l3_voltage', 'sensor.inverter_inverter_l1_current', 'sensor.inverter_inverter_l2_current', 'sensor.inverter_inverter_l3_current', 'sensor.inverter_inverter_l1_power', 'sensor.inverter_inverter_l2_power', 'sensor.inverter_inverter_l3_power', 'sensor.inverter_inverter_power', 'sensor.inverter_power_losses', 'sensor.inverter_inverter_frequency', 'sensor.inverter_load_ups_l1_power', 'sensor.inverter_load_ups_l2_power', 'sensor.inverter_load_ups_l3_power', 'sensor.inverter_load_ups_power', 'sensor.inverter_load_l1_voltage', 'sensor.inverter_load_l2_voltage', 'sensor.inverter_load_l3_voltage', 'sensor.inverter_load_l1_power', 'sensor.inverter_load_l2_power', 'sensor.inverter_load_l3_power', 'sensor.inverter_load_power', 'sensor.inverter_smartload_l1_voltage', 'sensor.inverter_smartload_l2_voltage', 'sensor.inverter_smartload_l3_voltage', 'sensor.inverter_smartload_l1_power', 'sensor.inverter_smartload_l2_power', 'sensor.inverter_smartload_l3_power', 'sensor.inverter_smartload_power', 'sensor.inverter_pv_power', 'sensor.inverter_pv1_power', 'sensor.inverter_pv2_power', 'sensor.inverter_pv1_voltage', 'sensor.inverter_pv1_current', 'sensor.inverter_pv2_voltage', 'sensor.inverter_pv2_current', 'switch.inverter', 'switch.inverter_battery_grid_charging', 'switch.inverter_program_1_grid_charging', 'switch.inverter_program_2_grid_charging', 'switch.inverter_program_3_grid_charging', 'switch.inverter_program_4_grid_charging', 'switch.inverter_program_5_grid_charging', 'switch.inverter_program_6_grid_charging', 'number.inverter_battery_equalization', 'number.inverter_battery_absorption', 'number.inverter_battery_float', 'number.inverter_battery_capacity', 'number.inverter_battery_empty', 'number.inverter_zero_export_power', 'number.inverter_battery_equalization_cycle', 'number.inverter_battery_equalization_time', 'number.inverter_battery_temperature_compensation', 'number.inverter_battery_max_charging_current', 'number.inverter_battery_max_discharging_current', 'number.inverter_battery_grid_charging_current', 'number.inverter_smartload_off', 'number.inverter_smartload_on', 'number.inverter_program_1_soc', 'number.inverter_program_2_soc', 'number.inverter_program_3_soc', 'number.inverter_program_4_soc', 'number.inverter_program_5_soc', 'number.inverter_program_6_soc', 'select.inverter_battery_control_mode', 'select.inverter_battery_operation_mode', 'select.inverter_smartload_mode', 'time.inverter_program_1_time', 'time.inverter_program_2_time', 'time.inverter_program_3_time', 'time.inverter_program_4_time', 'time.inverter_program_5_time', 'time.inverter_program_6_time', 'select.inverter_energy_pattern', 'switch.inverter_export_surplus', 'switch.inverter_time_of_use', 'number.inverter_program_1_power', 'number.inverter_program_2_power', 'number.inverter_program_3_power', 'number.inverter_program_4_power', 'number.inverter_program_5_power', 'number.inverter_program_6_power', 'number.inverter_program_1_voltage', 'number.inverter_program_2_voltage', 'number.inverter_program_3_voltage', 'number.inverter_program_4_voltage', 'number.inverter_program_5_voltage', 'number.inverter_program_6_voltage', 'number.inverter_battery_resistance', 'number.inverter_battery_ [... cut ...] y', 'number.inverter_battery_shutdown_soc', 'number.inverter_battery_restart_soc', 'number.inverter_battery_low_soc', 'number.inverter_battery_shutdown_voltage', 'number.inverter_battery_restart_voltage', 'number.inverter_battery_low_voltage', 'sensor.inverter_today_self_consumed', 'sensor.inverter_today_energy_bought_t1', 'sensor.inverter_today_energy_bought_t2', 'sensor.inverter_total_energy_bought_t1', 'sensor.inverter_total_energy_bought_t2', 'device_tracker.deye_hybrid_inverter_logger']
```

PV-Joe, 2024-08-06:
Thank you for your fast response and the list!

Started to rename the entityID from Stephen Joubert to the name by David Rapan.

… and guess what?

1. The history is gone and starts all over again.
2. Doing it the other way round: Renaming the entityID from David to Stephen’s preserved the history! The problem is that future updates will overwrite the entityID and you have to start all over again.

Searching the internet for a solution how to migrate history data to a new entityID came up with the information that it can take up to 24 hours until the history data will merge:

Source:
https://github.com/home-assistant/core/pull/89963

So I changed just one entityID and will be patient and see if it works after a while.

Keeping you informed if it will succeed or fails.

davidrapan, 2024-08-06:
You had to make mistake somewhere. I did it really many, many times and was successful every time. (Or maybe I forget to mention something in the steps). I'll double check!

davidrapan, 2024-08-06:
Make sure that the sensor to which you are doing the rename to does not exist! Meaning even the history. You can make sure using Developer Tools. In statistics page.

davidrapan, 2024-08-06:
What is proly happening in your situation is:
1. You added device with new entity names thus that created history for that new namíée already.
2. You changed entity names into you old one.
3. When you did the switch back. Because entities with that new names already existed it refused to transfer the history so the history stayed with your old entity names.

What  you have to do before renaming to new names:
1. First you need to enable Developer Tools (If you don't have already)
2. In Statistics page you will see entities with new naming with Fix button. you have to clear that entities using that button.
3. Then you can change your old names to new names.

You should see the history transferred immediately.

PV-Joe, 2024-08-07:
Just to let you know.
You were absolutely right 😄

The Statistics under Developer Tools was full of Solarman sensors. After deleting those I was able to rename entityID‘s to your logic and history was still there.

I‘m working my way through and have to edit the dashboards as well now.

After that is done, I‘m ready to update to your latest version and hope not to mess it up again.

davidrapan, 2024-08-07:
Magnificent! 🥳

PV-Joe, 2024-08-10:
I can’t believe it!
I finally managed to update to v2024.08.04 and everything including history works very well.
I’m totally satisfied now and I’m ready for future Updates.

Don’t even think about renaming any sensors please! It was a hell renaming and migrating all the entityID’s dashboards and automations.
Leave it compatible to earlier versions is the way to go in my opinion.

Have a sunny day ☀️