davidrapan/ha-solarman #27: Startup error

sofkaski, 2024-07-22:
I noticed an error in system startup when I needed to shutdown the Raspberry hosting my HA instance to resuffle power cabling.
The sequence was the following:

1. I shutdown the host (`sudo shutdown -h now`)
3. Pulled Rasberry power cable off and fiddled with cabling
4. Plugged rasberry power source back -> system boot
5. Solarman integration did not start up and there was the following in the log:
```
Error setting up entry Inverter for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 61, in async_setup_entry
    inverter = Inverter(inverter_host, inverter_serial, inverter_port, inverter_mb_slave_id, name, inverter_mac, lookup_path, lookup_file)
                                                                                                   ^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'inverter_mac' where it is not associated with a value
2024-07-22 17:38:27.164 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform bms_connector is taking over 10 seconds.
2024-07-22 17:38:31.817 WARNING (MainThread) [homeassistant.components.energy.sensor] Found unexpected state_class None for sensor.inverter_today_energy_bought
2024-07-22 17:38:31.817 WARNING (MainThread) [homeassistant.components.energy.sensor] Found unexpected state_class None for sensor.inverter_today_energy_sold
```
6. Reloaded the solarman integration -> the integration started normally

sofkaski, 2024-07-22:
Yep. I can confirm that the current main works with the sequence I mentioned. (I updated my fork to latest upstream)

closed 2024-07-22 by sofkaski

davidrapan, 2024-07-22:
No worries. 😉