davidrapan/ha-solarman #28: Configuring an existing device causes sensor naming issue

gedger, 2024-07-22:
If you configure an existing device the prefix name is not shown in the dialogue window. On pressing submit all sensors are renamed with the "None" prefix.

[image]
[image]

davidrapan, 2024-07-22:
Yes I noticed that too today. ConfigFlow and OptionsFlow still works (for me) in a mysterious ways. Common sense says that name should not be there as it's updateable from menu under those dots but I guess I'll put it back.

Thanks!

davidrapan, 2024-07-22:
[3808a54](https://github.com/davidrapan/ha-solarman/commit/3808a5412042650b35828bcd1debd2c070f7d390)

gedger, 2024-07-23:
Fixed although as you say it does make more sense to hide the name and use the rename function from the ... menu.

closed 2024-07-23 by gedger

davidrapan, 2024-07-23:
Yep exactly, will see in the future if I found out how it's supposed to work. Development documentation for HA is I don't wanna say bad, it's okay. But it has some holes here and there...

Cheers!