davidrapan/ha-solarman #29: Configuration Dialogue has an unnamed field.

gedger, 2024-07-22:
Configuration shows a field with no identification or description.

[image]

davidrapan, 2024-07-22:
That's a modbus id right there. I'll check it. Thanks.

davidrapan, 2024-07-22:
[3808a54](https://github.com/davidrapan/ha-solarman/commit/3808a5412042650b35828bcd1debd2c070f7d390)

gedger, 2024-07-23:
No change, still unnamed.

davidrapan, 2024-07-23:
Does you translation files containt "inverter_mb_slave_id" string?

gedger, 2024-07-23:
Confirmed translation is there. It's not really an issue so park it until someone else reports it. Could be something odd about my system.

gedger, 2024-07-24:
And magically it has now appeared 😕

closed 2024-07-24 by gedger

davidrapan, 2024-07-24:
Magic! hehe