davidrapan/ha-solarman #30: update_interval not working

gedger, 2024-07-22:
Adding the default section to the inverter configuration and then restarting HA does not change the update rate. Code from commit [c55af98](https://github.com/davidrapan/ha-solarman/commit/c55af98c75c62a2938c08f2c2064e501db7a5186)

```
default:
  update_interval: 30
  digits: 6
```

davidrapan, 2024-07-22:
Run it with debug enabled please a place the log here.

_Don't forget to obfuscate the serial._

gedger, 2024-07-22:
Here we go.
[home-assistant_solarman_2024-07-22T17-04-45.682Z.log](https://github.com/user-attachments/files/16337842/home-assistant_solarman_2024-07-22T17-04-45.682Z.log)

davidrapan, 2024-07-22:
Could you also show me how your inverter profile looks like?

davidrapan, 2024-07-22:
Ah never mind I know what is the issue...

> Registers which will be part of a request are decided dynamically (when not set in the inverter profile)
> - Can be requested individually in specific intervals according to their 'update_interval' value set in the inverter profile file (defaults to 60 seconds)

davidrapan, 2024-07-22:
[5a93aa1](https://github.com/davidrapan/ha-solarman/commit/5a93aa10ead0b697c81da26606718028f1f5376a)

```
default:
  update_interval: 30
  code: 0x04
  digits: 6
```

And delete whole requests segment.

davidrapan, 2024-07-23:
I still barely understand this implementation cause there is so little whings which really makes sense but yeah I would love to support as much as possible. :-)

When comes to the log.. oh god that's strange... can you show me your profile file?

davidrapan, 2024-07-23:
[f422188](https://github.com/davidrapan/ha-solarman/commit/f4221889ad6be8f1cfffe5b4d95f6698289f0316)

gedger, 2024-07-23:
Fixed it and the update_interval issue :clap:

closed 2024-07-23 by gedger