davidrapan/ha-solarman #36: Updating release deletes custom_parameters.yaml

gedger, 2024-07-25:
StephansJoubert original integration did not delete the file custom_parameters.yaml when used in the inverter_definitions folder. This was to allow tweaks to be made of an existing parameters file and ensure that it survive an update. This functionality appears to have been lost with the latest 24.07.24 update.

gedger, 2024-07-25:
https://github.com/StephanJoubert/home_assistant_solarman/blob/main/customization.md

davidrapan, 2024-07-25:
Yeah sure but there is nothing what says to hacs to not delete it though...

gedger, 2024-07-25:
I just tried re-downloading from HACS and it definitely removes it, no idea how SJ achieved it....I'll have a dig..

davidrapan, 2024-07-25:
it does, that's for sure. Maybe it's cause of the fact that this integration is added as custom repository?

davidrapan, 2024-07-25:
[v24.07.25 pre-release](https://github.com/davidrapan/ha-solarman/releases/tag/v24.07.25)

**It's important to redownload this beta (pre-release) version twice for hacs configuration to take effect!**

gedger, 2024-07-25:
Confirmed working.

closed 2024-07-25 by gedger