davidrapan/ha-solarman #39: Unable to add configuration

cyh0928, 2024-07-26:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 59, in async_setup_entry
    inverter = Inverter(inverter_host, inverter_serial, inverter_port, inverter_mb_slave_id, name, inverter_mac, lookup_path, lookup_file)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 132, in __init__
    super().__init__(address, serial, port, mb_slave_id, AUTO_RECONNECT)
  File "/config/custom_components/solarman/api.py", line 23, in __init__
    super().__init__(address, serial, port = port, mb_slave_id = mb_slave_id, logger = _LOGGER, auto_reconnect = auto_reconnect, socket_timeout = TIMINGS_SOCKET_TIMEOUT)
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 61, in __init__
    self.data_wanted_ev = Event()
                          ^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 93, in Event
    return Event(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 331, in __init__
    self._cond = ctx.Condition(ctx.Lock())
                               ^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 169, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system

[image]
Hello David, may I ask what is wrong with this?

davidrapan, 2024-07-26:
Hello! What device and what inverter profile? Were you migrating or is it fresh use of Solarman Integration?

Did you try to remove the device and then re-add?

Could you please translate from me the error message on the picture?

Thanks!

cyh0928, 2024-07-26:
The image means there is no device or entity

Collection device versionLSW3_15_840B_1.38
The inverter is megarevo，Modify and replace the configuration file yourself

cyh0928, 2024-07-26:
> 您好！什么设备和什么逆变器配置文件？您正在迁移还是刚刚开始使用 Solarman Integration？
> 
> 您是否尝试过移除该设备然后重新添加？
> 
> 您能帮我翻译一下图片上的错误信息吗？
> 
> 编辑：同时我[在这里](https://community.home-assistant.io/t/os-12-2-oserror-errno-30-read-only-file-system/715911)发现了一些有关您的错误消息的信息。
> 
> 谢谢！

> Hello! What device and what inverter profile? Were you migrating or is it fresh use of Solarman Integration?
> 
> Did you try to remove the device and then re-add?
> 
> Could you please translate from me the error message on the picture?
> 
> Edit: Meanwhile I found something about your error message [here](https://community.home-assistant.io/t/os-12-2-oserror-errno-30-read-only-file-system/715911).
> 
> Thanks!
I am migrating to use Solarman Integration and have tried deleting and re-adding

cyh0928, 2024-07-26:
> Hello! What device and what inverter profile? Were you migrating or is it fresh use of Solarman Integration?
>
> Did you try to remove the device and then re-add?
>
> Could you please translate from me [...]

I am migrating to use Solarman Integration and have tried deleting and re-adding

davidrapan, 2024-07-26:
Could you please maybe try to enable debug logging and then remove and add device with debug enabled? And upload full log here? (Don't forget to obfuscate the serial number)

cyh0928, 2024-07-26:
> Could you please maybe try to enable debug logging and then remove and add device with debug enabled? And upload full log here? (Don't forget to obfuscate the serial number)

Thanks David, I tried downgrading to HA OS 12.1 and it works fine now
[image]

davidrapan, 2024-07-26:
Great!

davidrapan, 2024-07-26:
Issue caused by HA OS 12.2.

Either downgrade to 12.1 or update to at least HA Supervisor 2024.04.0 as mentioned [here](https://community.home-assistant.io/t/os-12-2-oserror-errno-30-read-only-file-system/715911).

closed 2024-07-26 by davidrapan

VAdamec, 2024-08-15:
Running `ha core rebuild` fix this issue for me

```
Core: 2024.8.1
Supervisor: 2024.08.0
```

vinceg0267, 2024-09-02:
I have HA 2024.08.0 but not work

davidrapan, 2024-09-02:
@vinceg0267, did you run the command in the cli?

vinceg0267, 2024-09-02:
Excuse me, what is cli?

davidrapan, 2024-09-02:
https://www.home-assistant.io/blog/2019/02/04/introducing-home-assistant-cli/

What type of the HA installation are you running? HAOS or any of the other ones?

vinceg0267, 2024-09-02:
Ok, I did it with ssh terminal, thank you very much

davidrapan, 2024-09-02:
You are very welcome!