davidrapan/ha-solarman #43: Deye Inverter with LSW3_15_FFFF_1.0.9E

sebastiangebosz, 2024-07-27:
HELLO. I've used Stefan's integration so far. Version 1.4.3 only works. I wanted to try yours. Unfortunately it doesn't connect. I'm pasting the log. Can you help with this?

`2024-07-27 18:10:54.518 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'data': {'name': 'Falownik deye', 'inverter_discovery': True, 'inverter_host': '192.168.1.7', 'inverter_serial': 1782345394, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_string.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J3TF16B86WQXD9BQ5RQEDQ87', 'minor_version': 1, 'options': {'name': 'Falownik', 'inverter_discovery': True, 'inverter_host': '192.168.1.7', 'inverter_serial': 1782345394, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Falownik deye', 'unique_id': None, 'version': 1})
2024-07-27 18:10:54.518 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-07-27 18:10:54.573 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.1.7, 34EAE7B65472, 1782345394]
2024-07-27 18:10:55.862 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', '34:ea:e7:b6:54:72')}, 'identifiers': {('solarman', 1782345394)}, 'name': 'Falownik', 'manufacturer': 'Deye', 'model': 'HYBRID', 'serial_number': 1782345394}
2024-07-27 18:10:55.862 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-07-27 18:10:55.862 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 3, min_span: 25, digits: 6
2024-07-27 18:10:55.863 DEBUG (MainThread) [custom_components.solarman.api] Scheduling 4 query requests. #0
2024-07-27 18:10:55.864 DEBUG (MainThread) [custom_components.solarman.api] Querying (3 - 14) ...
2024-07-27 18:10:55.864 INFO (MainThread) [custom_components.solarman.api] Connecting to 192.168.1.7:8899
2024-07-27 18:10:55.893 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] SENT: a5 17 00 10 45 75 00 b2 6e 3c 6a 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 40 15
2024-07-27 18:10:56.027 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] RECD: a5 2d 00 10 15 75 a6 b2 6e 3c 6a 02 01 ae 0d ee 00 b1 10 00 00 60 0e b7 65 01 03 18 32 31 30 36 32 33 34 32 35 38 00 00 00 01 00 00 00 00 00 00 39 89 02 29 41 0d 00 00 83 15
2024-07-27 18:10:56.028 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [error: unpack requires a buffer of 24 bytes]
2024-07-27 18:10:57.029 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 4
2024-07-27 18:10:57.030 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-27 18:10:57.040 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] Successful reconnect
2024-07-27 18:10:57.041 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] SENT: a5 17 00 10 45 76 00 b2 6e 3c 6a 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 41 15
2024-07-27 18:10:57.170 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] RECD: a5 2d 00 10 15 76 a7 b2 6e 3c 6a 02 01 b0 0d ee 00 b2 10 00 00 60 0e b7 65 01 03 18 32 31 30 36 32 33 34 32 35 38 00 00 00 01 00 00 00 00 00 00 39 89 02 29 41 0d 00 00 88 15
2024-07-27 18:10:57.171 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [error: unpack requires a buffer of 24 bytes]
2024-07-27 18:10:58.173 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 3
2024-07-27 18:10:58.173 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-27 18:10:58.179 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] Successful reconnect
2024-07-27 18:10:58.180 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] SENT: a5 17 00 10 45 77 00 b2 6e 3c 6a 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 42 15
2024-07-27 18:10:58.436 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] RECD: a5 2d 00 10 15 77 a8 b2 6e 3c 6a 02 01 b1 0d ee 00 b3 10 00 00 60 0e b7 65 01 03 18 32 31 30 36 32 33 34 32 35 38 00 00 00 01 00 00 00 00 00 00 39 89 02 29 41 0d 00 00 8c 15
2024-07-27 18:10:58.437 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [error: unpack requires a buffer of 24 bytes]
2024-07-27 18:10:59.439 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 2
2024-07-27 18:10:59.440 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-27 18:10:59.445 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] Successful reconnect
2024-07-27 18:10:59.446 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] SENT: a5 17 00 10 45 78 00 b2 6e 3c 6a 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 43 15
2024-07-27 18:10:59.573 DEBUG (MainThread) [custom_components.solarman.api] [1782345394] RECD: a5 2d 00 10 15 78 a9 b2 6e 3c 6a 02 01 b2 0d ee 00 b4 10 00 00 60 0e b7 65 01 03 18 32 31 30 36 32 33 34 32 35 38 00 00 00 01 00 00 00 00 00 00 39 89 02 29 41 0d 00 00 90 15
2024-07-27 18:10:59.573 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [error: unpack requires a buffer of 24 bytes]
`

davidrapan, 2024-07-28:
Hi! Thank you for your post here!

I have some general idea why it is. So 1.4.3 of Stephans integration uses it's own implementation of solarman protocol. Newer version and then also mine integration uses pysolarmanv5 library which handles that.

That being sad I have planned to also implement some sort of a support for also ethernet stick which would involve adding that legacy implementation for that case so maybe I could also solve your issue with that. But it has a catch I would need your tight cooperation with me cause I have no way how to test it all afterwards, so...

What do you say?

Cheers!

Soltinioo, 2024-07-28:
Sure. As soon as you create something I will test it. From what I saw on Stefan's github, many people only run version 1.4.3. We will all be grateful to you if there are finally no errors in the logs. 😀

davidrapan, 2024-07-28:
Can you try something for me? Can you remove from the inverter definition profile which you are using (I believe it's deye_string?) that last sensor?:
```
- name: "Inverter ID"
  class: ""
  state_class: ""
  uom: ""
  scale: 1
  rule: 5
  registers: [0x0003, 0x0004, 0x0005, 0x0006, 0x0007]
```
And run it without it? Turn on the debug log, restart the HA and show me to logs please.

Thanks.

Soltinioo, 2024-07-29:
[error_log-1.txt](https://github.com/user-attachments/files/16411718/error_log-1.txt)

davidrapan, 2024-07-29:
Can you tell me the exact model of the inverter you are using?

Soltinioo, 2024-07-29:
inverter: DEYE SUN-9K-G03
Logger: LSW3 firmware LSW3_15_FFFF_1.0.9E

davidrapan, 2024-07-29:
Hi @Soltinioo, so in [1af1143](https://github.com/davidrapan/ha-solarman/commit/1af11436f4348dce5bd116130c61fb3e1ead8d61) is added some experimental support for your case. Could you try it?

_You will have to manually download and override the latest code from this repo though as those changes are not yet part of any release._

Soltinioo, 2024-07-30:
I did what you said. Everything works great! No errors in the logs! Thank you very much! I'm sure everyone using version 1.4.3 will switch to this integration 😁

davidrapan, 2024-07-30:
Great. Can you maybe gather some debug logs for us?

Best would couple of random samples from multiple different days? That would be awesome! :)

Soltinioo, 2024-07-30:
Sure, I'll be watching the integration. I'll post the logs soon

satyamedh, 2024-08-01:
Heya! Running into the same issue here, similar model. I ended up modifying the two files in that commit and it *does* work now, but the power is quite different from the stick logger's online web interface and the one from HA

attached screenshots
[image]
[image]

Not sure if you'd prefer a new issue for this as it's based on an unreleased commit, so just posting here

davidrapan, 2024-08-01:
Stick Logger local web is broken and I never saw it to show correct values with any inverter device.

satyamedh, 2024-08-01:
oh is it? aight then thanks!

davidrapan, 2024-08-01:
Yeah, mine (SG04LP3) for example shows with every restart random and totally different value than it should.

closed 2024-08-05 by davidrapan

davidrapan, 2024-09-29:
Hi @Soltinioo, not sure if this message will find you but I have a question for you. Could you please tell me what type of inverter it shows you in the `Device` sensor?

Thank you.

Soltinioo, 2024-09-29:
> 
[image]
is this what you mean?

davidrapan, 2024-09-29:
Magnificent! Thank you very much!