davidrapan/ha-solarman #44: Today‘s Power Losses with SG04LP3.yaml for DEYE 8K Hybrid shows high value

PV-Joe, 2024-07-28:
Hi David,

are you or anyone else able to explain the value of „Today‘s Power Losses“ to me? I think this value is not calculated the right way maybe?

**My configuration:**
Deye 8K Hybrid SG04 LP3-EU

**Inverter_definition file: (unedited orriginal file from release v24.07.24):**
 deye_sg04lp3.yaml

**Result:**
[image]
(These values are from 0:00 until 8:00 o‘clock)

Thanx again for your help!
Joe

davidrapan, 2024-07-28:
I guess it should be 0,4 right?

Here is mine:
[image]

PV-Joe, 2024-07-28:
Yes sure, here we go:

[image]
[image]

davidrapan, 2024-07-28:
Gosh If that would be right I would feel so sorry for u, lol!

Guess I'm gonna need Also the other values from the equation used...

My thoughts are that it's a scaling issue but if you used the same file (or copy pasted sensors even without changes) it's weird cause I'm using it too!

davidrapan, 2024-07-28:
Ah! forget everything! I know where is the issue! You proly have a version where is a typo!

Try find in that definition file: "soperator"

HAHA!

PV-Joe, 2024-07-28:
That was a quick fix 😄

I corrected the typo in:
deye_sg04lp3.yaml

From: soperator
To: operator

Result after fix:
[image]

Well done David!
I think it will be fixed permanently with your next release right?

Keep going ☀️

davidrapan, 2024-07-28:
Yes it's already committed and also as part of pre-releases I think. ;)

davidrapan, 2024-07-28:
Leave this Issue open for some time for maybe other which could experience the same thing. ;)

closed 2024-08-05 by davidrapan