davidrapan/ha-solarman #46: Failed setup

orangedmind, 2024-07-29:
Hi,
I tried setting up your integration but if fails when trying to connect:
[image]

Please find below the home-assistant.log

2024-07-29 16:26:36.229 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'data': {'name': 'Solarman', 'inverter_discovery': True, 'inverter_host': '192.168.10.52', 'inverter_serial': 4176382624, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J3ZE3X2CHZRAHA6FZMGDVZ2C', 'minor_version': 1, 'options': {'name': 'Solarman', 'inverter_discovery': True, 'inverter_host': '192.168.10.52', 'inverter_serial': 4176382624, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Solarman', 'unique_id': None, 'version': 1})
2024-07-29 16:26:36.229 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-07-29 16:26:36.236 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.10.52, E8FDF87C2F24, 4176382624]
2024-07-29 16:26:37.366 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', 'e8:fd:f8:7c:2f:24')}, 'identifiers': {('solarman', 4176382624)}, 'name': 'Solarman', 'manufacturer': 'Deye', 'model': 'HYBRID', 'serial_number': 4176382624}
2024-07-29 16:26:37.366 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-07-29 16:26:37.366 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 3, min_span: 25, digits: 6
2024-07-29 16:26:37.367 DEBUG (MainThread) [custom_components.solarman.api] Scheduling 4 query requests. #0
2024-07-29 16:26:37.367 DEBUG (MainThread) [custom_components.solarman.api] Querying (3 - 14) ...
2024-07-29 16:26:37.367 INFO (MainThread) [custom_components.solarman.api] Connecting to 192.168.10.52:8899
2024-07-29 16:26:37.392 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 9b 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf b0 15
2024-07-29 16:26:37.482 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 2b 00 10 15 9b b4 a0 8a ee f8 02 01 89 e2 35 00 58 12 00 00 12 c4 71 66 01 03 18 32 32 31 30 32 39 30 32 35 33 00 01 00 00 12 0c 07 00 00 00 01 14 12 12 50 f4 22 15
2024-07-29 16:26:37.483 DEBUG (MainThread) [custom_components.solarman.api] Querying succeeded.
2024-07-29 16:26:37.483 DEBUG (MainThread) [custom_components.solarman.api] Querying (59 - 112) ...
2024-07-29 16:26:37.483 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-29 16:26:37.503 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-29 16:26:37.503 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 9c 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3b 00 36 b4 11 54 15
2024-07-29 16:26:37.683 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 7f 00 10 15 9c b5 a0 8a ee f8 02 01 8a e2 35 00 58 12 00 00 12 c4 71 66 01 03 6c 00 04 00 13 00 00 00 00 1e 7b 00 00 00 09 00 09 00 00 00 00 0e c5 00 00 0e a7 00 00 09 7e 00 00 00 00 00 17 00 00 00 00 13 7d 00 00 00 00 00 00 00 00 00 00 00 00 15 f4 00 00 00 00 00 00 16 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 3e 00 5c 01 4c 00 51 13 55 7a 15
2024-07-29 16:26:37.684 DEBUG (MainThread) [custom_components.solarman.api] Querying succeeded.
2024-07-29 16:26:37.685 DEBUG (MainThread) [custom_components.solarman.api] Querying (150 - 195) ...
2024-07-29 16:26:37.685 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-29 16:26:37.703 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-29 16:26:37.703 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 9d 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 96 00 2e 25 fa 02 15
2024-07-29 16:26:37.876 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 6f 00 10 15 9d b6 a0 8a ee f8 02 01 8a e2 35 00 58 12 00 00 12 c4 71 66 01 03 5c 0a be 07 08 14 50 12 8e f7 2c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 48 1a 15
2024-07-29 16:26:37.877 DEBUG (MainThread) [custom_components.solarman.api] Querying succeeded.
2024-07-29 16:26:37.877 DEBUG (MainThread) [custom_components.solarman.api] Querying (244 - 279) ...
2024-07-29 16:26:37.877 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-29 16:26:37.895 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-29 16:26:37.896 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 9e 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 f4 00 24 04 23 5f 15
2024-07-29 16:26:39.919 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 10 00 10 15 9e b7 a0 8a ee f8 02 01 8c e2 35 00 5a 12 00 00 12 c4 71 66 05 00 5e 15
2024-07-29 16:26:39.920 WARNING (MainThread) [custom_components.solarman.api] Querying (244 - 279) failed. #0 [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-07-29 16:26:40.920 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 4
2024-07-29 16:26:40.921 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-29 16:26:40.941 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-29 16:26:40.941 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 9f 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 f4 00 24 04 23 60 15
2024-07-29 16:26:42.983 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 10 00 10 15 9f b8 a0 8a ee f8 02 01 8f e2 35 00 5e 12 00 00 12 c4 71 66 05 00 67 15
2024-07-29 16:26:42.984 WARNING (MainThread) [custom_components.solarman.api] Querying (244 - 279) failed. #0 [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-07-29 16:26:43.985 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 3
2024-07-29 16:26:43.985 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-29 16:26:44.008 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-29 16:26:44.008 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 a0 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 f4 00 24 04 23 61 15
2024-07-29 16:26:46.040 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 10 00 10 15 a0 b9 a0 8a ee f8 02 01 92 e2 35 00 61 12 00 00 12 c4 71 66 05 00 6f 15
2024-07-29 16:26:46.040 WARNING (MainThread) [custom_components.solarman.api] Querying (244 - 279) failed. #0 [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2024-07-29 16:26:47.041 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 2
2024-07-29 16:26:47.042 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-29 16:26:47.064 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-29 16:26:47.064 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 a1 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 f4 00 24 04 23 62 15
2024-07-29 16:26:51.067 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 10 00 10 15 a1 ba a0 8a ee f8 02 01 97 e2 35 00 66 12 00 00 12 c4 71 66 05 00 7b 15
2024-07-29 16:26:51.067 WARNING (MainThread) [custom_components.solarman.api] Querying (244 - 279) failed. #0 [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
[... cut ...]
2024-07-29 16:26:55.128 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Solarman data in 17.762 seconds (success: False)
2024-07-29 16:26:55.129 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-07-29 16:26:55.129 INFO (MainThread) [custom_components.solarman.api] Disconnecting from 192.168.10.52:8899
2024-07-29 16:26:55.129 DEBUG (MainThread) [custom_components.solarman.api] Connection lost can be during closing ignored.

It seems this one is causing the issue "5FrameError: V5 frame does not contain a valid Modbus RTU frame"

Any ideas to get rid of it?

davidrapan, 2024-07-29:
Hi @orangedmind,

It looks like it's going to be something similar to the #43 :-/

davidrapan, 2024-07-29:
@orangedmind,

Look, try to install beta (pre-release) of this integration.

You can do it using its HACS page (by clicking here on that big blue button) then those 3 dots in top right corner and click on Redownload there will be a switch which says "Show beta versions" then procced to select v24.07.29 wait a sec and at the end click DOWNLOAD. Restart.

orangedmind, 2024-07-30:
@davidrapan 

I installed the mentioned beta, restarted HA and checked again.
Same behavior, same message

2024-07-30 10:50:14.581 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'data': {'name': 'Solarman', 'inverter_discovery': True, 'inverter_host': '192.168.10.52', 'inverter_serial': 4176382624, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J3ZE3X2CHZRAHA6FZMGDVZ2C', 'minor_version': 1, 'options': {'name': 'Solarman', 'inverter_discovery': True, 'inverter_host': '192.168.10.52', 'inverter_serial': 4176382624, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Solarman', 'unique_id': None, 'version': 1})
2024-07-30 10:50:14.582 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-07-30 10:50:14.591 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.10.52, E8FDF87C2F24, 4176382624]
2024-07-30 10:50:15.732 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', 'e8:fd:f8:7c:2f:24')}, 'identifiers': {('solarman', 4176382624)}, 'name': 'Solarman', 'manufacturer': 'Deye', 'model': 'HYBRID', 'serial_number': 4176382624}
2024-07-30 10:50:15.732 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-07-30 10:50:15.732 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 3, min_span: 25, digits: 6
2024-07-30 10:50:15.733 DEBUG (MainThread) [custom_components.solarman.api] Scheduling 4 query requests. #0
2024-07-30 10:50:15.733 DEBUG (MainThread) [custom_components.solarman.api] Querying (3 - 14) ...
2024-07-30 10:50:15.733 INFO (MainThread) [custom_components.solarman.api] Connecting to 192.168.10.52:8899
2024-07-30 10:50:15.756 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 54 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 69 15
2024-07-30 10:50:15.852 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 2b 00 10 15 54 60 a0 8a ee f8 02 01 6b 69 36 00 8c 18 00 00 db 3f 72 66 01 03 18 32 32 31 30 32 39 30 32 35 33 00 01 00 00 12 0c 07 00 00 00 01 14 12 12 50 f4 70 15
2024-07-30 10:50:15.854 DEBUG (MainThread) [custom_components.solarman.api] Querying succeeded.
2024-07-30 10:50:15.854 DEBUG (MainThread) [custom_components.solarman.api] Querying (59 - 112) ...
2024-07-30 10:50:15.854 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-30 10:50:15.879 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-30 10:50:15.879 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 55 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 3b 00 36 b4 11 0d 15
2024-07-30 10:50:16.062 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 7f 00 10 15 55 61 a0 8a ee f8 02 01 6b 69 36 00 8c 18 00 00 db 3f 72 66 01 03 6c 00 04 00 01 00 00 00 00 1e 91 00 00 00 00 00 00 00 00 00 00 0e d0 00 00 0e b2 00 00 09 38 00 00 00 00 00 05 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 04 e2 00 00 00 00 00 00 10 86 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 29 00 14 01 20 00 16 bf dd 3d 15
2024-07-30 10:50:16.063 DEBUG (MainThread) [custom_components.solarman.api] Querying succeeded.
2024-07-30 10:50:16.064 DEBUG (MainThread) [custom_components.solarman.api] Querying (150 - 195) ...
2024-07-30 10:50:16.064 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-30 10:50:16.085 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-30 10:50:16.085 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 56 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 96 00 2e 25 fa bb 15
2024-07-30 10:50:16.243 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] RECD: a5 6f 00 10 15 56 62 a0 8a ee f8 02 01 6b 69 36 00 8c 18 00 00 db 3f 72 66 01 03 5c 0a be 07 08 14 50 12 8e f7 2c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 48 67 15
2024-07-30 10:50:16.244 DEBUG (MainThread) [custom_components.solarman.api] Querying succeeded.
2024-07-30 10:50:16.244 DEBUG (MainThread) [custom_components.solarman.api] Querying (244 - 279) ...
2024-07-30 10:50:16.244 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-07-30 10:50:16.268 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] Successful reconnect
2024-07-30 10:50:16.268 DEBUG (MainThread) [custom_components.solarman.api] [4176382624] SENT: a5 17 00 10 45 57 00 a0 8a ee f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 f4 00 24 04 23 18 15
[... cut ...]
2024-07-30 10:50:31.518 INFO (MainThread) [custom_components.solarman.api] Disconnecting from 192.168.10.52:8899
2024-07-30 10:50:31.519 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Solarman data in 15.787 seconds (success: False)
2024-07-30 10:50:31.520 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-07-30 10:50:31.520 INFO (MainThread) [custom_components.solarman.api] Disconnecting from 192.168.10.52:8899
2024-07-30 10:50:31.520 DEBUG (MainThread) [custom_components.solarman.api] Connection lost can be during closing ignored.

davidrapan, 2024-07-30:
You sure you are running 24.07.29 version?

orangedmind, 2024-07-30:
Sorry, you are right, there was a pending update I haven't seen.
[image]

Now I get the entities from solarman. Thank you
I think I have to figure out which data matches the information I see in the Solarman App.
[image]
[image]
[image]
[image]

davidrapan, 2024-07-30:
What exact device model it is?

orangedmind, 2024-07-31:
Do you mean the inverter?
Type: SUN600G3-EU-230

davidrapan, 2024-07-31:
You should be using this profile file: 'deye_2mppt.yaml'

davidrapan, 2024-07-31:
Device used with incorrect inverter definition profile.

orangedmind, 2024-07-31:
Thanks a lot, now I get values.
[image]