davidrapan/ha-solarman #47: HACS Update for Stephen Jouberts Solarman still remains after updating to Solarman by David Rapan.

PV-Joe, 2024-07-29:
Hi David,

after updating from Solarman by Stephen Joubert to HA-Solarman by David Rapan the HACS Update from Stephen still remains.

[image]

Have you got a clue how to get rid of it?

davidrapan, 2024-07-29:
Yeah, you have to just REMOVE that one trough its HACS page and then REDOWNLOAD this one.

PV-Joe, 2024-07-29:
I’m getting a heat flush already thinking about messing up something again.

Will try to though after the sun goes down and a fresh backup, just in case.

Thanks for your fast reply. ☀️

davidrapan, 2024-07-29:
Now I remembered that you could maybe have problems cause your sensor renames :-/ Cause it will proly asks you to remove the device first :-/

I'll look of there is any manual way of doing that...

PV-Joe, 2024-07-29:
Sensor names are the ones from your integration. I just renamed the entityID’s for your sensor name to the ones before updating. That kept all my history values instant.

davidrapan, 2024-07-29:
Yes I know I remember but it will get lost (those renames not history) if you delete your inverter device from HA. :) So you have to do it all over again after re-adding it back.

davidrapan, 2024-07-29:
Okay so inside '/homeassistant/.storage/core.device_registry' when you will search for 'stephanjoubert' you should find an entry which will look something like this:
```
{"area_id":null,"config_entries":["68a258ff31941ee930c340dbdd289960"],"configuration_url":"homeassistant://hacs","connections":[],"disabled_by":null,"entry_type":"service","hw_version":null,"id":"83f522cfacafdf9639c6f53b018ce54e","identifiers":[["hacs","822730719"]],"labels":[],"manufacturer":"davidrapan","model":"integration","name_by_user":null,"name":"Solarman","primary_config_entry":"68a258ff31941ee930c340dbdd289960","serial_number":null,"sw_version":null,"via_device_id":null}
```
but instead there will be "manufacturer":"stephanjoubert" instead, if you delete that whole row from there that should do it. Just be careful about formatting.

Just to be sure before you edit anything show me the line you found in ther together with one line above and one line under whatever its there, okay? And I tell you what next.

PV-Joe, 2024-07-30:
Hey David,

thanks again for trying to help me.

I looked into the file and searched for “stephen” and tried “david” also with no results.

I looked at every single entry. The only possible entry I’ve found that has a hint to one of the two integrations is:

"area_id":"solar","config_entries":["63dd0cfed036f8b2010cd683f6665b3a","75bccd9a330d6078f2c53646c0256bf9"],"configuration_url":null,"connections":[["mac","e8:fd:f8:3b:67:18"]],"disabled_by":null,"entry_type":null,"hw_version":null,"id":"bd522119c74fcfa0f79001c0d28b9bb6","identifiers":[["solarman",xxxxxxxxxx]],"labels":[],"manufacturer":"Deye","model":"SG04LP3","name_by_user":null,"name":"Solarman","primary_config_entry":"63dd0cfed036f8b2010cd683f6665b3a","serial_number":xxxxxxxxxx,"sw_version":null,"via_device_id":"385077ec822ae0e786a3fe98b874f124"},

I did not delete it.

davidrapan, 2024-07-30:
No do not delete this one, this one is about your installed device and it would disappear!

Try look for "solarman" then? if there is more entries?

PV-Joe, 2024-07-30:
No more entries for Solarman.

davidrapan, 2024-07-30:
Bro that's so weird! What version of HA are you running? There should be somewhere entries for all of your HACS integrations you are using and such...

PV-Joe, 2024-07-30:
I am on:
[image]

Lots of “hacs” entries.

Your integration must use the entry above then.

davidrapan, 2024-07-30:
Entry you showed is for the device within my integration. You need an entry for the integration within hacs.

Guess you gonna have to show me those entries then... there should be one which belongs to the Stephans integration within hacs.