davidrapan/ha-solarman #49: Warning Release 24.07.29 causes issues

gedger, 2024-07-30:
Just tried the above and it raises a load of errors. Downgrading to .28 fixes.

[home-assistant_solarman_2024-07-30T08-21-37.172Z.log](https://github.com/user-attachments/files/16424576/home-assistant_solarman_2024-07-30T08-21-37.172Z.log)

davidrapan, 2024-07-30:
As it's unfortunate this pre-release version is mainly for testing some niche cases of some users which are not able to run at all it proly will change multiple times before the release. But still thanks for the report for sure! I'm also running it but without any problems, maybe it's device dependent hm?

gedger, 2024-07-31:
For info I've just tried the latest from main and it still causes issues so will need a tweak before releasing. Log attached which will hopefully help.

[home-assistant_solarman_2024-07-31T14-46-38.733Z.log](https://github.com/user-attachments/files/16443526/home-assistant_solarman_2024-07-31T14-46-38.733Z.log)

davidrapan, 2024-07-31:
Not sure what exactly could cause that but I tried: [fix: Reverted partial request fulfillment](https://github.com/davidrapan/ha-solarman/commit/045986ba4f588716dd0c6fb9ba854b09f4bf65b8) as it turned out this feature is not necessary and original issue was caused by incorrect profile used.

gedger, 2024-08-01:
Sorry, only just seen your Edit2: above. I have been seeing some really variable update intervals which I think has been also caused by the auto discovery. I have now disabled auto discovery which has improved matters plus have loaded 24.07.31 with no issues so far.

Going off-line

[image]

Coming on-line

[image]

I'm still struggling to debug what happened at 8.05am when I restarted HA with a slightly different profile that had 3 disabled parameters. Now I've disabled auto discovery and running the latest build I will evaluate again over the coming days.

davidrapan, 2024-08-02:
Could you maybe do a test for me with 5 sec update interval for lets say a hour during the day? And also share the graph here?

gedger, 2024-08-03:
Just realised, would you like a debug log too? If so I'll run again.

[image]

davidrapan, 2024-08-03:
Thanks! And no it is not necessary I just wanted to see what are the usual update intervals on other installations.

Very appreciated!

davidrapan, 2024-08-03:
What about the issues you were seeing? Are they still present in the last pre-release?

gedger, 2024-08-04:
Latest release has been fine, I was just giving it a few days to confirm. I  couldn't determine the cause of the variable polling time when it varies between 30-40s as shown on the "coming on line" graph above.  Integration, inverter or network? If I ever find a way of provoking it then I'll raise an issue (if it is the integration) so I think it's safe to close this.

closed 2024-08-04 by gedger