davidrapan/ha-solarman #51: Discovery process too frequent

gedger, 2024-08-01:
If the inverter cannot be reached the integration seems to go into a discovery loop of trying to find / connect to the inverter. This is too aggressive and causes serious impact to the running of Home Assistant as the GUI becomes slow and unresponsive. The inverter also fails to come back on line as it is being bombarded with packets. Running a ping from another host the inverter does not respond but as soon as the integration is stopped then immediately starts responding to the ping and comes back on-line.

[home-assistant_solarman_using_non_custom_config.log](https://github.com/user-attachments/files/16452285/home-assistant_solarman_using_non_custom_config.log)

I triggered this by trying to get to the route of another issue of variable update rates. The following graph shows the update interval.
[image]

davidrapan, 2024-08-01:
Discover happens only during the startup and at first directly to the saved IP and then when nothing responds on all interfaces. It really only depends on what Spook does to the integration.

gedger, 2024-08-01:
Spook was not in the loop in this instance, plus I've looked through the code and it only calls the official API to disable polling in exactly the same as if you had done it from the HA GUI so shouldn't cause an issue......

I've now hopefully stopped HA using those interfaces, they are from my VPN and other docker containers so it has no need to scan those.

I still think there should be a significantly longer delay between the auto discovery, surely 20-30 seconds would be quick enough?

davidrapan, 2024-08-01:
OMG that's a private range too. Never heard of that one lol.

davidrapan, 2024-08-01:
I mean those are few packets anyway. The problem was the exception not the "size" of the discovery.

20-30 seconds is for HA eternity and will complain that whole initialization did not fit in 10 seconds range. 😆

gedger, 2024-08-01:
I don't understand how/why it was slowing down HA so much and why my inverter stopped responding to pings but as soon as I stopped the integration then everything returned to normal, crazy. I've disabled discovery now so we'll see how it goes.

davidrapan, 2024-08-01:
That discovery concise of UDP broadcast packets. That in any way can cripple the inverter...

davidrapan, 2024-08-01:
Also, you said that Spook was not involved. But that discovery can kick in only when you restart the integration. As I said the discovery is only present at the start of integration.

gedger, 2024-08-01:
Presumably that includes when you load a different inverter profile?

davidrapan, 2024-08-01:
Changing the configuration does reload the integration.

gedger, 2024-08-04:
Haven't had an issue since disabling discovery on my setup so we can close this.

closed 2024-08-04 by gedger

davidrapan, 2024-08-04:
I also adjusted the discovery flow to do it only once with 5 attempts and then one confirmation for the MAC if it succeeds.