davidrapan/ha-solarman #52: [BUG in Custom Sensor]: Correction for "Today Consumption [kWh]" working for Deye Hybrid (deye_sg04lp3.yaml) - [Fault: Operator missing for substraction]

PV-Joe, 2024-08-03:
Hi David,

I added your Custom Sensor for "Today's Consumption [kWh]" from your Wiki and realised that the operators for substraction were missing. The result was a much to high value as the values weren't substracted.

I tested the following YAML Code for the Custom Sensor (deye_sg04lp3.yaml) and it is working fine:

```
  - group: Custom Sensors
    items:
      # Calculates today's energy consumption
      # Formula: Today Energy Bought [0x0208] + Today Production [0x0211] + Today Battery Discharge [0x0203] - Today Energy Sold [0x0209] - Today Battery Charge [0x0202]
      - name: Today Consumption
        friendly_name: Today's Consumption
        alt: Daily Consumption
        description: Includes today's house consumption as well inverter consumption and AC/DC conversion losses
        class: energy
        state_class: total_increasing
        uom: kWh
        rule: 1
        digits: 1
        registers: [0x0208, 0x0211, 0x0203, 0x0209, 0x0202]
        sensors:
          - scale: 0.1
            registers: [0x0208]
          - scale: 0.1
            registers: [0x0211]
          - scale: 0.1
            registers: [0x0203]
          - operator: subtract
            scale: 0.1
            registers: [0x0209]
          - operator: subtract
            scale: 0.1
            registers: [0x0202]   
``` 
**Add this sensor for Deye Hybrid SG04LP3:**

1. Copy code to your "deye_sg04lp3.yaml" or "custom_configurations.yaml"
2. Restart Home Assistant
3. Add newly created sensor "sensor.solarman_today_consumption" to your dashboard

**Description:**
This Sensor will add a calculated value from today's house consumption as well inverter consumption and AC/DC conversion losses.

davidrapan, 2024-08-03:
They are in the wiki according to the latest (pre-release) version of the integration. You can either look at other "custom sensors" in the profile and adjust the sensors accordingly or update to the latest version.

PV-Joe, 2024-08-03:
I’m on Version v24.07.24.
So this belongs to the latest official release.

I never tested any of the prereleases.
Added the other sensor for Total Consumption as well. 👍

davidrapan, 2024-08-05:
Should be all good now in 24.08.04, hm?

PV-Joe, 2024-08-05:
Did not injekt 24.08.04 yet.
I’m using a custom_configuration.yaml at the moment with operator and it works well with the last official release.

Probably will injekt the latest release soon to keep up. 😄

davidrapan, 2024-08-05:
Don't forget that it has now dedicated place for your custom profile! So back it up before the update through HACS.

closed 2024-08-09 by davidrapan

PV-Joe, 2024-08-10:
Here is the sensor for Total Consumption [kWh]:

```
      # Calculates total energy consumption
      - name: Total Consumption
        friendly name: Total Consumption
        alt: Total Consumption
        description: Includes total house consumption as well inverter consumption and AC/DC conversion losses
        class: energy
        state_class: total_increasing
        uom: kWh
        rule: 3
        digits: 1
        registers:
          [
            0x020A,
            0x020B,
            0x0216,
            0x0217,
            0x0206,
            0x0207,
            0x020C,
            0x020D,
            0x0204,
            0x0205,
          ]
        sensors:
          - scale: 0.1
            registers: [0x020A, 0x020B]
          - scale: 0.1
            registers: [0x0216, 0x0217]
          - scale: 0.1
            registers: [0x0206, 0x0207]
          - operator: subtract
            scale: 0.1
            registers: [0x020C, 0x020D]
          - operator: subtract
            scale: 0.1
            registers: [0x0204, 0x0205]
```