davidrapan/ha-solarman #56: [Add Custom Sensor] Today‘s PV Consumption [kWh]

PV-Joe, 2024-08-05:
Hi David,

is it possible to add a permanent custom sensor in the deye_sgo4lp3.yaml definition file to fulfill the following task?

**Today‘s PV Consumption [kWh]**

1. This sensor should calculate the consumption taken from the sun only.
2. The sensor should not include: Battery Charge/Discharge or Energy Export.

This sensor would be great to visualize the consumed Energy with the energy source.

Any idea how to implement this?

PV-Joe, 2024-08-05:
This formula should leave just the consumed sun for today:

**Today's Production - Today's Energy Export - Today's Battery Charge - Today’s Losses**

Am I right or did I miss out?

The result is probably negative in the morning before production starts because of the substraction of “Today’s Losses”.

Maybe you can find a more elegant way to split the “Today’s Consumption into their sources?

I’d like to split the (orange) Today’s Consumption into:

1. Today’s PV Consumption
2. Today’s Grid Import (blue)
3. Today’s Battery Import
4. Today’s Losses (red)

[image]

PV-Joe, 2024-08-05:
> _Also for this topic are more suitable discussions cause it's not really an issue but whatever, hit me._ 😆

On your Code page you have written:
leave [🚩 Issues](https://github.com/davidrapan/ha-solarman/issues) for 🐞 bug reporting, 🎁 feature requests and such...

… so I thought this was a feature request an changed the title 😄
I can move it if you want or maybe it is better that you explain to just use the section for BUGS / ISSUES only?

davidrapan, 2024-08-05:
No you somehow right. I can be seen as feature request... 😆 

I'll look into it. Even I'm not really convinced of the value of such sensor. 😝

davidrapan, 2024-08-05:
I'm gonna soon end up with the PTSD from all that logic in the naming though... 😆

PV-Joe, 2024-08-05:
> PTSD

Don’t worry too much about the naming. I got used to it and like it how it is for our DEYE.
I think it hasn’t got a very high priority in my opinion.

New features please 😄

Your integration works like a charm by the way. That’s what I call fast and reliable. 👍
You are a magician!

Keep up the brilliant work but watch out for PTSD 😂

davidrapan, 2024-08-05:
But shouldn't be battery charging counted towards self-consumed?

PV-Joe, 2024-08-05:
The Battery charging is already separated with the “ Solarman Today's Battery Charge” as a source for energy.

For the Dashboard it would be nice to have separated energy sources to make it more clear where the consumed energy comes from.

davidrapan, 2024-08-05:
Battery Discharge is source of the energy. Battery Charge is consumer of the energy.

davidrapan, 2024-08-05:
First experiment:
[image]

PV-Joe, 2024-08-05:
Did you calculate without any battery charge/discharge?

Confusing isn’t it but there must be a way somehow.

davidrapan, 2024-08-05:
It's just PV without exported energy and losses. I showed you all my today's values so you can check whenever it matches your expectations or not. 😉

davidrapan, 2024-08-05:
Which formula? F1? 😆

I just test it using template sensor within HA for now.

CrazyUs3r, 2024-08-05:
You can see if you can use this

from __future__ import annotations

import logging
import voluptuous as vol

from homeassistant.config_entries import ConfigEntry
from homeassistant.core import HomeAssistant
from homeassistant.const import EntityCategory
from homeassistant.components.select import SelectEntity
from homeassistant.helpers.entity_platform import AddEntitiesCallback

_LOGGER = logging.getLogger(__name__)

_PLATFORM = "calculator"

def _create_calculator_entity():
    try:
        entity = CalculatorEntity()
        entity.update()
        return entity
    except BaseException as e:
        _LOGGER.error(f"Configuring calculator failed. [{e}]")
        raise

async def async_setup_entry(hass: HomeAssistant, config: ConfigEntry, async_add_entities: AddEntitiesCallback) -> bool:
    _LOGGER.debug(f"async_setup_entry: {config.options}")
    
    async_add_entities([_create_calculator_entity()])

    return True

async def async_unload_entry(hass: HomeAssistant, config: ConfigEntry) -> bool:
    _LOGGER.debug(f"async_unload_entry: {config.options}")
    return True

class CalculatorEntity(SelectEntity):
    def __init__(self):
        self._result = 0
        self._attr_entity_category = EntityCategory.CONFIG
        self.entity_id = f"{_PLATFORM}.calculator"
        self._attr_options = ["add", "subtract", "multiply", "divide", "reset"]

    def update(self):
        _LOGGER.debug(f"CalculatorEntity initialized with result {self._result}")

    @property
    def current_option(self):
        """Return the current option of this select."""
        return None

    async def async_select_option(self, option: str) -> None:
        """Perform the selected operation."""
        if option == "reset":
            self._result = 0
        else:
            # This example uses fixed values for demonstration. You can modify it to accept dynamic input.
            value = 10
            if option == "add":
                self._result += value
            elif option == "subtract":
                self._result -= value
            elif option == "multiply":
                self._result *= value
            elif option == "divide":
                if value != 0:
                    self._result /= value
                else:
                    _LOGGER.error("Division by zero is not allowed")
        self._attr_state = self._result
        self.async_write_ha_state()
        _LOGGER.debug(f"Performed {option} operation. Result is now {self._result}")

if __name__ == "__main__":
    logging.basicConfig(level=logging.DEBUG)
    entity = CalculatorEntity()
    entity.update()
    entity.async_select_option("add")
    entity.async_select_option("subtract")
    entity.async_select_option("multiply")
    entity.async_select_option("divide")
    entity.async_select_option("reset")

PV-Joe, 2024-08-11:
> First experiment: ![self-consumed_experiments_1](https://private-user-images.githubusercontent.com/71029912/355148845-9bbd965f-d174-46d5-a810-2d8fb9749cd7.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMzNjg1NzUsIm5iZiI6MTcyMzM2ODI3NSwicGF0aCI6Ii83MTAyOTkxMi8zNTUxNDg4NDUtOWJiZDk2NWYtZDE3NC00NmQ1LWE4MTAtMmQ4ZmI5NzQ5Y2Q3LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODExVDA5MjQzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTExZDdiOTExYWEyYmEyOTVmY2ZmM2U3MjhlMTdiYzNjOGRjZjgwNGRjODMwMmRkYzliNWNmMzUwOWNlMDNkY2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6OUxXhvSEVfmddgrGkenEy1DL4H5_oKCIjSzYwPtG_g)
> 
> But it does not match the Energy Dashboard. Proly cause they don't care about losses.

I think the value **Today’s PV Self-Consumed** does the trick and would be a nice addition to your next final release 👍

closed 2024-08-22 by PV-Joe