davidrapan/ha-solarman #62: Cannot connect to inverter using domain

hechay, 2024-08-05:
Logger: homeassistant.config_entries
Source: config_entries.py:586
First occurred: 5:42:07 AM (1 occurrences)
Last logged: 5:42:07 AM

Error setting up entry SolarBK-DEYE-Domain for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 39, in async_setup_entry
    if IPv4Address(inverter_host).is_private:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ipaddress.py", line 1319, in __init__
    self._ip = self._ip_int_from_string(addr_str)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ipaddress.py", line 1206, in _ip_int_from_string
    raise AddressValueError("Expected 4 octets in %r" % ip_str)
ipaddress.AddressValueError: Expected 4 octets in 'mydomain.net'

I hope you can develope the new function that connect to inverter via domain name. 

Thanks a lot!

davidrapan, 2024-08-06:
Hi @hechay, [fix: Could not use with domain name instead of IP](https://github.com/davidrapan/ha-solarman/commit/a507d21c4916dfe8149f63b8e9d88a087ac1a200)

hechay, 2024-08-06:
Great. I'll test and let your know immediately

hechay, 2024-08-06:
Work great!

closed 2024-08-06 by hechay