davidrapan/ha-solarman #64: Connecting to INVT iMars BG20KTR-US inverter with LSW-3

Jochey, 2024-08-07:
Hello,

I am trying to connect to my solar inverter INVT iMars BG20KTR-US with an LSW-3 wifi stick connected. Firmware LSW3_14_8401_1.15

I am able to connect, but get the error "The data address received in the request is not an allowable address for
the server."

Any advice on how to solve this issue? I have tested several of the different invert configs.

2024-08-07 19:58:21.629 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'data': {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.30.111', 'inverter_serial': 1724679032, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J4PYZZDM4RR5XFGEE23P04N0', 'minor_version': 1, 'options': {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.30.111', 'inverter_serial': 1724679032, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Inverter', 'unique_id': None, 'version': 1})
2024-08-07 19:58:21.629 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-08-07 19:58:21.636 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.30.111, 98D863D0A7B4, 1724679032]
2024-08-07 19:58:22.922 DEBUG (MainThread) [custom_components.solarman.api] {'connections': {('mac', '98:d8:63:d0:a7:b4')}, 'identifiers': {('solarman', 1724679032)}, 'name': 'Inverter', 'manufacturer': 'Deye', 'model': 'HYBRID', 'serial_number': 1724679032}
2024-08-07 19:58:22.922 DEBUG (MainThread) [custom_components.solarman] async_setup: coordinator.async_config_entry_first_refresh
2024-08-07 19:58:22.922 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 10, code: 3, min_span: 25, digits: 6
2024-08-07 19:58:22.923 DEBUG (MainThread) [custom_components.solarman.api] Scheduling 4 query requests. #0
2024-08-07 19:58:22.923 DEBUG (MainThread) [custom_components.solarman.api] Querying (3 - 14) ...
2024-08-07 19:58:22.924 INFO (MainThread) [custom_components.solarman.api] Connecting to 192.168.30.111:8899
2024-08-07 19:58:22.935 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] SENT: a5 17 00 10 45 04 00 78 83 cc 66 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 36 15
2024-08-07 19:58:22.980 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] RECD: a5 13 00 10 15 04 70 78 83 cc 66 02 01 56 0f f3 01 ba 0e 00 00 67 a6 c0 64 01 83 02 c0 f1 65 15
2024-08-07 19:58:22.981 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [IllegalDataAddressError: The data address received in the request is not an allowable address for
the server.
]
2024-08-07 19:58:23.182 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 4
2024-08-07 19:58:23.182 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-08-07 19:58:23.188 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] Successful reconnect
2024-08-07 19:58:23.189 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] SENT: a5 17 00 10 45 05 00 78 83 cc 66 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 37 15
2024-08-07 19:58:23.241 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] RECD: a5 13 00 10 15 05 71 78 83 cc 66 02 01 56 0f f3 01 ba 0e 00 00 67 a6 c0 64 01 83 02 c0 f1 67 15
2024-08-07 19:58:23.242 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [IllegalDataAddressError: The data address received in the request is not an allowable address for
the server.
]
2024-08-07 19:58:23.643 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 3
2024-08-07 19:58:23.644 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-08-07 19:58:23.656 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] Successful reconnect
2024-08-07 19:58:23.657 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] SENT: a5 17 00 10 45 06 00 78 83 cc 66 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 38 15
2024-08-07 19:58:23.712 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] RECD: a5 13 00 10 15 06 72 78 83 cc 66 02 01 56 0f f3 01 ba 0e 00 00 67 a6 c0 64 01 83 02 c0 f1 69 15
2024-08-07 19:58:23.713 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [IllegalDataAddressError: The data address received in the request is not an allowable address for
the server.
]
2024-08-07 19:58:24.315 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 2
2024-08-07 19:58:24.315 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-08-07 19:58:24.322 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] Successful reconnect
2024-08-07 19:58:24.323 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] SENT: a5 17 00 10 45 07 00 78 83 cc 66 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 39 15
2024-08-07 19:58:24.380 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] RECD: a5 13 00 10 15 07 73 78 83 cc 66 02 01 57 0f f3 01 bb 0e 00 00 67 a6 c0 64 01 83 02 c0 f1 6d 15
2024-08-07 19:58:24.381 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [IllegalDataAddressError: The data address received in the request is not an allowable address for
the server.
]
2024-08-07 19:58:25.183 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 1
2024-08-07 19:58:25.183 DEBUG (MainThread) [custom_components.solarman.api] Reader Task done: False, cancelled: False.
2024-08-07 19:58:25.190 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] Successful reconnect
2024-08-07 19:58:25.191 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] SENT: a5 17 00 10 45 08 00 78 83 cc 66 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 0c b5 cf 3a 15
2024-08-07 19:58:25.251 DEBUG (MainThread) [custom_components.solarman.api] [1724679032] RECD: a5 13 00 10 15 08 74 78 83 cc 66 02 01 58 0f f3 01 bc 0e 00 00 67 a6 c0 64 01 83 02 c0 f1 71 15
2024-08-07 19:58:25.252 WARNING (MainThread) [custom_components.solarman.api] Querying (3 - 14) failed. #0 [IllegalDataAddressError: The data address received in the request is not an allowable address for
the server.
]
2024-08-07 19:58:26.253 DEBUG (MainThread) [custom_components.solarman.api] Querying attempts left: 0, aborting.
2024-08-07 19:58:26.253 DEBUG (MainThread) [custom_components.solarman.api] Request failed. [Previous Status: Disconnected]
2024-08-07 19:58:26.253 INFO (MainThread) [custom_components.solarman.api] Disconnecting from 192.168.30.111:8899
2024-08-07 19:58:26.256 DEBUG (MainThread) [custom_components.solarman.coordinator] Finished fetching Inverter data in 3.334 seconds (success: False)
2024-08-07 19:58:26.257 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-08-07 19:58:26.257 INFO (MainThread) [custom_components.solarman.api] Disconnecting from 192.168.30.111:8899

davidrapan, 2024-08-07:
Hi @Jochey, that means you are requesting wrong registers from the device so you need to use different definition profile. You gonna have to either try some other available and see if any will match your device or write your own.

Jochey, 2024-08-07:
> Hi @Jochey, that means you are requesting wrong registers from the device so you need to use different definition profile. You gonna have to either try some other available and see if any will match your device or write your own.

Thank you @davidrapan. I have tried all the definition profiles but none of them is working straight out of the box. For the afore_hybrid I get the message 2024-08-07 20:50:14.370 WARNING (MainThread) [custom_components.solarman.api] Querying (507 - 560) failed. #0 [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]. I guess this might be a good place to start?

Any hints on how to identify the changes I need to make for a new profile?

davidrapan, 2024-08-07:
I was trying to find something but no luck yet :-/ Crucial will be to find similar device from different brand as many of them share hw inside and just branding is changing and it looks like that Afore could be good candidate.

Also on the github page of library which is used here: https://github.com/jmccrohan/pysolarmanv5 you can find out how to manually scan registers but it is very tedious and long job... :-/

closed 2024-09-30 by github-actions[bot]