davidrapan/ha-solarman #65: Total Production is dropping to zero

switschel, 2024-08-08:
Since release v24.08.04 my microinverter total production value is shortly dropping to "0" and afterwards to the correct values. This leads to wrong statistics in the energy dashboard saying that the total production happend every day.

Would be great if the drop to zero on microinverter start can be avoided each day morning.

davidrapan, 2024-08-08:
Ah, damn...

Would you be willing to test which beta version between versions 24.07.24 - 24.08.04 introduced this mess?

davidrapan, 2024-08-09:
Yeah I guess you gonna have to do it manually then.

Even the latest commits as they are not part of any release yet.

davidrapan, 2024-08-09:
Try 24.08.08

switschel, 2024-08-10:
> Try 24.08.08 pre-release.

Tried it, same issue. Dropping to zero every day the microinverter starts.

davidrapan, 2024-08-10:
What device are you using?

switschel, 2024-08-10:
Deye sun600g3-eu with deye 2mppt profile

davidrapan, 2024-08-10:
Your issue is proly cause by this: [fix: Deye microinverters validation lowered to 0.0](https://github.com/davidrapan/ha-solarman/commit/2a10012e5516655f3066aa1eb4e1fbdaa6625fa5)

it now somehow makes sense why it was there... But it caused issues with new inverters...

davidrapan, 2024-08-10:
Try it with definition profile from this commit: [fix: Deye microinverters PV validation to 0.1](https://github.com/davidrapan/ha-solarman/commit/6ea759340755d80be6f96aadeaf39c509a3236e6)

davidrapan, 2024-08-12:
Confirmed as fixed in: https://github.com/davidrapan/ha-solarman/issues/76

closed 2024-08-12 by davidrapan