davidrapan/ha-solarman #68: modbus write register

amiko24, 2024-08-10:
Hello David, how can I write the registers.

I use such an automation to write the wallbox registers via Modbus. (See yaml code below).
Does it work the same way with the Deye Iverters?
What is the name of the hub for Deye sg04lp3? Slave number is 1 or 0?

`- id: '1680707528756'
  alias: Setze Ladestrom
  description: ''
  trigger:
  - platform: state
    entity_id:
    - sensor.output_strom_wallbox
  condition: []
  action:
  - service: modbus.write_register
    data:
      hub: modbus heidelberg wallbox
      address: 261
      slave: 1
      value:
      - '{{ (states.sensor.output_strom_wallbox.state) }}'`

PV-Joe, 2024-08-10:
I don’t know about the write registers but:

Slave number is 1

… for my Deye SG04LP3

It works fine!

gedger, 2024-08-10:
When I look at the services provided by solarman I only see.
[image]

gedger, 2024-08-10:
So this is a working call to change a holding register.
[image]

amiko24, 2024-08-10:
it should also work via device

`      - action: solarman.write_holding_register
        metadata: {}
        data:
          wait_for_attempts: 5
          device: 70ddcba05b16bf94857f18e9836eb017
          register: 80
          value: 1`

amiko24, 2024-08-10:
it should also work via device

`      - action: solarman.write_holding_register
        metadata: {}
        data:
          wait_for_attempts: 5
          device: 70ddcba05b16bf94857f18e9836eb017
          register: 80
          value: 1
  mode: single`

gedger, 2024-08-10:
It only works with device, the GUI translates the friendly device name to device id for you. if you look at the service call I showed above then the yaml is .

```
      - service: solarman.write_holding_register
        data:
          wait_for_attempts: 5
          device: db2d333f97371cec16797b695551cef5
          register: 3051
          value: 10000
```