davidrapan/ha-solarman #70: HYD 6000-EP sofar + SOLARMAN + HA = imposible to connect

miralpeix, 2024-08-10:
Dear friends

After a week trying different things to connect HA with my solarman inverter… I need some help.

Current setup:

Inverter: HYD 6000-EP sofar

[image]

As first option I have tried to use the SOLARMAN INTEGRATION, however all devices are show as “unknown”.

[image]

Following another recomendations I have tried in oder to solve:

1. Different versions of solarman integrations (1.5… 1.4…)
2. Different yalm configurations (sofar_lsw3.yalm, deye_hybrid.yalm….)
3. Due to possible problems in HA, I have reinstalled in different raspberrys…
4. I have spoken with sofar to update the logger FW to the latest version available
5. Also I have trie another alternative solutions like MQQT + SUN SINK: https://www.youtube.com/watch?v=BJzJSi6TZMk but didn’t work too.

Now thanks a one good recommendation im trying to use the https://github.com/davidrapan/ha-solarman

But the problem is here again, here the screenshots and attach the log

[image]
[image]

I feel like it is impossible to connect with my logger from any way using HA.
[5.log](https://github.com/user-attachments/files/16569634/5.log)

Some recommendations?? I’m about to surrender after this week…

davidrapan, 2024-08-10:
Hi @miralpeix,

In the definition profile you are using try change the function code in:
```
- start: 0x0200
  end: 0x0255
  mb_functioncode: 0x03
```
to 0x04

or even remove that request set completely so it runs only with the second one

PV-Joe, 2024-08-10:
I found a Sofar Solar Modbus documentation.
Haven’t checked if these are the registers from your integration.

Check it out here:
https://solaranzeige.de/phpBB3/download/file.php?id=5254

miralpeix, 2024-08-10:
> Hi @miralpeix,
> 
> In the definition profile you are using try change the function code in:
> 
> ```
> - start: 0x0200
>   end: 0x0255
>   mb_functioncode: 0x03
> ```
> 
> to 0x04
> 
> or even remove that request set completely so it runs only with the second one.
> 
> _And you don't have a problem with connection to the inverter, you have a problem with reading registers from the inverter._ 😉
> 
> And send logs.

David, Joe, the knowledge you have seems incredible to me.... and I feel like an idiot asking this, but, how can I change this code? Can you give a little advise in order to find it in home assistant? THANKS 🙏

miralpeix, 2024-08-10:
Find it with file editor :)

I have change it, but the problem continues... :-(

First change to 0x04 and then removing all the request.

Attach the log and shots

You can't imagine how grateful I am for your help.  [8.log](https://github.com/user-attachments/files/16569864/8.log)

[image]

[image]

PV-Joe, 2024-08-10:
> David, Joe, the knowledge you have seems incredible to me....

David is the expert, not me.
He helped me to get mine going as well and hopefully can solve your problem as well.

This Integration has multiple extras and advantages compared to Stephen’s version. By the way.

Hope it works out for you!

davidrapan, 2024-08-10:
The 8.log is after you changed that code or after you removed the first request set? Would love to see both logs, aswell another try where you run it only with the second request but 0x03 code.

I'm proly gonna have to pass you problem to the guys which made the lib we are using in the integration.

davidrapan, 2024-08-10:
Another thing I want you to try is to put there single request for 0x0200 to see if it is able to at least read single register.

Adjust the request block so it looks like for example something like this:
```
requests:
  - start: 0x0200
    end: 0x0200
    mb_functioncode: 0x03
  - start: 0x0201
    end: 0x0255
    mb_functioncode: 0x03
  - start: 0x10B0
    end: 0x10BC
    mb_functioncode: 0x04
```

davidrapan, 2024-08-10:
> I found a Sofar Solar Modbus documentation. Haven’t checked if these are the registers from your integration.
> 
> Check it out here: https://solaranzeige.de/phpBB3/download/file.php?id=5254

The doc is really exactly for his model? Or perhaps some similar one?

miralpeix, 2024-08-10:
There we go! First all, thanks David for your work, your time, and your patience!

After to continue only a comment from what I have seen in the different forums and internet searches, every time there are problems with HA+ Solarman integrations with the inverter values ​​and they are shown as "unknown" in 90% of the cases the inverter is like mine: HYD 6000-EP sofar. 

Any test that needs to be done, I put my HA and my inverter at the service of the community. If you need to connect to it remotely, we can talk privately. At your service.

There we go with the logs: log+shot

[03.log](https://github.com/user-attachments/files/16570909/03.log)
[image]

*-*-*-*
[04.log](https://github.com/user-attachments/files/16570910/04.log)
[image]

*-*-*-*
[delete.log](https://github.com/user-attachments/files/16570911/delete.log)
[image]

*-*-*-*
[all.log](https://github.com/user-attachments/files/16570912/all.log)
[image]

*-*-*-*
[alternative 1.log](https://github.com/user-attachments/files/16570913/alternative.1.log)
[image]

*-*-*-*
[alternative 2.log](https://github.com/user-attachments/files/16570916/alternative.2.log)
[image]

*-*-*-*
[alternative 3.log](https://github.com/user-attachments/files/16570917/alternative.3.log)
[image]

PV-Joe, 2024-08-10:
> @PV-Joe? The doc is really exactly for his model? Or perhaps some similar one?
 I don‘t know, just did a search and found it.

Here is another source that might be helpful:
https://github.com/AndyWhittaker/HYD6000

davidrapan, 2024-08-10:
I have a suspicion that cause his device is EP it could be just wrong registries we are asking, cause the profile is for ES. Who knows...

githubDante, 2024-08-11:
Hi @davidrapan, 

SOFAR uses some kind of proprietary MODBUS protocol for some of the functions. Not the first complain from this brand. However the PDFs that can be found state that some info still can be obtained with the standard read_holding/read_input methods.

@miralpeix can you experiment with different addresses and provide the log from the test. Ignore the comments. Thank you in advance 

###### Registry_Details.pdf - error @ 0x0200 - fault codes through 0x0205 
###### Used also in the AndyWhittaker repo 
```
start: 0x0206
end: 0x020c
function_code: 0x03

start: 0x10B0
end: 0x10B4 
function_code: 0x04
```

###### This one is directly from the example in the documentation 
```
start: 0x0000
end: 0x0024
function_code: 0x03   

```
###### SOFARSOLAR Protocol.pdf

```
start: 0x0006
end: 0x000e
function_code: 0x03 

start: 0x1000 
end: 0x1005
function_code: 0x04 
```

davidrapan, 2024-08-11:
Hey @githubDante, thanks for joining the convo!

> SOFAR uses some kind of proprietary MODBUS protocol for some of the functions. Not the first complain from this brand.

Fuh! That's unfortunate. 😞

So it's going to be trial and error of what could be read from that device. :-/

@miralpeix, so it's up to you if you want to get your hands dirty. Or maybe we could set up some remote access then and I could look into it myself when I'll have some time to spare. 😶‍🌫️

miralpeix, 2024-08-11:
There we go with the homeworks for today: log+picture

[AndyWhittaker repo.log](https://github.com/user-attachments/files/16576321/AndyWhittaker.repo.log)
[image]

**************
[from the example.log](https://github.com/user-attachments/files/16576323/from.the.example.log)
[image]

**************
[SOFARSOLAR Protocol.log](https://github.com/user-attachments/files/16576324/SOFARSOLAR.Protocol.log)
[image]
[image]

@davidrapan I wish I had the technical ability to get my hands dirty... I wish!!!!
we will need to work in remote or give me very accurate instructions, but I don't know if I am ready...!

davidrapan, 2024-08-11:
Well yeah you forgot to put the dash in there :-/
```
  - start: 0x10B0
    end: 0x10BC
    code: 0x04
```

**Every start end code block needs to start with dash and be properly spaced.**

githubDante, 2024-08-11:
Probably the function code key also needs to be properly specified. Edited version:

###### Registry_Details.pdf - error @ 0x0200 - fault codes through 0x0205 

###### Used also in the AndyWhittaker repo 
```
start: 0x0206
end: 0x020c
mb_functioncode: 0x03

start: 0x10B0
end: 0x10B4 
mb_functioncode: 0x04
```
###### This one is directly from the example in the documentation 
```
start: 0x0000
end: 0x0024
mb_functioncode: 0x03   
```

###### SOFARSOLAR Protocol.pdf

```
start: 0x0006
end: 0x000e
mb_functioncode: 0x03 

start: 0x1000 
end: 0x1005
mb_functioncode: 0x04 
```

miralpeix, 2024-08-12:
Thanks and sorry! 😆 again:

[Used also in the AndyWhittaker repo.log](https://github.com/user-attachments/files/16578450/Used.also.in.the.AndyWhittaker.repo.log)
[image]

[directly from the example in the documentation.log](https://github.com/user-attachments/files/16578451/directly.from.the.example.in.the.documentation.log)
[image]

[SOFARSOLAR Protocol.log](https://github.com/user-attachments/files/16578455/SOFARSOLAR.Protocol.log)
[image]

githubDante, 2024-08-12:
Unfortunately only the strange  `01 90 02 00 00` responses. 
I have already sent a request for documentation to SOFAR for this model. Let's see what their answer will be.

githubDante, 2024-08-13:
Some documents which may help:

1.  [SOFAR Modbus User Guide](https://github.com/user-attachments/files/16595295/SOFAR.Modbus.User.Guide.2023-07-23.pdf)
2.  [sofar-hyd-3ph-and-sofar-g3-modbus-protocol](https://github.com/user-attachments/files/16595302/1697444348057-211116_sofar-hyd-3ph-and-sofar-g3-modbus-protocol-2021-10-14_client.xlsx)

According to 1 the protocol version which must be used for the EP inverters is SOFAR-G3. The second document contains the register mapping.

miralpeix, 2024-08-13:
> Hi @miralpeix
> 
> I had a similar issue, in fact pretty much the same I believe, but mine is the HYD 3000-EP Sofar. Also with similar firmware version.
> 
> Your post push me to find and search a bit more on the solution. And I think I got it!
> 
> On the 'original' [home assistant solarman repo](https://github.com/StephanJoubert/home_assistant_solarman) there a "sort of new file", that doesn't show on the "Automatic Config wizard". That file is this one: [hyd-zss-hp-3k-6k.yaml](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/hyd-zss-hp-3k-6k.yaml)
> 
> By attempting manual configurations, I got my **HYD 3000-EP Sofar working in HA**!
> 
> So after that success, I looked into getting that specific configuration `.yaml` onto this own [repo/solution](https://github.com/davidrapan/ha-solarman).
> 
> So here a the simple steps you can follow to attempt to replicate my results:
> 
> Prerequisites:
> 
> * Installed file editor on HA
> 
> Steps:
> 
> * Install `davidrapan/ha-solarman` on HA through HACS
> * Normally configure your options as prompted, don't worry about the `.yaml` file for now! Just select a random one!
> * It will say that there issues attempting to connect to your device, etc.... Normal behaviour!
> * Go to the file editor. From the root folder follow the path to: `/homeassistant/custom_components/solarman/inverter_definitions/`
> * Create a new file and name it whatever you want, ending of course with `.yaml`
>   
>   * Personally I preferred to choose the same file name, from where we will copy it's contents: `hyd-zss-hp-3k-6k.yaml`
> * Copy/Paste the contents of this file from `StephanJoubert/home_assistant_solarman` into our newly created one: [hyd-zss-hp-3k-6k.yaml](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/hyd-zss-hp-3k-6k.yaml)
> * Now go back to the configurations of Solarman ( `Settings`-> `Devices & Services` -> _on tab `Integrations`_ -> `Solarman`
> * Onto the `Devices`section click `Configure` and change the `.yaml` file to the newly create one (as see you can see in the image below.
> 
> <img alt="Screenshot 2024-08-13 at 04 04 51" width="120" src="https://private-user-images.githubusercontent.com/9058173/357262548-cff8004b-0fa9-4839-9bba-9da0b6fff35d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1MzkwMDEsIm5iZiI6MTcyMzUzODcwMSwicGF0aCI6Ii85MDU4MTczLzM1NzI2MjU0OC1jZmY4MDA0Yi0wZmE5LTQ4MzktOWJiYS05ZGEwYjZmZmYzNWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTNUMDg0NTAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2IwMzA1NmYyY2U4NWVjYmQ1NzllYTQ1ZDc1MjU1OWJmMjBmYmFmNTRmZjI2MTJhZWRhOGYwYjdkZjQyMzkwMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Oh5LuAPQYkZ7rvOo6sC649jIaTHvhuAvt0Y9L4uXGCY">
> Well that's it! At least for me that solved and immediately started getting data!
> 
> Hope this works for you, and also can help someone else!
> 
> Good luck!

I can't believe it! WORKS!!!!!
[image]

closed 2024-08-13 by miralpeix

davidrapan, 2024-08-13:
Great!

Anyway @miralpeix, this repo also contains that file. It's called: zcs_azzurro-hyd-zss-hp.yaml

miralpeix, 2024-08-13:
Thanks David and all the community for the amazing support. A friend here

davidrapan, 2024-08-13:
I can add it also under sofar brand if you want. :)

miralpeix, 2024-08-13:
> I can add it also under sofar brand if you want. :)

all that I can contribute, please, will be happy!

miralpeix, 2024-08-13:
Yes, works perfectly too

ssimoe50, 2024-08-13:
I will try later today that file you spoke too @davidrapan, but great if soo! 

 Glad I could help @miralpeix ! 😄 
 
 @githubDante Thanks for closing that issue too. Indeed it was "just as simple" as changing and testing all alternatives I could find, and on find one that worked, thankfully!
 
 Have a nice day you all!

githubDante, 2024-08-13:
> Awesome!
> 
> @githubDante? So that CRC Error was just for trying to read incorrect registers?

@davidrapan yes. In fact the packet does no contain a CRC code at all.