davidrapan/ha-solarman #72: [Feature Request] Add facility to be able to stop / start integration from an Automation

gedger, 2024-08-11:
Issue https://github.com/davidrapan/ha-solarman/issues/17 requested to reduce the number of errors / warnings logged when Solar only powered inverters go off-line at night and then come online in the morning. The changes implemented in https://github.com/davidrapan/ha-solarman/issues/17 have improved the situation but the final and best solution is to be able to instruct the integration to cleanly disconnect from the inverter and to stop trying to connect until such time as the integration is enabled again. This could be achieved by the integration providing a binary_sensor or switch that can be enabled / disabled by an automation at sunset and sunrise.

gedger, 2024-09-11:
Keep Alive

davidrapan, 2024-10-12:
Keep Alive

davidrapan, 2024-11-13:
Keep Alive

Thiiib, 2025-01-05:
If we can help let us know