davidrapan/ha-solarman #73: [Feature Request] Add facility for binary_sensors in inverter definitions

gedger, 2024-08-11:
Solar Inverters have several single but status flags that are currently represented in HA as normal sensors. The advantage of using a binary_sensor is that HA will better show the on/off state by changing the icon colour and icon.

PV-Joe, 2024-08-11:
This sounds like a great idea. 👍

I colored the icon using a mushroom chip card for the Battery State using the following code. I think your idea does the same trick.

```
type: custom:mushroom-chips-card
chips:
  - type: entity
    entity: sensor.solarman_battery_state
    card_mod:
      style: |
        ha-card {
          {% if states('sensor.solarman_battery_state') == 'standby' %}
            --card-mod-icon-color: green;
          {% elif states('sensor.solarman_battery_state') == 'charging' %}
            --card-mod-icon-color: orange;
          {% elif states('sensor.solarman_battery_state') == 'discharging' %}
            --card-mod-icon-color: red;
          {% endif %}
        } 
```

davidrapan, 2024-08-12:
[feat: Add Binary Sensor platform](https://github.com/davidrapan/ha-solarman/commit/e09984815f3e3b8e74c0b5a7be7547e60de229c3)

But I will also change that "class is sometimes platform" mess so wait some more time before you start using it. 😉

davidrapan, 2024-08-13:
@PV-Joe, this is OT here but it's better like this:
```
type: custom:mushroom-chips-card
chips:
  - type: entity
    entity: sensor.solarman_battery_state
    card_mod:
      style: |
        ha-card {
          {% if states('sensor.solarman_battery_state') == 'charging' %}
            --card-mod-icon-color: orange;   
          {% elif states('sensor.solarman_battery_state') == 'charging' %}
            --card-mod-icon-color: red;
          {% else %}
            --card-mod-icon-color: green;
          {% endif %}
        } 
```

davidrapan, 2024-08-15:
It's done.

[refactor: Differentiation using platform property](https://github.com/davidrapan/ha-solarman/commit/1dfc4b066a47f7423b8603412009833092320eae)

gedger, 2024-08-16:
Excellent, thank you. So for my clarity and for others before updating their configurations we can now specify.

**platform**: sensor, binary_sensor, switch, number, select, time
**state_class**: measurement, total, total_increasing
**device_class**: [sensor](https://www.home-assistant.io/integrations/sensor#device-class), [binary_sensor](https://www.home-assistant.io/integrations/binary_sensor/`)
**uom**: whatever you like (if missing then presumably it will use the home_assistant uom associated with the device_class?)

Is the number platform a numeric sensor? So this could also be achieved by setting sensor and an appropriate device_class?

davidrapan, 2024-08-16:
**platform** - Yes. (when not set sensor is default value)
**state_class** - Yes.
**device_class** - No, here you put things like (or did I misunderstood that you provided those links as for examples what can be put in there? :-D):
- battery, power, energy, ...: for sensor platform or number platform
- battery, power, connectivity, ...: for binary_sensor platform
- switch, outlet: for switch platform
**uom** - Not sure if it's set by default by HA 🤔

> Is the number platform a numeric sensor? So this could also be achieved by setting sensor and an appropriate device_class?

_If I understand HA logic correctly they use term sensor for read-only values._

**Now you should put platform: number for R/W entity (but for backward compatibility reasons "configurable:" is still treated as number platform).**

davidrapan, 2024-08-16:
**And I also added "bit:" property (works with rule: 1) to easily check if specific bit on given index is set or not.**

davidrapan, 2024-08-16:
@PV-Joe, @gedger, I'm gonna use this thread as it's bout the binary_sensor so what are your thoughts on...

When you set for that sensor "device_class: power". It shows it like this:
[image]

gedger, 2024-08-16:
That looks correct, Power should be shown as Detected when on and Clear when off as per the [binary_sensor](https://www.home-assistant.io/integrations/binary_sensor/). The docs aren't the clearest though....I haven't had the chance to play yet....

gedger, 2024-08-16:
Here's a grid sensor from my Powerwall.
[image]

davidrapan, 2024-08-16:
Okay then! 👍

gedger, 2024-08-16:
I come from the generation when you only supplied compiled code and libraries so documentation was everything. My first work was in assembler for the intel 4040....Still trying to get used to this pesky snake code called Python 🤣

gedger, 2024-08-16:
> I don’t think this is very helpful in my opinion. I suggest to leave this sensor as it as it was.

binary_sensor didn’t exist before so this is only applies to these. Normal power sensors are exactly the same. This is also how HA has decided to show a binary power sensor so the change request needs to be made there or just pick another binary_sensor type that has off/on or however you want it displayed. Underneath the sensor is on/off the device_class simply tells HA how to interpret and display.

gedger, 2024-08-16:
What have I missed, just tried latest main at updated my config file and get errors.
[home-assistant_2024-08-16T15-14-58.475Z.log](https://github.com/user-attachments/files/16638638/home-assistant_2024-08-16T15-14-58.475Z.log)

davidrapan, 2024-08-16:
Yeah I was fixing that like an hour ago.

gedger, 2024-08-16:
OK, I'll wait for a new commit.

davidrapan, 2024-08-16:
It should be already there in this one: [fix: ParameterParser KeyError during init](https://github.com/davidrapan/ha-solarman/commit/75a6809fcc17b1310865c6a1b0f6bb6ffe7da856)

gedger, 2024-08-16:
Doh! Don't use HACS redownload main, it doesn't work!

davidrapan, 2024-08-16:
> Doh! Don't use HACS redownload main, it doesn't work!

I know that. 😆

gedger, 2024-08-16:
Works great 👏  Has thrown up one small addition....can the binary_sensor have the option to invert or flip? Several of my sensors are the wrong way round, for instance the grid sensor is true when the grid has failed and zero when it hasn't 🙄

gedger, 2024-08-16:
It looks like the platform: number has an issue when configurable: option is not present. Adding configurable: clears the error.

```
2024-08-16 17:34:45.915 ERROR (MainThread) [homeassistant.components.number] Error while setting up solarman platform for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 372, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 631, in async_add_entities
    for entity in new_entities:
  File "/config/custom_components/solarman/number.py", line 31, in <genexpr>
    async_add_entities(create_entity(lambda s: SolarmanNumberEntity(coordinator, s), sensor) for sensor in sensors if is_platform(sensor, _PLATFORM) or "configurable" in sensor)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/entity.py", line 21, in create_entity
    entity = creator(sensor)
             ^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/number.py", line 31, in <lambda>
    async_add_entities(create_entity(lambda s: SolarmanNumberEntity(coordinator, s), sensor) for sensor in sensors if is_platform(sensor, _PLATFORM) or "configurable" in sensor)
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/number.py", line 56, in __init__
    configurable = sensor["configurable"]
                   ~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'configurable'
```

davidrapan, 2024-08-16:
[fix: Number platform KeyError: 'configurable'](https://github.com/davidrapan/ha-solarman/commit/585358d5b84f992cd9a822b95536934f7958b5c7)
[feat: Inverted property for binary_sensor platform](https://github.com/davidrapan/ha-solarman/commit/08d530dca171d8b2ff0f57d316b0e9854a1af21a)
- inverted: True

gedger, 2024-08-16:
Confirm fixed. Will test some more tomorrow, time for a beer.

davidrapan, 2024-08-16:
@gedger? Can I have a question? What "Inverter AC Export Power" means? In the contrast to the "Inverter AC Power" for example? And "Total DC Power" Means like PV1 power + PV2 power?

gedger, 2024-08-17:
I have a grid export limitation at my property, this means when the export power reaches the power limit the inverter has to back off. This function is called EPM or Export Power Management but the measurement to make it possible is Inverter AC Export Power, perhaps it would be better called Grid Export Power as that is what is being measured via a current transformer.

Yup, Total DC power is PV1+PV2, I now call it DC Power.

Sure, although I may not have strictly followed the naming convention changes....

davidrapan, 2024-08-17:
So it's the same as the Zero Export Power in SG04LP3? Which value you have there?

> Yup, Total DC power is PV1+PV2, I now call it DC Power.

For the record I also think it should be called with universal DC but... 😆 

And at the end I really don't care about the naming or simply how are the entities called. My only goal is to have the same name for entities with the same purpose (across all profile files) for clarity so it's easier and cleaner for people who are using several different devices (sure I'm now one of them so call me out for being a biased! 😆).

gedger, 2024-08-17:
It’s the same principle but in the UK each property gets an export amount (never zero) with a minimum of 3.6kW. So rather than Zero Export Power it’s ‘n’ Export Power which is why I called it Inverter Export Power. This is a real time measurement of how much power is currently be exported to the grid.

davidrapan, 2024-08-17:
Yeah I mean the Zero there means something different (at least from what I understood).

It is a value which inverter considers as "zero" and tries to balance power from grind around that value. I have it set to 100W which for inverter means that he tries to balance power which flows from the grid around 100W.

Is that it?

gedger, 2024-08-17:
The official name is Backflow Power but it's continually showing how much power is currently being exported to the grid. The inverter monitors this value and when it reaches your maximum export amount it backs off. The actual power value the inverter back off at is another modbus register that I never found. 

https://midsummerwholesale.co.uk/pdfs/solis-export-power-manager-manual-.pdf

So this register is the value measured by that CT mentioned in the pdf.

gedger, 2024-08-17:
Here's my latest inverter definition using the latest binary_sensor. Note I removed name and inverter ID as it no longer worked for me after a software update.

[solis_1p8k-5g.yaml.zip](https://github.com/user-attachments/files/16645525/solis_1p8k-5g.yaml.zip)

Here's also the Solis inverter ModBus spec if you'd like to include it on your docs page, may help someone. Unfortunately not all inverters support all registers so it's a suck it and see approach...

[_Without Control Grid-Tied EN 2021.5.26_RS485_MODBUS.pdf](https://github.com/user-attachments/files/16645546/_Without.Control.Grid-Tied.EN.2021.5.26_RS485_MODBUS.pdf)

davidrapan, 2024-08-17:
> The official name is Backflow Power but it's continually showing how much power is currently being exported to the grid.

So it is value for reading and not for configuration?

> Here's my latest inverter definition using the latest binary_sensor. Note I removed name and inverter ID as it no longer worked for me after a software update.

Thanks! Which software you mean? 😮 

> Here's also the Solis inverter ModBus spec if you'd like to include it on your docs page, may help someone. Unfortunately not all inverters support all registers so it's a suck it and see approach...

Great! I sure will!

gedger, 2024-08-18:
> Thanks! Which software you mean? 😮
The inverter software was updated and the name and ID were no longer correct, they served no useful purpose to me so I didn't  bother to debug. 

> So it's for the 1p-5g and presumably some others?
In theory yes.

gedger, 2024-08-18:
Happy to close this now?

davidrapan, 2024-08-18:
> The inverter software was updated and the name and ID were no longer correct, they served no useful purpose to me so I didn't bother to debug.

Detection of the type of the device for example 😋 (I would love to be able to do that in the future maybe...) But that's more for the integration than you, sure, haha.

> Happy to close this now?

😉

closed 2024-08-18 by davidrapan