davidrapan/ha-solarman #76: Production Values & Alert Entities not working anymore

aViN-187, 2024-08-11:
Hi there,

i'm kind of new to home-assistant therefore please be kind to me. After the Update Update of 2024.8 (at least that is my assumption) I'm got the following issues:

Value of my 2MTTP Deye in total comsumption does not match. 

in the past i always got values between 2-5 kwh. A typical day looks like this: 
[image]
After the update it looks like this:
[image]
Values add 200 kwh to a day ... same goes on the next days:
[image]

Value=sensor.deye_total_production
[image]

on 6th August something happened maybe? But now clue ... 

2nd Issue: 
The Value: Altert on the Deye Decvice - every 5 secounds the value chances. In my mind i can't remember this behaviour. When i check the micro inverter manually via the ip/statuspage - all looks good. 

[image]

Maybe someone could help me out - thanks

davidrapan, 2024-08-11:
Hi @aViN-187, I think it's proly the same as https://github.com/davidrapan/ha-solarman/issues/65. I made the fix in: [fix: Deye microinverters PV validation to 0.1](https://github.com/davidrapan/ha-solarman/commit/6ea759340755d80be6f96aadeaf39c509a3236e6). And it's already part of pre-release v24.08.10 but you will have to enable beta versions in the HACS under redownload button in the right top corner to see beta (pre-release) versions.

davidrapan, 2024-08-11:
Duplicate of https://github.com/davidrapan/ha-solarman/issues/65

aViN-187, 2024-08-11:
thanks for the quick reply - i activated beta and did the installation. I will report tomorrow. I assume the "wrong" values from the past days will not be corrected by the dashboard?

davidrapan, 2024-08-11:
Sadly no :-/ But you can at least maybe clear them using Developer tools and then STATISTICS page. You can find the Total Production sensor and adjust the data.

aViN-187, 2024-08-12:
Issue resolved: 

Data is looking good so far - thanks! 
[image]

nethertheless Alert Issue is still the same - what can i do to help out?
[image]

davidrapan, 2024-08-12:
I'm glad to hear that!

> Alert Issue is still the same - what can i do to help out?

I'll look into it and let you know. 😉

davidrapan, 2024-08-12:
@CrazyUs3r? You were doing the changes there. Could you maybe elaborate? Thanks.

CrazyUs3r, 2024-08-12:
@aViN-187 Which inverter do you use with which firmware

aViN-187, 2024-08-12:
hey @CrazyUs3r 

Modell: deye sun800g3
[image]

The behavior startet 6. August. Before that no changes on the Alert.

[image]
After that - see post above

davidrapan, 2024-08-12:
Try definition profile from this commit: [feat: Updated Deye 2MPPT and 4MPPT](https://github.com/davidrapan/ha-solarman/commit/6f9e9e70d39aacee83a8b8297a40c43f19896d4a)

CrazyUs3r, 2024-08-12:
@davidrapan with mod protocol are you using?

aViN-187, 2024-08-12:
> Try definition profile from this commit: [feat: Updated Deye 2MPPT and 4MPPT](https://github.com/davidrapan/ha-solarman/commit/6f9e9e70d39aacee83a8b8297a40c43f19896d4a)

Not working at all - 9 Duplicate Keys, etc. ... at least for the 2MPPT Version

davidrapan, 2024-08-12:
Oh did I f*cked up something?

CrazyUs3r, 2024-08-12:
i I'll try tomorrow morning since I don't have any sun anymore

davidrapan, 2024-08-12:
Yeah well and be aware that this version already did go trough naming standardization and for example Daily Production sensors changed into Today Production, etc. So migrate your history if you need that under new name using step by step guide from wiki. (But that sensor is not that important anyway as all the data are already in the Total version of that sensor but letting you know so you are not surprised) 😉

aViN-187, 2024-08-13:
Update:

Total Production looks strange ... i got values in the statistics, but not in the energy overview. 

[image]

[image]

Alert Management looks fixed - since i restartet HA with the new 2MPPT file - all goods, thanks!

[image]

davidrapan, 2024-08-13:
Look at the entity id of that sensor, didn't it got messed up somehow?

CrazyUs3r, 2024-08-13:
I have no Problem. I night is all offline but total go normal up without 0

[image]
[image]

CrazyUs3r, 2024-08-13:
Load the firmware 32-D1 again up in inverter

davidrapan, 2024-08-13:
@CrazyUs3r, great!

@aViN-187,
> Alert Management looks fixed - since i restartet HA with the new 2MPPT file - all goods, thanks!

Does the "Value" in attributes changes? If so does it appear simply random or some specific values? If so, what values?

CrazyUs3r, 2024-08-13:
and my device status = fault is a problem from the ac wire this i check at the weekend

aViN-187, 2024-08-13:
in the meantime values did appear .. so i assume all good for now.
[image]

in regards to the error message - not value at all, but now since no sun in shining the state changes to error. 
[image]

I assume that behaviour correct? :) Nethertheless big thanks for your work

davidrapan, 2024-08-13:
@aViN-187 That's awesome and you are very welcome! 🥂

Anyway, still suggest you to delete and re-add your device to clear out those nonexistent sensors. 😉

closed 2025-02-03 by davidrapan