davidrapan/ha-solarman #81: Total Battery Life Cycles and Battery SOH show low values

PV-Joe, 2024-08-14:
Hi David,

The sensor „Total Battery Life Cycles“ and Battery SOH shows low values in my opinion.

[image]

My battery:
1x Pytes E-Box 5,12 KW
(running since two years)

Config Solarman:
[image]

Did I misconfigure or is it a bug?

davidrapan, 2024-08-14:
What is your "Corrected Capacity" from Battery attributes?

PV-Joe, 2024-08-14:
Sorry, I might be dumb but really don’t know what to do.

I did not correct anything and don’t know where I have to.

Battery Capacity is:
[image]

davidrapan, 2024-08-14:
So Battery Capacity is set ~ correctly (according info you provided).

And the Corrected capacity can be seen here:
[image]

PV-Joe, 2024-08-14:
Here is mine:

[image]

Did I miss something?

davidrapan, 2024-08-14:
So yeah your inverter thinks you have 10 kWh battery. You can correct that by switching your inverter battery control to: "Use Batt %" and let it run in that mode until your Corrected Capacity value does not correct itself to ~ your set Battery Capacity. But it's kinda important you also have correctly set other values about your batt for inverter to work properly in that mode.

_I'll post here for you some pics from my setting in a bit. If you decide to do something about it._

PV-Joe, 2024-08-14:
Thanks for explaining!

My solar technician chose the battery settings and I never changed anything. As I don’t really know what exactly I’m doing  by changing values I’d rather leave it like he set it up.

My battery cycles when it is full from 99% to 100% and back to 99% and so on if you mean that.

I there another way of getting the sensor going?

davidrapan, 2024-08-14:
> My battery cycles when it is full from 99% to 100% and back to 99% and so on if you mean that.

Really? It shouldn't do that. If there is enough PV and battery is not needed it should stay idle.

PV-Joe, 2024-08-15:
No, no charging or discharging from grid at all.
Experts told me, that it is normal for Lithium batteries and is because of cell balancing.

So is there a way to get the sensors running by modifying something in the integration or Home Assistant itself?

If yes, how exactly should I do it?

davidrapan, 2024-08-15:
> No, no charging or discharging from grid at all.
Experts told me, that it is normal for Lithium batteries and is because of cell balancing.

Oh yeah that in fact is normal but not really every time or all the time. It should stop after is the balancing somehow done.

**I still do wonder what is your Battery BMS Alarm showing when is your battery full? Does it stays Inactive the whole time?**

But anyway mine do charge or discharge from grid when full so that'S why I have setup that "hack".

> So is there a way to get the sensors running by modifying something in the integration or Home Assistant itself?

I'll see what I can do about it. 😉

PV-Joe, 2024-08-15:
Battery BMS Alarm says “inactive” since switching to your integration.

Maybe you could make the installed battery capacity configurable in the setup menue?

Otherwhise I could help myself with template sensor for Battery SOH.

Just asked ChatGPT and this is the result:

```
sensor:
  - platform: template
    sensors:
      battery_soh:
        friendly_name: "Battery SOH"
        unit_of_measurement: "%"
        value_template: >
          {% set battery_capacity = 5.12 %}
          {% set life_cycle = 6000 %}
          {% set total_capacity_kwh = battery_capacity * life_cycle %}
          {% set consumed_kwh = states('sensor.solarman_total_battery_charge') | float(0) %}
          {% set soh_percentage = (consumed_kwh / total_capacity_kwh) * 100 %}
          {% set result_soh = 100 - soh_percentage %}
          {{ result_soh | round(2) }}
```

davidrapan, 2024-08-15:
> Maybe you could make the installed battery capacity configurable in the setup menue?

I really want this to be tied to a data from the inverter. As there are two values, one set by configuring of the inverter and the other which inverter provides using one of the registers.

Maybe I could somehow detect when the provided values is way off and used the set one instead.

> Just asked ChatGPT and this is the result:

Haha! That's funny! And still so scary though! 😆 He will take all the jobs from us one day. 😉

davidrapan, 2024-08-15:
[fix: Battery SOH when wrong corrected capacity](https://github.com/davidrapan/ha-solarman/commit/369e2fb786d764651025c80568fa149a76178bfa)

davidrapan, 2024-08-22:
So how is it in new version?

PV-Joe, 2024-08-22:
**This is the result for my 5,12 kWh battery:**

[image]

[image]
[image]

Can you validate?

davidrapan, 2024-08-22:
I forgot to also do the changes for cycles: [fix: Battery cycles when wrong corrected capacity](https://github.com/davidrapan/ha-solarman/commit/b69f577b058c2ed0c91300a76f995b47d355d191)

**What about SOH? That should be correct for some time already.**

PV-Joe, 2024-08-22:
Here it is:

[image]

Do I just have to overwrite the *.py and it will be permanently fixed with the next update?

davidrapan, 2024-08-22:
> Do I just have to overwrite the *.py and it will be permanently fixed with the next update?

Yes. If you replace the content of sensor.py with this: https://github.com/davidrapan/ha-solarman/blob/main/custom_components/solarman/sensor.py

PV-Joe, 2024-08-22:
Will do it today and send my feedback here.

Thank you for your continuous and extra fast support!

PV-Joe, 2024-08-22:
Here ist the value after replacing the sensor.py:

[image]

Are the values constistent in your opinion?
If yes, we can close as completed :-)

davidrapan, 2024-08-22:
Yes it should double as it did after the changes. So seems now okay to me. 😉

closed 2024-08-22 by davidrapan

PV-Joe, 2024-08-25:
> Yes it should double as it did after the changes. So seems now okay to me. 😉 

But the Battery SOH in % doesn't fit to the Lifecycles?

[image]
[image]

davidrapan, 2024-08-25:
6000 cycles and you are at 80% so I don't think so, hehe.

PV-Joe, 2024-08-25:
Ahh you are setting 80% as the limit. 😂 That confused me.

I think we should add and explain that in the description to make it clear. Maybe together with the formula that is used.

It should be clear for non-experts I think and will reduce silly questions like mine 😂

davidrapan, 2024-08-25:
There is no limit... And that is a literal description of a life cycles of the battery. 80% at 6000 cycles as the worst case scenario.

davidrapan, 2024-08-25:
How did you calculate your number anyway? 93% after just almost 400 cycles would be really bad battery, though.

> This will make the calculation useless, right?

Not necessarily. Just less precise. It is tied to the inverter anyway so replacing the battery would be the act which could render this calculation as useless for sure. You have to keep in mind that it's just for a "having idea" about the state of the battery. Nothing you should be willing to put your hand in the fire for, though. 😆

I added this feature cause I'm charging the battery during T2 (cheaper electricity) so I wanted to know how fast is my battery proly degrading cause of that. So you should be treating the values more like as it changes with time rather than current value. You know what I mean? 😉

davidrapan, 2024-08-28:
> I would add not replace.

I wasn't talking about your case but giving an example of what would render the values useless.

> Yeah, I like monitoring it as well. I'm mostly autark but as you have seen I don't need much energy. I'm more an enthusiast and like the technology.

Maybe there could be a way of solving it by adding some kind of a offset which you would change with your battery addition. Let me know when you'll do that and we can try something. 😉

PV-Joe, 2024-08-28:
Great news. Thank you. I will see when my technician has time for it. The Battery firmwares need to be updated to match each other and he probably wants to update the Deye Hybrid as well. Don't know if I want a Deye Update as some of them are reported as not very good.