davidrapan/ha-solarman #84: Update Intervals not working

Steinbichler, 2024-08-15:
First thank you for this fine solarman integration! Great Job!

I am running homeassistant in actual version. My converter is a Deye 600. Relais is installed.

I have your ha-slarman up an running installed via HACS. Nearly everthing works fine but:

I have requested at Deye support to set the update interval for my Deye 600 to 1 Minute. 
They agreed to have it done. I see in the Homepage of the converter the update time is set to 1.

But nothing changed in the ha-solarman. I changed the number in the definition file: deye_2mppt.yaml.
After the installation the update_interval: 5 was set. 
What ever the number I enter, and for sure I did a restart after the change, nothing happens. 
The update interval is always 5 Minutes.

Can you help please?

Thank you an regards Roland

davidrapan, 2024-08-15:
Hi @Steinbichler, I'm little bit worried that you are mixing different things here together. 

> I have requested at Deye support to set the update interval for my Deye 600 to 1 Minute.
They agreed to have it done. I see in the Homepage of the converter the update time is set to 1.

This integration and HA has nothing to do with Deye and their support and is completely separated thing. No configuration from Deye side won't affect this integration at all.

And at the same time nothing set in this integration will affects Deye Cloud nor Solarman App. Again it is completely separate thing from those.

> But nothing changed in the ha-solarman. I changed the number in the definition file: deye_2mppt.yaml.
After the installation the update_interval: 5 was set.
What ever the number I enter, and for sure I did a restart after the change, nothing happens.
The update interval is always 5 Minutes.

What are you trying to achieve? update_interval: 5 in the 2mppt.yaml means 5 seconds not 5 minutes.

Steinbichler, 2024-08-15:
Hi if set to 5 as ist should be 5 seconds I have this:
[image]
Update was 6 Minutes before.

davidrapan, 2024-08-15:
What shows Update Interval sensor in diagnostic card in device detail view?

CrazyUs3r, 2024-08-15:
@Steinbichler 
Wie gut ist das wlan Signal 
Mein m80 macht auch nur alle 5min ist aber auch auf 1min gestellt. In der solarman app kommen die Signale aber auch nur im 5min intervall an

Steinbichler, 2024-08-15:
@davidrapan
[image]

do you mean this one

@CrazyUs3r 
Das Wlan kann das Problem nicht sein es ist sehr gut und in der Solarman App kommt der update nach einer Minute

Steinbichler, 2024-08-15:
[home-assistant_solarman_2024-08-15T17-36-23.029Z.log](https://github.com/user-attachments/files/16628192/home-assistant_solarman_2024-08-15T17-36-23.029Z.log)

I hope it helps.Thank you an Regards Roland

davidrapan, 2024-08-15:
What the actual f*ck is happening in there!

Could I please ask you to do the following please:
1. Disable debug logging (if you have it enabled)
2. Restart Home Assistant
3. Remove the device from the integration
4. Enable debug logging
5. Add your device back again and go trough configuration
6. Let it run for some time
7. Disable debug logging and post here the log file again

Thank you for your cooperation! 😉

PV-Joe, 2024-08-15:
I remember this behavior after my first installation.

The Update information in seconds or minutes from Home Assistant was showing wrong values but the sensors updates as expected.

I haven’t done anything about it and wondered. The problem desolved by itself after a while.

Steinbichler, 2024-08-16:
@davidrapan 

Hallo I have done it in your requested way. After the update and the delete of the converter in HO the automatic recognition does not work any more,

[image]

[image]

and the Log file

[home-assistant_solarman_2024-08-16T07-55-56.658Z.log](https://github.com/user-attachments/files/16634545/home-assistant_solarman_2024-08-16T07-55-56.658Z.log)

Steinbichler, 2024-08-16:
Sorry about the confusion in the deleted comment.
It was a problem with my HO installation everything woks fine I will upload the log asap.
Regards Roland

CrazyUs3r, 2024-08-16:
My log
[home-assistant_solarman_2024-08-16T08-13-04.070Z.log](https://github.com/user-attachments/files/16634730/home-assistant_solarman_2024-08-16T08-13-04.070Z.log)

davidrapan, 2024-08-16:
@Steinbichler and when you are observing the value does it really update with such a long period? Or just the HA shows it incorrectly?

davidrapan, 2024-08-16:
Maybe [fix: Proper implementation of sensor platform](https://github.com/davidrapan/ha-solarman/commit/16b4522b51d8006814d84438c7dc42c20e1d7a37) will do something about that strange update behavior.

CrazyUs3r, 2024-08-16:
I test it also

Steinbichler, 2024-08-16:
@davidrapan.

I did a test to cover my pannels. And yes ist last at least the 5 to 6 Minutes till I see the change in HA.

A question to my understanding: Even if the content of the register in my inverter did not change,  the update in HA-solarman should read the unchanged content after the interval time? Is it not in this way?

And for the link in your comment to the definition file? Should I change ist like it is in the description? and clear all the names?

davidrapan, 2024-08-16:
> I did a test to cover my pannels. And yes ist last at least the 5 to 6 Minutes till I see the change in HA.

Fuu, that is really strange :-/

> A question to my understanding: Even if the content of the register in my inverter did not change, the update in HA-solarman should read the unchanged content after the interval time? Is it not in this way?

It will be read from the inverter. BUT! **Ha won't update it's "Last updated" until the value really changes.**

> And for the link in your comment to the definition file? Should I change it like it is in the description? and clear all the names?

**It was link to a commit, it affects whole source code not just definition profile.**

CrazyUs3r, 2024-08-16:
[image]

davidrapan, 2024-08-16:
@CrazyUs3r, you did not wait long enough for the file to finish uploading, hehe.

davidrapan, 2024-08-17:
@Steinbichler Try [v24.08.17](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.17) and see if it won't fix your issue.

bjoernmayer, 2024-08-19:
Heyho,
I am an `24.08.17` and I can see the same behaviour. Data is updated around every 5ish minutes

davidrapan, 2024-08-19:
Hey @bjoernmayer. Please enable debug logging and post the log here.

Steinbichler, 2024-08-19:
Hallo I am on 24.08.17 here again a log from my Installation.
[home-assistant_solarman_2024-08-19T14-20-56.963Z.log](https://github.com/user-attachments/files/16661444/home-assistant_solarman_2024-08-19T14-20-56.963Z.log)
Regards

davidrapan, 2024-08-19:
Hey @Steinbichler, from sensor named "Fault information Word 2" remove:
```
        lookup:
          - key: 0x0000
            value: "OK"
```

Steinbichler, 2024-08-19:
Hallo davidrapan done, but no change.
Here again a logfile from the time after the change in the devinition file.

[home-assistant_solarman_2024-08-19T16-00-04.808Z.log](https://github.com/user-attachments/files/16662697/home-assistant_solarman_2024-08-19T16-00-04.808Z.log)

[image]
[image]

CrazyUs3r, 2024-08-19:
i have add a 2mppt.yaml and hier my log
[home-assistant_solarman_2024-08-19T16-00-44.380Z.log](https://github.com/user-attachments/files/16662711/home-assistant_solarman_2024-08-19T16-00-44.380Z.log)

davidrapan, 2024-08-19:
@Steinbichler, according to the log there is clearly visible that HA talks with the inverter every 5 seconds and also gets the data correctly.

Whatever is causing the issue is happening after that in the HA cause there are also others like here for example @CrazyUs3r with the same device and it's working okay for him... So I'm really out of ideas what's causing issues for you.

Do you have the auto polling enabled?

CrazyUs3r, 2024-08-19:
By the way, I still have the upload time of 5 minutes. I tried to change it with github.com/s10 his exe but it doesn't work. I already wrote to deye. I have also already changed was @davidrapan changed and everything that was done.

CrazyUs3r, 2024-08-19:
No i have 1min
[image]

davidrapan, 2024-08-19:
@CrazyUs3r, this is irrelevant for our integration.

davidrapan, 2024-08-19:
I have also on the cloud refresh rates in minutes but it just means that servers from deye/solarman are asking for data in that interval.

CrazyUs3r, 2024-08-19:
Update is comming in 1min
[image]

davidrapan, 2024-08-19:
@CrazyUs3r What about PV1 Power also every 1m?

davidrapan, 2024-08-19:
I just checked it and it your devices really are not giving out new data with every request... My device does (It's also deye, even with set that on deye cloud to 1m) so i thought it should work the same for you guys too.

**But that means it should be set somewhere in the device, right?**

davidrapan, 2024-08-19:
@CrazyUs3r, can you show me screen of your: http://{inverter_ip}/config_hide.html

CrazyUs3r, 2024-08-19:
@davidrapan sure I'll send it to you tomorrow today it's too dark

CrazyUs3r, 2024-08-19:
@davidrapan here all from ha
[image]

davidrapan, 2024-08-19:
Nono i was talking about http://{inverter_ip}/config_hide.html it's a page for the stick logger.

davidrapan, 2024-08-19:
Ok.. mh..

davidrapan, 2024-08-19:
What happens when you switch it to transparency and hit save?

CrazyUs3r, 2024-08-19:
I haven't figured out what the difference is yet. I switched last year and nothing changed. Maybe you know what the difference is.

davidrapan, 2024-08-19:
I was wondering if it maybe won't affect the update_interval in some way.

CrazyUs3r, 2024-08-19:
So now it is separated again and all data is gone
[image]

davidrapan, 2024-08-19:
What shows the log?

davidrapan, 2024-08-19:
Anyway wrap up of this issue: As the inverter itself is the one who is sending (old data) and changing them for us once every 1m for @CrazyUs3r or 5m for @Steinbichler it's is sadly not an issue of the integration but either some sort of a hidden setting of the inverter cause it looks like this period can be at least change to 5m or 1m or it's hardware limitation. :-/

CrazyUs3r, 2024-08-20:
> You mean cause of night, right?

Yes

davidrapan, 2024-08-20:
Would like to know from where that value of 60s or 5m comes from.

CrazyUs3r, 2024-08-20:
That's a good question in my m80 the solarman says 1min for upload 60s for the other but the values ​​in the app are only in 5min intervals

davidrapan, 2024-08-20:
And did you also try deye cloud instead of solarman app?

CrazyUs3r, 2024-08-20:
I just have both apps to look at, but I'm slowly moving over to the deye cloud

davidrapan, 2024-08-20:
And is it the same? The update_interval?

I'm currently not using any of the apps cause now I'm finally completely satisfied with what my integration + HA is offering to me so I really not just not need them but they don't even remotely offer comparable value at all.

And soon I'm gonna also try to disconnect my devices from their servers completely.

CrazyUs3r, 2024-08-20:
Yes, both apps show the same thing.

I'm on your side when it comes to your decision to separate. The project in HA is running very well for me too.

Oh, if you want to separate, the update time won't be cut off any more. Or the daily values ​​won't be reset.

CrazyUs3r, 2024-08-20:
Look in my log and I see that data is pulled every 5s

[home-assistant_solarman_2024-08-20T10-05-02.693Z.log](https://github.com/user-attachments/files/16674272/home-assistant_solarman_2024-08-20T10-05-02.693Z.log)

davidrapan, 2024-08-20:
Yes they are but the content of the data is the same every time. That's the issue.

CrazyUs3r, 2024-08-20:
I agree with you completely and always overwriting data with the same value would not be any better

bjoernmayer, 2024-08-20:
Heyho,

this is my `config_hide.html`:
[image]

The tcp timeout is set to 5 minutes, but this sounds unrelated

CrazyUs3r, 2024-08-20:
And what inverter typ??

davidrapan, 2024-08-20:
@Steinbichler or @bjoernmayer don't you want to also try to write to your supp as @CrazyUs3r did to lower that update period as an experiment if it won't (for some "really mysterious" reason) help?

bjoernmayer, 2024-08-20:
> And what inverter typ??

Deye 600 s3

> @Steinbichler or @bjoernmayer don't you want to also try to write to your supp as @CrazyUs3r did to lower (make it more often) that update period as an experiment if it won't (for some "really mysterious" reason) help?

I just did. Let's see what happens.

Steinbichler, 2024-08-22:
Hallo David, that is the point I mentioned at the beginning of this item.

I have requested at Deye support to set the update interval for my Deye 600 to 1 Minute.
They agreed to have it done. I see in the Homepage of the converter the update time is set to 1.

I now again send them a mail to ensure they have done it and I got the the answer: Yes we did!

davidrapan, 2024-08-22:
@Steinbichler. Yeah I know but still it has nothing to do with the integration. Your device is still sending the same data for 5 minutes... What about update interval in their app? Did that change?

[55 more replies]