davidrapan/ha-solarman #86: Error setting up entry Sofar for solarman

manuel833, 2024-08-16:
Logger: homeassistant.config_entries
Quelle: config_entries.py:604
Erstmals aufgetreten: 14:56:07 (3 Vorkommnisse)
Zuletzt protokolliert: 14:57:30

Error setting up entry Sofar for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 72, in async_setup_entry
    await inverter.load(name, inverter_mac, lookup_path, lookup_file)
  File "/config/custom_components/solarman/api.py", line 80, in load
    self.profile = ParameterParser(self.parameter_definition)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/parser.py", line 51, in __init__
    self._code = self._registers_table[0]
                 ~~~~~~~~~~~~~~~~~~~~~^^^
KeyError: 0

davidrapan, 2024-08-16:
What definition profile it is?

manuel833, 2024-08-16:
I have update all new files 
[home-assistant_2024-08-16T14-33-54.877Z.log](https://github.com/user-attachments/files/16638230/home-assistant_2024-08-16T14-33-54.877Z.log)

davidrapan, 2024-08-16:
You proly did not otherwise this would not happen:
```
NameError: name 'is_platform' is not defined
```
That method is located in the common.py

davidrapan, 2024-08-16:
I suggest to you to completely replace content of the custom_components/solarman folder.

davidrapan, 2024-08-16:
[v24.08.16](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.16) beta (pre-release) fixes the issue.

closed 2024-08-19 by davidrapan