davidrapan/ha-solarman #87: no values by disconnect

CrazyUs3r, 2024-08-16:
Can you change something so that when you disconnect it will continue to be displayed.

davidrapan, 2024-08-16:
But not everyone wants that so I'm gonna have to think about it. 😉

CrazyUs3r, 2024-08-16:
Here you  can see what me Problem is to 0 go it from the inverter this not direct a problem from the py

[image]

CrazyUs3r, 2024-08-16:
2024-08-16 20:36:56.897 ERROR (MainThread) [custom_components.solarman.coordinator] Timeout fetching solarmanm80 data
2024-08-16 20:39:58.335 ERROR (MainThread) [custom_components.solarman.coordinator] Error fetching solarmanm80 data: [3907049102] Querying 192.168.178.226:8899 failed.
2024-08-16 20:48:31.958 WARNING (MainThread) [custom_components.solarman.api] [3907049102] Querying (0 - 112) failed. #0 [NoSocketAvailableError: Cannot open connection to 192.168.178.226]
2024-08-16 20:48:32.158 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Querying attempts left: 4
2024-08-16 20:48:32.158 INFO (MainThread) [custom_components.solarman.api] [3907049102] Connecting to 192.168.178.226:8899
2024-08-16 20:48:35.221 WARNING (MainThread) [custom_components.solarman.api] [3907049102] Querying (0 - 112) failed. #0 [NoSocketAvailableError: Cannot open connection to 192.168.178.226]
2024-08-16 20:48:35.623 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Querying attempts left: 3
2024-08-16 20:48:35.623 INFO (MainThread) [custom_components.solarman.api] [3907049102] Connecting to 192.168.178.226:8899
2024-08-16 20:48:38.678 WARNING (MainThread) [custom_components.solarman.api] [3907049102] Querying (0 - 112) failed. #0 [NoSocketAvailableError: Cannot open connection to 192.168.178.226]
2024-08-16 20:48:39.278 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Querying attempts left: 2
2024-08-16 20:48:39.278 INFO (MainThread) [custom_components.solarman.api] [3907049102] Connecting to 192.168.178.226:8899
2024-08-16 20:48:42.390 WARNING (MainThread) [custom_components.solarman.api] [3907049102] Querying (0 - 112) failed. #0 [NoSocketAvailableError: Cannot open connection to 192.168.178.226]
2024-08-16 20:48:43.191 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Querying attempts left: 1
2024-08-16 20:48:43.191 INFO (MainThread) [custom_components.solarman.api] [3907049102] Connecting to 192.168.178.226:8899
2024-08-16 20:48:46.293 WARNING (MainThread) [custom_components.solarman.api] [3907049102] Querying (0 - 112) failed. #0 [NoSocketAvailableError: Cannot open connection to 192.168.178.226]
2024-08-16 20:48:47.295 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Querying attempts left: 0, aborting.
2024-08-16 20:48:47.295 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Request failed. [Previous State: Disconnected (-1)]
2024-08-16 20:48:47.295 INFO (MainThread) [custom_components.solarman.api] [3907049102] Disconnecting from 192.168.178.226:8899
2024-08-16 20:48:51.895 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Scheduling 1 query request. #0
2024-08-16 20:48:51.896 DEBUG (MainThread) [custom_components.solarman.api] [3907049102] Querying (0 - 112) ...

davidrapan, 2024-08-16:
What am I looking at? Is that the time when your inverter goes offline?

CrazyUs3r, 2024-08-17:
Yes the gaps or the 0 values ​​are during the night

gedger, 2024-08-17:
I have had intermittent problems with connecting but my sensors go unavailable which is the correct state, HA displays it as a gap. 

I discovered that the wireless stick was randomly connecting to another AP with a lower signal level which caused the connection issues. I solved the problem by creating a dedicated SSID on the nearest AP and then connecting to that. I still get occasional connection problems but have always found that powering off/on the Wi-Fi router and stick has solved them. I don't think the stick hardware is the best....

If the connection problem can't be solved by the integration then you can create a template sensor that checks the value and only updates when the value is valid i.e. available and not zero

davidrapan, 2024-08-17:
I mean I could add configuration item which you toggle when you don'T want to loose the values after dc. BUT, I'm not sure if it won't bring more issues than it solves....

gedger, 2024-08-18:
I have tried blocking the connection to my inverter many times and on all occasions I get a blank in the graph where the sensor went unavailable. None of the sensors went to Zero so I can't replicate the issue @CrazyUs3r is having on my system.

davidrapan, 2024-08-18:
As far as I know this is happening only on some models which are using profiles deye_2mppt/4mppt and deye_string. So it's a "device specific thing" and it's complete BS!!!! But we were able to (thanks to our "amazing" integration 😆) mitigate it by applying validation (has to be > 0) on the values which were causing issues. It think @CrazyUs3r was able to figure it out first. 👏

CrazyUs3r, 2024-08-18:
@gedger What you mean is that you want to block your inverter from the internet. When I block mine, the values ​​are also only there, although I think the inverter sends the data to the Chinese server, but if it can't get there, it doesn't have the current time and just shows me nonsense.

davidrapan, 2024-08-18:
Ho you are blocking it? Removing info about the server (Server A Setting, maybe also Optional Server Setting) on this page: [http://{inverter_ip}/config_hide.html](http://{inverter_ip}/config_hide.html) is not enough?

CrazyUs3r, 2024-08-18:
First I blocked the connection in the fritzbox and later in adguard to see both variants. I had not yet deleted the connection under config_hiden

davidrapan, 2024-08-18:
If you decide to try it let me know the result!

CrazyUs3r, 2024-08-18:
If I lock everything then the only problem is that the device time is wrong, then the daily values, for example, are not reset but reading and writing the values ​​still works.

PV-Joe, 2024-08-18:
I would be interested to be able to run the inverter completely local.

As far as I understand that is not possible cause of the time.

CrazyUs3r, 2024-08-18:
I think I'll try the docker

https://github.com/Hypfer/deye-microinverter-cloud-free/tree/master/dummycloud

CrazyUs3r, 2024-08-18:
I've just tested it locally but the docker doesn't offer any option to set the time, so the inverter gets the GMT time but I can see that the inverter gets its response to changing the time.

gedger, 2024-08-18:
Can't you write to the date/time registers via Solarman, that works for me.

CrazyUs3r, 2024-08-18:
No. Your time does not fit the time setting because low year high hour and so on in the pdf

VoyteckPL, 2024-08-19:
+1 for values visible when disconnected. Even optionally. I would rather have an option to choose if I want to see 0 or unknown.

gedger, 2024-08-19:
> No. Your time does not fit the time setting because low year high hour and so on in the pdf

I thought the Solarman stick was completely optional for inverters, well it is for Solis. I operated mine without the stick plugged in and instead used a direct RS485 connection to read/write all the values including setting the date and time. I can still do the same when the Solarman stick is plugged in. I inky have the stick as Solis couldn't offer support without the remote connection.

>+1 for values visible when disconnected. Even optionally. I would rather have an option to choose if I want to see 0 or unknown.

So if your inverter goes off-line or because there is a temporary connection problem then you want all the values to go to zero! You'll end up with the graph shown when this issue was raised which, in my view, is highly undesirable. Also what about monthly totals should they go to zero too? A alternative solution is to leave the values at whatever the last read value was. However, then all the values are in incorrect as they no longer reflect the true state of the inverter. That's why HA has the unavailable state, I can't connect to the inverter so I can't tell you the value hence it marks it as unavailable.

If anyone wishes for this behaviour then it can easily be achieved by using template sensors. The value can then be set to whatever you like. So my vote is to leave as is.

VoyteckPL, 2024-08-19:
Ok maybe you are right! :) The problem is that I use **current production sensor** to calculate different things like excess power etc.
I managed to create sensor which basically works like this - if unknown then = 0.
Other sensors I don't actually use so this is not a big problem for me.

gedger, 2024-08-19:
I think each user and each application may have a slightly different requirement so template sensors seem the way to go.

Another idea would be to add a new option to the validation. Min, Max, "Unavailable":value ?

VoyteckPL, 2024-08-19:
I think some sensors have 0 during night so it could be shown as fake 0. :)

gedger, 2024-08-19:
I do hope my grid voltage stays above zero though 😆

davidrapan, 2024-08-19:
> I do hope my grid voltage stays above zero though 😆

😆 😆 😆 A good one. Fking yes! 😆

davidrapan, 2024-08-19:
I have this strong feeling that we will mess more things up with this than we solve.. If there is anything to solve in the first place, imo.

CrazyUs3r, 2024-08-23:
I have ask chatgpt he say the coordinator need samthing like this. @davidrapan can you look

`class SolarmanDataUpdateCoordinator(DataUpdateCoordinator):
    def __init__(self, hass, api):
        super().__init__(hass, _LOGGER, name="solarman", update_interval=timedelta(minutes=1))
        self.api = api
        self.last_data = None

    async def _async_update_data(self):
        try:
            data = await self.api.get_data()
            if data:
                self.last_data = data
            return self.last_data
        except Exception as err:
            _LOGGER.warning("Error fetching data: %s", err)
            return self.last_data
`

gedger, 2024-08-23:
I'm sorry but this makes no sense to me as you'll be reporting incorrect values! When the inverter is not connected then the integration has no idea what the true value is so it should be displayed as unavailable. If you display the last value then you will have to check that the inverter is connected to determine whether the sensor value being shown is actually correct which is surely is the wrong way round?

gedger, 2024-08-23:
You can use a template sensor to achieve this.

https://community.home-assistant.io/t/support-retrieving-the-last-known-value-in-templates-for-unavailable-sensor-entities/663710

gedger, 2024-08-23:
Thinking about this it sounds like you should bet trying to work out why the inverter is off-line so much. When I had an issue with a device I setup a command line sensor to regularly run nmap to determine the state of the device. You need to change the IP address to match your inverter.

You get 3 states returned.

3 - Inverter is up and port 8899 is open
2 - Inverter is up but port 8899 is not open
1 - Host not available

```
command_line:
  - sensor:
      name: Inverter Network Status
      command: "nmap -p 8899 192.168.1.40 | grep -cE 'Starting|8899/tcp open|Host is up'"
      scan_interval: 60   # every 60 seconds
      value_template: "{{ value }}"
```

mhoogenbosch, 2024-08-28:
I too have this request. And while I do understand everything above I think it makes sense for a few sensors to keep the value. For example 

The inverter total prodcution:
[image]

there is a gap when the inverter is offline, which is understandable but the value goes to unavaiable which is a weird view on a dashboard. 

for the total production and daily production I therefor have created the following:

```
- trigger:
    - platform: state
      entity_id: sensor.inverter_today_production
      not_to:
        - unknown
        - unavailable
  sensor:
    name: "SofarSolar_vandaag_zero"
    state: '{{ trigger.to_state.state }}'
    unique_id: "SofarSolar_vandaag_zero"
    unit_of_measurement: "kWh"
    state_class: "total"
    device_class: "Energy"

- trigger:
    - platform: state
      entity_id: sensor.inverter_total_production
      not_to:
        - unknown
        - unavailable
  sensor:
    name: "SofarSolar_totaal_zero"
    state: '{{ trigger.to_state.state }}'
    unique_id: "SofarSolar_totaal_zero"
    unit_of_measurement: "kWh"
    state_class: "total_increasing"
    device_class: "Energy"
```
for the current active output I've created the following:

```
  - name: "SofarSolar_actueel_zero"
    state: "{{ states('sensor.inverter_output_active_power')|float(0) }}"
    unique_id: "SofarSolar_actueel_zero"
    unit_of_measurement: "W"
    state_class: "total"
    device_class: "Energy"
```

These are all in the template.yaml file and are only used for displaying on a dashboard, to make sure there is just a 0 instead of "unavailable".

davidrapan, 2024-08-28:
Maybe we could show unavailable only for `measurement` values...

But still it has so many POV not just that, "Yeah the value proly did not change during that offline period." But what if the inverer just got disconnected from the network? That means it will show again wrong value... But (another one) `total` wouldn't be that much wrong as `measurement` values...

**And many more... lol.**

CrazyUs3r, 2024-09-25:
Hey @davidrapan what you think for this coordinator Skript

```
from __future__ import annotations
import logging
from typing import Any
from homeassistant.core import HomeAssistant
from homeassistant.helpers.update_coordinator import DataUpdateCoordinator
from .const import *

_LOGGER = logging.getLogger(__name__)

class InverterCoordinator(DataUpdateCoordinator[dict[str, Any]]):
    def __init__(self, hass: HomeAssistant, inverter):
        super().__init__(hass, _LOGGER, name=inverter.name, update_interval=TIMINGS_UPDATE_INTERVAL, always_update=False)
        self.inverter = inverter
        self._counter = -1
        self._last_data = None  # Hinzugefügt: Variable zum Speichern der letzten Daten

    def _accounting(self) -> int:
        if self.last_update_success:
            self._counter += 1
        return int(self._counter * self._update_interval_seconds)

    async def _async_update_data(self) -> dict[str, Any]:
        try:
            # Daten vom Wechselrichter abrufen
            data = await self.inverter.get(self._accounting())
            self._last_data = data  # Die abgerufenen Daten speichern
            return data
        except Exception:
            # Wenn ein Fehler auftritt (z.B. der Wechselrichter ist offline)
            if self._last_data is not None:
                _LOGGER.warning("Wechselrichter offline, zeige letzte bekannte Daten an.")
                return self._last_data  # Letzte bekannte Daten zurückgeben
            else:
                self._counter = 0
                raise  # Wenn keine vorherigen Daten vorhanden sind, Fehler auslösen

    async def async_shutdown(self) -> None:
        _LOGGER.debug("async_shutdown")
        await super().async_shutdown()
        await self.inverter.shutdown()

```

davidrapan, 2024-09-25:
Hi @CrazyUs3r, it's not necessary to save the last state like this. The state is already there within the sensors data and is just hidden behind `Unavailable` label whenever there is disconnect from the device.

davidrapan, 2024-09-25:
> And we dont want to see this one info in all other 40 entities. For example why after sunshine "today production" is not available if today production is 10kwh etc.

@szyx22, That is true until 11:59 PM. The only sensors which are potential candidates for preserving on device disconnects are **Total \*** sensors.

davidrapan, 2024-09-29:
@CrazyUs3r, did you had the chance to try set time using new Date & Time entity?

CrazyUs3r, 2024-09-29:
@davidrapan I still have to configure a lot of things in the Python server, but reading and writing works with the current version. This morning there was a small bug with the wrong time, which was due to my server.

davidrapan, 2024-09-29:
What do you mean? I was talking just about configuration from the HA side w/ no server involved.

CrazyUs3r, 2024-09-29:
sorry yes it works from the ha side the inverter hit it

davidrapan, 2024-09-29:
So It could be used to set the time for example every time HA and inverter connects using automation, no? And it could potentially solve time sync issue completely in a cloudless environment? Or am I missing something?

CrazyUs3r, 2024-09-29:
For that part I don't think you've made a mistake or forgotten anything. Unfortunately it just has no effect on the data when the sun is gone

CrazyUs3r, 2024-09-29:
I'll test it tomorrow morning, switch on my local server on the inverter and disconnect it completely from the network, then I'll let you know tomorrow what happens.

Laho812, 2024-09-30:
> As I mentioned before the **Total** * sensors could be made persistent during inverter disconnects. Is that enough for you?

I would love to see this! 

Writing date & time would be great as well. I blocked internet connection for my inverter so it won't talk to anything outside my network. Sadly the daily production values are not resetting anymore.
Besides that is it possible to make the daily production sensors persistent as well and reset them to zero at 0:00am ?

My model: Deye SUN-M200G4-EU-Q0

davidrapan, 2024-10-01:
> I would love to see this!

Okay then,

> Writing date & time would be great as well. I blocked internet connection for my inverter so it won't talk to anything outside my network. Sadly the daily production values are not resetting anymore.

That is already possible.

> Besides that is it possible to make the daily production sensors persistent as well and reset them to zero at 0:00am ?

That is somewhat out of scope of the integration but you can certainly do it yourself using utility meters.

gedger, 2024-10-01:
I still think this is the wrong approach, the whole point of the unavailable state is to say that the sensor is offline and its state is unknown. Even the total sensors are questionable, if the last state of the day_ total is kept then it will be wrong from midnight until the inverter comes back online. The monthly_total will also be wrong but only 1 day per month at midnight.

If people want to display the old state and possibly reset them at midnight then HA has all the tools to do this by using template sensors and utility meters. HA also has the conditional option on most of the UI components now so you can display something meaningful on the dashboard rather than unavailable. 

Perhaps a better option is for all sensors to have an attribute "last_state" so the sensor can go unavailable but the last_state still found?

davidrapan, 2024-10-01:
> if the last state of the day_ total is kept then it will be wrong from midnight until the inverter comes back online. The monthly_total will also be wrong but only 1 day per month at midnight.

Everytime I use term **Total \*** i realy mean total, not today or daily not even monthly. 😉

So that's why I said that **Today \*** sensors are out of scope.

davidrapan, 2024-10-01:
Date & Time entity was transferred from a sensor under proper `datetime` platform and should be in "Configuration" section.

Laho812, 2024-10-03:
> Date & Time entity was transferred from a sensor under proper datetime platform and should be in "Configuration" section.

You mean this entity listet in the "Configuration" category, right:
[image]
In my case `datetime.bkw1_date_time`

It was never available since I set up the integration.

davidrapan, 2024-10-03:
Hmmm that's weird.. could you show me debug log?

CrazyUs3r, 2024-10-03:
At my its running [image]

Laho812, 2024-10-03:
In my systemlog are two messages:
```
Logger: homeassistant.components.datetime
Quelle: helpers/entity_platform.py:595
Integration: Datum/Uhrzeit (Dokumentation, Probleme)
Erstmals aufgetreten: 12:57:55 (4 Vorkommnisse)
Zuletzt protokolliert: 13:00:31

Error adding entity datetime.bkw1_date_time for domain datetime with platform solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 595, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 909, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1366, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/datetime/__init__.py", line 104, in state
    if (value := self.native_value) is None:
                 ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/functools.py", line 995, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/datetime.py", line 60, in native_value
    return self._attr_native_value.replace(tzinfo = ZoneInfo(self.coordinator.hass.config.time_zone))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: str.replace() takes no keyword arguments
```
and this one:

```
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: homeassistant
Quelle: custom_components/solarman/datetime.py:60
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 12:58:01 (15 Vorkommnisse)
Zuletzt protokolliert: 13:00:41

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 44, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/datetime/__init__.py", line 104, in state
    if (value := self.native_value) is None:
                 ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/functools.py", line 995, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/datetime.py", line 60, in native_value
    return self._attr_native_value.replace(tzinfo = ZoneInfo(self.coordinator.hass.config.time_zone))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: str.replace() takes no keyword arguments
```

I don't know how i can create debug log of solarman only. When enableing it in the device view, I get a log of all my integrations.

davidrapan, 2024-10-03:
It looks like you don't have properly set time zone. Or have some issue with it.

Laho812, 2024-10-03:
Don't know what i can do wrong with the time zone :D I can just set it in Settings -> General

davidrapan, 2024-10-03:
Me neither honestly... But if you could discover what's stored in `self.coordinator.hass.config.time_zone` in your case maybe it will help.

Dubists, 2024-10-09:
[image]

Like Laho812  I'm also not able to set datetime, the field is greyed out. (Deye Sun 1600 G4, 4 MPPT Profile).
Timezone is set in configuration.yaml with coordinates.

CrazyUs3r, 2024-10-09:
Please share the debug log

Laho812, 2024-10-09:
> Me neither honestly... But if you could discover what's stored in `self.coordinator.hass.config.time_zone` in your case maybe it will help.

Can you tell me how to check the value this variable?

CrazyUs3r, 2024-10-09:
@davidrapan in the last 2 day is at night the same All states a store in ha

[image]

Dubists, 2024-10-10:
> Please share the debug log

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: homeassistant
Quelle: custom_components/solarman/datetime.py:63
Integration: Solarman (Dokumentation, Probleme)
Erstmals aufgetreten: 11:28:06 (50 Vorkommnisse)
Zuletzt protokolliert: 11:35:38

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solarman/datetime.py", line 61, in native_value
    return datetime.strptime(self._attr_native_value, DATETIME_FORMAT).replace(tzinfo = ZoneInfo(self.coordinator.hass.config.time_zone))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 554, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 333, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '48/0/0 0:0:0' does not match format '%y/%m/%d %H:%M:%S'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 453, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in async_update_listeners
    update_callback()
  File "/config/custom_components/solarman/entity.py", line 43, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/datetime/__init__.py", line 104, in state
    if (value := self.native_value) is None:
                 ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/datetime.py", line 63, in native_value
    _LOGGER.debug(f"SolarmanDateTimeEntity.native_value: {format_exception(e)}")
                                                                           ^
NameError: name 'e' is not defined

Timezone is set in the configuration.yaml. 
I am able to set the datetime with this Integration on my two Deye 12k, but not on the 1600 G4. Maybe the registers are incorrect?

[50 more replies]

closed 2025-01-19 by CrazyUs3r