davidrapan/ha-solarman #89: v24.08.14 - Failed to Setup

cvroque, 2024-08-16:
**Describe the bug**
After updating from v24.08.04 to v24.08.14 the integrations fails to setup. Removing and adding again didn't help.

**Attach the debug log**
```
2024-08-16 14:51:01.963 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Solis for solarman
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solarman/__init__.py", line 72, in async_setup_entry
await inverter.load(name, inverter_mac, lookup_path, lookup_file)
File "/config/custom_components/solarman/api.py", line 80, in load
self.profile = ParameterParser(self.parameter_definition)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solarman/parser.py", line 51, in __init__
self._code = self._registers_table[0]
~~~~~~~~~~~~~~~~~~~~~^^^
KeyError: 0
2024-08-16 14:52:50.190 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-08-16T17:51:00.740913+00:00', 'data': {'name': 'Solis', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'solis_3p-4g.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J5E50VM492P99NJ0KS3NY55Z', 'minor_version': 1, 'modified_at': '2024-08-16T17:51:00.740917+00:00', 'options': {'name': 'Solis', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'solis_3p-4g.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Solis', 'unique_id': None, 'version': 1})
2024-08-16 14:52:50.193 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-08-16 14:52:50.310 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.68.124, 34EAE7EA644A, 4089846851]
2024-08-16 14:52:51.405 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 5, code: 4, min_span: 25, digits: 6
2024-08-16 14:52:51.406 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Solis for solarman
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solarman/__init__.py", line 72, in async_setup_entry
await inverter.load(name, inverter_mac, lookup_path, lookup_file)
File "/config/custom_components/solarman/api.py", line 80, in load
self.profile = ParameterParser(self.parameter_definition)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solarman/parser.py", line 51, in __init__
self._code = self._registers_table[0]
~~~~~~~~~~~~~~~~~~~~~^^^
KeyError: 0
2024-08-16 14:53:00.752 DEBUG (MainThread) [custom_components.solarman.config_flow] ConfigFlowHandler.async_step_user: None
2024-08-16 14:53:00.753 DEBUG (MainThread) [custom_components.solarman.config_flow] step_user_data_process: discovery: <custom_components.solarman.discovery.InverterDiscovery object at 0x7fccaf1a3cb0>
2024-08-16 14:53:00.753 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-08-16 14:53:00.753 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all
2024-08-16 14:53:00.753 DEBUG (MainThread) [custom_components.solarman.discovery] _discover_all: Broadcasting on 192.168.68.0/24
2024-08-16 14:53:00.856 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.68.124, 34EAE7EA644A, 4089846851]
2024-08-16 14:53:01.871 DEBUG (MainThread) [custom_components.solarman.config_flow] step_user_data_schema: data: {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, custom_components/solarman/inverter_definitions/: ['afore_BNTxxxKTL-2mppt.yaml', 'afore_hybrid.yaml', 'deye_2mppt.yaml', 'deye_4mppt.yaml', 'deye_hybrid.yaml', 'deye_sg01hp3.yaml', 'deye_sg04lp3.yaml', 'deye_string.yaml', 'kstar_hybrid.yaml', 'pylontech_Force-H.yaml', 'sofar_g3hyd.yaml', 'sofar_hyd3k-6k-es.yaml', 'sofar_lsw3.yaml', 'sofar_wifikit.yaml', 'solarman_dtsd422-d3.yaml', 'solis_1p8k-5g.yaml', 'solis_3p-4g.yaml', 'solis_3p-5g.yaml', 'solis_hybrid.yaml', 'solis_s6-gr1p.yaml', 'zcs_azzurro-hyd-zss-hp.yaml', 'zcs_azzurro-ktl-v3.yaml']
2024-08-16 14:53:01.872 DEBUG (MainThread) [custom_components.solarman.config_flow] step_user_data_schema: STEP_USER_DATA_SCHEMA: {'name': <class 'str'>, 'inverter_discovery': <class 'bool'>, 'inverter_host': <class 'str'>, 'inverter_serial': <class 'int'>, 'inverter_port': <class 'int'>, 'inverter_mb_slave_id': <class 'int'>, 'inverter_passthrough': <class 'bool'>, 'lookup_file': In(['afore_BNTxxxKTL-2mppt.yaml', 'afore_hybrid.yaml', 'deye_2mppt.yaml', 'deye_4mppt.yaml', 'deye_hybrid.yaml', 'deye_sg01hp3.yaml', 'deye_sg04lp3.yaml', 'deye_string.yaml', 'kstar_hybrid.yaml', 'pylontech_Force-H.yaml', 'sofar_g3hyd.yaml', 'sofar_hyd3k-6k-es.yaml', 'sofar_lsw3.yaml', 'sofar_wifikit.yaml', 'solarman_dtsd422-d3.yaml', 'solis_1p8k-5g.yaml', 'solis_3p-4g.yaml', 'solis_3p-5g.yaml', 'solis_hybrid.yaml', 'solis_s6-gr1p.yaml', 'zcs_azzurro-hyd-zss-hp.yaml', 'zcs_azzurro-ktl-v3.yaml']), 'battery_nominal_voltage': <class 'int'>, 'battery_life_cycle_rating': <class 'int'>}
2024-08-16 14:53:09.165 DEBUG (MainThread) [custom_components.solarman.config_flow] ConfigFlowHandler.async_step_user: {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'solis_3p-4g.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}
2024-08-16 14:53:09.165 DEBUG (MainThread) [custom_components.solarman.config_flow] validate_input: {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'solis_3p-4g.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}
2024-08-16 14:53:09.165 DEBUG (MainThread) [custom_components.solarman.config_flow] ConfigFlowHandler.async_step_user: validation passed: {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'solis_3p-4g.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}
2024-08-16 14:53:09.168 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-08-16T17:53:09.166137+00:00', 'data': {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'solis_3p-4g.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J5E54S1ECKYWE27PQYWE9DQ6', 'minor_version': 1, 'modified_at': '2024-08-16T17:53:09.166144+00:00', 'options': {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.68.124', 'inverter_serial': 4089846851, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'solis_3p-4g.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Inverter', 'unique_id': None, 'version': 1})
2024-08-16 14:53:09.168 DEBUG (MainThread) [custom_components.solarman.discovery] discover
2024-08-16 14:53:09.173 DEBUG (MainThread) [custom_components.solarman.discovery] _discover: [192.168.68.124, 34EAE7EA644A, 4089846851]
[... cut ...] d422-d3.yaml', 'solis_1p8k-5g.yaml', 'solis_3p-4g.yaml', 'solis_3p-5g.yaml', 'solis_hybrid.yaml', 'solis_s6-gr1p.yaml', 'zcs_azzurro-hyd-zss-hp.yaml', 'zcs_azzurro-ktl-v3.yaml']), 'battery_nominal_voltage': <class 'int'>, 'battery_life_cycle_rating': <class 'int'>}

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Add Device
2. Select 'solis_3p-4g.yaml'
3. Press submit
4. See error

**Metadata:**
Version:  v24.08.14

**Additional context**
Rolling back to v24.08.04 works normally.

davidrapan, 2024-08-16:
Duplicate of #86

davidrapan, 2024-08-16:
Hi @cvroque, thanks for the report, it's already solved [fix: ParameterParser KeyError during init](https://github.com/davidrapan/ha-solarman/commit/75a6809fcc17b1310865c6a1b0f6bb6ffe7da856) and will be part of tonight's beta release. 😉

davidrapan, 2024-08-16:
[v24.08.16](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.16) beta (pre-release) fixes the issue.

Baloo42300, 2024-08-17:
Hello for me too isn't run 😔 please help

davidrapan, 2024-08-17:
Hi @Baloo42300, enable beta releases on the HACS page of the integration in top right corner menu under REDOWNLOAD. And REDOWNLOAD latest v24.08.16 version. 😉

davidrapan, 2024-08-17:
> I no longer have access to my devices since the update.

I don't understand. What do you mean by that? You really have to be more specific.

Baloo42300, 2024-08-17:
@davidrapan since the update 1 hour ago. my devices have been disconnected and are no longer present in my table

davidrapan, 2024-08-17:
@Baloo42300

What version you have installed?

**How to nebale debug:**
In the Home Assistant go to settings, devices, solarman and then you will see:
[image]
Then you let it run for some time and click on that button again and it will download a log file.

Baloo42300, 2024-08-17:
Now i have version 24.08.04 and when i  install the latest version, my devices and entities disappear, I have nothing left.. I managed to reinstall the old one and everything works again as before

Baloo42300, 2024-08-17:
I started the log, how long should I leave it?

davidrapan, 2024-08-17:
@Baloo42300 Try [v24.08.17](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.17) and see if it won't fix your issue.

> I started the log, how long should I leave it?

Couple of minutes.

cvroque, 2024-08-19:
Just confirming that [v24.08.17](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.17) fixed the issue. 

Sorry, I didn't see the other issue when opening mine. Thanks a lot!

Baloo42300, 2024-08-19:
Sorry @davidrapan i haven't confirmed to you but yes version v24.08.17 has solved the problem. Everything is working normally again now thank you very much 👌🏾👌🏾

closed 2025-02-03 by davidrapan