davidrapan/ha-solarman #91: The sensors do not update automatically

manuel833, 2024-08-17:
The sensors only update when the integration is reloaded.

[image]

davidrapan, 2024-08-17:
Hi @manuel833, didn't you accidentally turn of the polling?

**Anyway I will really need debug logs.**

manuel833, 2024-08-17:
Where can I find polling?
[home-assistant_solarman_2024-08-17T17-03-14.083Z.log](https://github.com/user-attachments/files/16645722/home-assistant_solarman_2024-08-17T17-03-14.083Z.log)

davidrapan, 2024-08-17:
When you click on the 3 dots next to the device and then under System options.

manuel833, 2024-08-17:
sofar hyd 10ktl-3ph
sofar_g3hyd.yaml 

Solarman 24.08.16
Core: 2024.8.2
Supervisor: 2024.08.0
Operating System: 13.0

davidrapan, 2024-08-17:
Hey @manuel833,

Open sofar_g3hyd.yaml profile file and on the 15 row you will find:
```
        class: ""
```
Change it to:
```
        class: "enum"
        options: ["waiting", "detection", "grid-connected", "emergency power supply", "recoverable fault", "permanent fault", "upgrade", "self-charging"]
```
**Notice the added row.**

And let me know if it helps. 😉

davidrapan, 2024-08-17:
Oh I just noticed it also has:
```
        state_class: "measurement"
```
If you will still have issues change it to:
```
        state_class: ""
```

manuel833, 2024-08-17:
The Solarman Inverter status is now working and the sensors are also functioning. Thank you.

closed 2024-08-17 by manuel833

davidrapan, 2024-08-17:
[v24.08.17](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.17) also fixes the issue. 😉