davidrapan/ha-solarman #94: Update of data not working

ballonchef, 2024-08-17:
**Describe the bug**
I moved from solarman to ha-solarman. i can retrieve data from my inverter, but the data does not update automatically. Only after i press "reload" in the intregration manually.
With the solarman integration, the automatic update worked well.

**Attach the debug log**
I don't know how to insert a debug log.

**To Reproduce**
-

**Expected behavior**
Data should automatically update every 10 seconds

**Screenshots**
[image]

Here you can see that the sensor does not update automatically, only when I reload manually.
**Metadata:**
Version: [e.g. v24.08.14]

davidrapan, 2024-08-17:
Duplicate of #91

davidrapan, 2024-08-17:
Hi @ballonchef Try update to [v24.08.17](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.17) and see if it won't fix your issue.

davidrapan, 2024-08-17:
If not here is how to enable debug logging:
[image]

ballonchef, 2024-08-17:
Thanks for the very quick reply!! I'll update and report.

ballonchef, 2024-08-17:
The update to 

> Hi @ballonchef Try update to [v24.08.17](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.17) and see if it won't fix your issue.

The update to 24.08.17 solved the issue. Thanks a lot, again!!

davidrapan, 2024-08-17:
Excellent! You're very welcome!

PV-Joe, 2024-08-17:
@ballonchef

Nice to see you over here.
Hope you are happy with the fast update interval and the other advantages of this integration.

Happy monitoring ☀️

closed 2025-02-03 by davidrapan