davidrapan/ha-solarman #100: v24.08.17 - Failed to Setup

Mikkao, 2024-08-18:
**Describe the bug**
Connection to inverter fails. Log says "V5_SEQ_NO_MISMATCH".
192.168.178.30 and 192.168.178.66 are WLAN repeaters. 192.168.178.80:8899 is the LAN connection to the Deye.

**Attach the debug log**

```
2024-08-18 22:42:41.773 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-18 22:42:41.775 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-18 22:42:46.316 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/frontend", "/config/custom_components/hacs/hacs_frontend_experimental", False)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 49: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-18 22:42:46.324 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/iconset.js", "/config/custom_components/hacs/iconset.js", True)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 57: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-18 22:42:46.331 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-18 22:42:50.881 ERROR (SyncWorker_0) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.66', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f59908470>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:42:50.883 ERROR (SyncWorker_3) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.30', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f598db9b0>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:43:19.205 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.helpers.event. This is deprecated and will stop working in Home Assistant 2024.11, it should be updated to import functions used from event directly at custom_components/hacs/base.py, line 642: self.hass.helpers.event.async_track_time_interval(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-18 22:43:22.754 ERROR (SyncWorker_6) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.30', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f5978d550>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:43:22.755 ERROR (SyncWorker_7) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.66', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f57e04050>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:43:36.065 ERROR (SyncWorker_2) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.66', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f56afcfb0>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:43:36.070 ERROR (SyncWorker_7) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.30', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f56afdfa0>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:43:39.664 WARNING (MainThread) [custom_components.solarman.api] [0123456789] Querying (0 - 22) failed. #0 [TimeoutError]
2024-08-18 22:43:39.768 DEBUG (MainThread) [custom_components.solarman.api] [0123456789] V5_SEQ_NO_MISMATCH: a5 17 00 10 45 03 00 b3 02
2024-08-18 22:43:39.865 DEBUG (MainThread) [custom_components.solarman.api] [0123456789] Querying attempts left: 4
2024-08-18 22:43:39.870 DEBUG (MainThread) [custom_components.solarman.api] [0123456789] Successful reconnect
2024-08-18 22:43:39.874 DEBUG (MainThread) [custom_components.solarman.api] [0123456789] SENT: a5 17 00 10 45 f2 00 b3 2f 45 89 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 00 00 17 05 c4 f4 15
2024-08-18 22:43:40.984 DEBUG (MainThread) [custom_components.solarman.api] [0123456789] V5_SEQ_NO_MISMATCH: a5 17 00 10 45 03 00 b3 02
2024-08-18 22:43:50.652 DEBUG (MainThread) [custom_components.solarman.api] [0123456789] Request failed. [Previous State: Disconnected (-1)]
2024-08-18 22:43:50.653 INFO (MainThread) [custom_components.solarman.api] [0123456789] Disconnecting from 192.168.178.80:8899
2024-08-18 22:43:50.658 DEBUG (MainThread) [custom_components.solarman.coordinator] async_shutdown
2024-08-18 22:43:50.658 INFO (MainThread) [custom_components.solarman.api] [0123456789] Disconnecting from 192.168.178.80:8899
2024-08-18 22:43:59.297 ERROR (SyncWorker_8) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.66', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f58b65bb0>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:43:59.553 ERROR (SyncWorker_9) [fritzconnection] Unable to get a connection: HTTPConnectionPool(host='192.168.178.30', port=49000): Max retries exceeded with url: /igddesc.xml (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f56ae6420>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2024-08-18 22:44:00.751 DEBUG (MainThread) [custom_components.solarman] async_setup_entry({'created_at': '2024-08-18T20:02:48.681810+00:00', 'data': {'name': 'Inverter', 'inverter_discovery': True, 'inverter_host': '192.168.178.80', 'inverter_serial': 2107021875, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'deye_hybrid.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'disabled_by': None, 'domain': 'solarman', 'entry_id': '01J5KHBM79A1H5S75CECYB4E5V', 'minor_version': 1, 'modified_at': '2024-08-18T20:24:50.892303+00:00', 'options': {'name': 'Inverter', 'inverter_discovery': False, 'inverter_host': '192.168.178.80', 'inverter_serial': 0123456789, 'inverter_port': 8899, 'inverter_mb_slave_id': 1, 'inverter_passthrough': False, 'lookup_file': 'deye_sg04lp3.yaml', 'battery_nominal_voltage': 48, 'battery_life_cycle_rating': 6000}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'title': 'Inverter', 'unique_id': None, 'version': 1})
2024-08-18 22:44:01.516 DEBUG (MainThread) [custom_components.solarman.parser] Defaults for update_interval: 60, code: 3, min_span: 25, digits: 6
2024-08-18 22:44:01.516 DEBUG (MainThread) [custom_components.solarman.api] {'identifiers': {('solarman', 0123456789)}, 'name': 'Inverter', 'manufacturer': 'Deye', 'model': 'SG04LP3', 'serial_number': 0123456789}
[... cut ...]
2024-08-18 22:46:33.713 INFO (MainThread) [custom_components.solarman.api] [0123456789] Disconnecting from 192.168.178.80:8899
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add Solarman in HACS
2. Configure (Auto filled in the SN of the dongle, not the inverter, corrected that)
3. Start
4. Message "Setup failed"

**Expected behavior**
Solarman connects to the Deye and fetches data

**Metadata:**
v24.08.17

**Additional context**
HA on raspberry 4, Deye SUN-12K-SG04LP3-EU, both connected via LAN to the PC over a FritzBox router.

davidrapan, 2024-08-18:
Hi @Mikkao, Try enable passthrough mode.

davidrapan, 2024-08-19:
Duplicate of #21

Mikkao, 2024-08-19:
Hello David, thank you so much for your fast answer, you are really committed to this project!
I had already tried enabling passthrough mode which didn't help.
Editing the api.py file didn't help, either; the line is 53 now rather than 32. I changed it to 23 whicht didn't help, either.
Would be great if you'd be successful in solving that problem. Tried to use my WiFi logger, but no avail due to weak signal.

davidrapan, 2024-08-19:
> the line is 53 now rather than 32

My bad. Anyway some number in range 0-25 should work...

Mikkao, 2024-08-19:
Looking forward to it!

PV-Joe, 2024-08-19:
Tried to use my WiFi logger, but no avail due to weak signal.

Wouldn’t it be easier to repeat the signal with a repeater or even an old router?
I have done that so the signal reaches the cellar where the inverter with LSW3 Stick-Logger is installed and it works flawlessly.

davidrapan, 2024-08-19:
@Mikkao, try this: [feat: Experimental testing of ethernet stick](https://github.com/davidrapan/ha-solarman/commit/5d44f3cfe0bc56e84997ffefab60c040b3aa7912)

davidrapan, 2024-08-19:
[v24.08.18](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.18)

davidrapan, 2024-08-21:
[v24.08.21](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.21)

closed 2025-02-03 by davidrapan