davidrapan/ha-solarman #95: 24.08.17 Connection Issue File __init__.py", line 70, in async_setup_entry

hackepeterOli, 2024-08-18:
Hi, 
i cannot establish a connection to my sg04lp3. I found the following in the log:

`2024-08-18 12:32:33.218 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-18 12:32:36.535 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Deye10kW for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 70, in async_setup_entry
    inverter = Inverter(inverter_host, inverter_serial, inverter_port, inverter_mb_slave_id, inverter_passthrough)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/api.py", line 62, in __init__
    super().__init__(address, serial, port, mb_slave_id, passthrough)
  File "/config/custom_components/solarman/api.py", line 24, in __init__
    super().__init__(address, serial, port = port, mb_slave_id = mb_slave_id, logger = _LOGGER, auto_reconnect = AUTO_RECONNECT, socket_timeout = TIMINGS_SOCKET_TIMEOUT)
  File "/usr/local/lib/python3.12/site-packages/pysolarmanv5/pysolarmanv5_async.py", line 65, in __init__
    self.data_wanted_ev = Event()
                          ^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 93, in Event
    return Event(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 331, in __init__
    self._cond = ctx.Condition(ctx.Lock())
                               ^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/context.py", line 68, in Lock
    return Lock(ctx=self.get_context())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 169, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/local/lib/python3.12/multiprocessing/synchronize.py", line 57, in __init__
    sl = self._semlock = _multiprocessing.SemLock(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system`

PV-Joe, 2024-08-18:
@hackepeterOli 

Hi there,
did you update to this integration?
Delete the old one first!

hackepeterOli, 2024-08-18:
Hi PV-Joe,

Yes, I have deleted the old one. Still no success.
What surprises me is that the input fields with IP address and serial number are always pre-filled when creating a new connection. Are there possibly old data from the old integration?

PV-Joe, 2024-08-18:
As far as I know it sets it up automatically.

Did you install the latest Version of HA and HACS?

hackepeterOli, 2024-08-18:
Yes, i think so:
HACS: 1.34.0
HA:
Core
2024.8.2
Supervisor
2024.08.0
Operating System
12.4
Frontend
20240809.0

PV-Joe, 2024-08-18:
Let’s wait for David. He will understand what’s going on.

As I said in the other thread, it works perfectly with my Deye SG04LP3 and is very stable and updating the most important values every 5 seconds.

davidrapan, 2024-08-18:
Duplicate of https://github.com/davidrapan/ha-solarman/issues/39

davidrapan, 2024-08-18:
In the thread above are provided two possible solutions.

hackepeterOli, 2024-08-18:
Perfect, Thanks!

ha core rebuild fix the problem.

PV-Joe, 2024-08-18:
@hackepeterOli 

That sounds great 👍

If you want to express yourself about the integration go here:

https://github.com/davidrapan/ha-solarman/discussions/57

closed 2025-02-03 by davidrapan