davidrapan/ha-solarman #97: Exception when loading new kstar_hybrid profile

frsantos, 2024-08-18:
**Describe the bug**
After updating to latest version 24.08.17, I get an error on the log, and sensor is not updated

**Attach the debug log**
```txt
2024-08-18 15:42:43.028 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.kstar_dsp_alarm_code for domain sensor with platform solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 662, in state
    numerical_value = float(value)  # type:ignore[arg-type]
                      ^^^^^^^^^^^^
TypeError: float() argument must be a string or a real number, not 'list'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 598, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 912, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1366, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 664, in state
    raise ValueError(
ValueError: Sensor sensor.kstar_dsp_alarm_code has device class 'None', state class 'measurement' unit 'None' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '['0x0', '0x0']' ()
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Metadata:**
Version: [e.g. v24.08.17]

**Additional context**
Add any other context about the problem here.

davidrapan, 2024-08-18:
Hey,

Try remove `state_class: "measurement"` from entities which are causing issues.

frsantos, 2024-08-18:
OK, working on removing the "measurement" on problematic entities...

closed 2024-08-18 by frsantos