davidrapan/ha-solarman #98: Moved from another solarman to for my Hybrid Solis

OneScuzzie2k, 2024-08-18:
I have moved from a old working addon, but this one will not work with my Solis Hybrid I am getting this

Logger: custom_components.solarman.api
Source: custom_components/solarman/api.py:206
integration: Solarman ([documentation](https://github.com/davidrapan/ha-solarman), [issues](https://github.com/davidrapan/ha-solarman/issues))
First occurred: 15:56:34 (34 occurrences)
Last logged: 15:59:40

    [4090370252] Querying (33029 - 33152) failed. #0 [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
    [4090370252] Querying (33029 - 33152) failed. #0 [IllegalDataAddressError: The data address received in the request is not an allowable address for the server. ]
    [4090370252] Querying (33022 - 33147) failed. #0 [IllegalDataValueError: The value contained in the request data field is not an allowable value for the server. ]

davidrapan, 2024-08-18:
Hey @OneScuzzie2k, those errors means that you are trying to read incorrect registers. What exactl model and definition profile are you using?

And from which version are you migrating?

OneScuzzie2k, 2024-08-18:
Hi David thanks for getting back to me so quickly, I am coming from Solaman v1.5.1. I am using a Solis Hybrid, this is what I also use on StephanJoubert one and it all seems to work great.

I tried a install over the top to keep all the names for my setup, and a uninstall of the old one and a new install of yours.

davidrapan, 2024-08-18:
So you used your own definition profile?

OneScuzzie2k, 2024-08-18:
I tried my own one that I used on the StephanJoubert one and your solis_hybrid.yaml neither work. I just get failed to setup

OneScuzzie2k, 2024-08-18:
I am rolled back to StephanJoubert to get it working works straigh away. Just putting your version back on just incase you want some more informtion

davidrapan, 2024-08-18:
Try to put in the profile file into default section:
```
  min_span: 1
```

OneScuzzie2k, 2024-08-18:
Same I am afraid 
[image]
[image]

OneScuzzie2k, 2024-08-18:
Its just clicked in and started working.... let me try my yaml

davidrapan, 2024-08-18:
Good. Can you then please test for me continuously increasing the value until it stops working again? And tell me the highest number you could put there?

OneScuzzie2k, 2024-08-18:
OK on it now....

davidrapan, 2024-08-18:
Anyway to explain what is happening. In our integration are requested registers decided by sensor descriptions. And somewhere in the range 33029-33152 are registers which can't be read from your device for some reason. And lowering the min_span from 25 will create more requests from that range and just single one and that's why it works. 😉

OneScuzzie2k, 2024-08-18:
Seems to have stopped all together now even with min_span set to 1 doh!

OneScuzzie2k, 2024-08-18:
Ignore that its kicked in, i thing anything other than 1 breaks it

OneScuzzie2k, 2024-08-18:
Think its gonna have to stay on 1!

davidrapan, 2024-08-18:
With lower values there is going to be clearly visible which registers causes the issues. So maybe you could look for them? If you want ofc.

OneScuzzie2k, 2024-08-18:
OK, writing back, can you tell me on these:

    - name: "Solis Timed Charge Start Hour"
      class: "current"
      state_class: "measurement"
      uom: ""
      scale: 1
      rule: 1
      registers: [43143]
      icon: 'mdi:clock'
    - name: "Solis Timed Charge Start Minute"
      class: "current"
      state_class: "measurement"
      uom: ""
      scale: 1
      rule: 1
      registers: [43144]
      icon: 'mdi:clock'
    - name: "Solis Timed Charge End Hour"
      class: "current"
      state_class: "measurement"
      uom: ""
      scale: 1
      rule: 1
      registers: [43145]
      icon: 'mdi:clock'
    - name: "Solis Timed Charge End Minute"
      class: "current"
      state_class: "measurement"
      uom: ""
      scale: 1
      rule: 1
      registers: [43146]
      icon: 'mdi:clock'

If i set 43142 to 3, 43143 to 31, 43144 to 6 and 43145 to 32 I get these figures? 
[image]

Sorry really greatful for your cleverness in doing this plugin, I would not have a clue, this is the other reason I needed to move as I cannot write times on demand to change charge on demand and dump to grid on demand.

OneScuzzie2k, 2024-08-18:
In matter of fact it does the same as the other driver does it shows in the solarman driver as these times, yet it does not change them at all, they do not match on the invertor.

OneScuzzie2k, 2024-08-18:
[image]

OneScuzzie2k, 2024-08-18:
Strange the Modbus reads something that is not set

davidrapan, 2024-08-18:
I'm doing it too with my device (Deye SG04LP3) but we have in the profile properly setup registers as writable and we can configure them without that service.

Maybe we could try to implement it for your device too hm?

OneScuzzie2k, 2024-08-18:
I am happy to try anything, I want to be able to change the times on the invertor to force a discharge to grid or to charge on demand, I wanted to be able to set the Discharge start hour to "{{ states('sensor.solarman_solis_hour')}}" and the discharge minute to "{{ states('sensor.solarman_solis_minute') | int + 2 }}" and reverse on charge on demand

davidrapan, 2024-08-18:
Technical question, how many "rows" or entries it has?

OneScuzzie2k, 2024-08-18:
I only use the 4 Charge and 4 dischard so as below, but I did read the extra time 2 and time 3 ones but not needed.

43142
    - name: "Solis Timed Charge Start Hour"
43143
    - name: "Solis Timed Charge Start Minute"
43144
    - name: "Solis Timed Charge End Hour"
43145
    - name: "Solis Timed Charge End Minute"

43146
    - name: "Solis Timed Discharge Start Hour"
43147
    - name: "Solis Timed Discharge Start Minute"
43148
    - name: "Solis Timed Discharge End Hour"
43149
    - name: "Solis Timed Discharge End Minute"

davidrapan, 2024-08-18:
Okay but when you change those values in the inverter, are they then properly displayed in the Home Assistant?

OneScuzzie2k, 2024-08-18:
Yes

OneScuzzie2k, 2024-08-18:
But If i change them on the write holding, the show on HA as incorrect values to what I set, but dont update the invertor

davidrapan, 2024-08-18:
And did you also tried the other service for writing?

davidrapan, 2024-08-18:
Anyway, try add into the profile this:
```
      - name: "Program 1 Time Start Hour"
        platform: number
        state_class: measurement
        uom: "H"
        scale: 1
        rule: 1
        registers: [43143]
        icon: "mdi:wrench-clock"
```
Restart the HA or reload the integration and then you should see in the device detail page configurable value.

OneScuzzie2k, 2024-08-18:
No as these are the only ones I need to change, I had the same experince with StephanJoubert Solarman. In mater of face I changed them using the Modbus, but they still display incorrect on the HA, yet different on the invertor still.

OneScuzzie2k, 2024-08-18:
[image]
Added let me reboot and try

davidrapan, 2024-08-18:
And when you change the value there try wait let's say at least 30 seconds before you check the inverter.

OneScuzzie2k, 2024-08-18:
Perfect I actually added:

    - name: "Program 1 Time Start Hours"
      platform: number
      state_class: measurement
      uom: "H"
      scale: 1
      rule: 1
      registers: [43142]
      icon: "mdi:wrench-clock"
      - name: "Program 1 Time Start Minutes"
      platform: number
      state_class: measurement
      uom: "H"
      scale: 1
      rule: 1
      registers: [43143]
      icon: "mdi:wrench-clock"

davidrapan, 2024-08-18:
You have a spacing typo before the second dash.

OneScuzzie2k, 2024-08-18:
Yeah noticed corrected before restart :)

davidrapan, 2024-08-18:
Another typo: Hour is: 43143, Minutes are: 43144

OneScuzzie2k, 2024-08-18:
Doh!

davidrapan, 2024-08-18:
I'll return back for a sec to the request problem best for you will be to set requests like this (instead of setting that min_span):
```
requests_fine_control:
requests:
  - start: 33022
    end:  33095
    mb_functioncode: 0x04
  - start: 33116
    end:  33179
    mb_functioncode: 0x04
  - start: 33206
    end:  33282
    mb_functioncode: 0x04
  - start: 43000
    end: 43150
    mb_functioncode: 0x03
```
Notice the `requests_fine_control:` it's right above and aligned with the `requests:` and it will ensure the requesting will follow wxactly what's set in the requests section. 😉 

And remove the `min_span: 1`

OneScuzzie2k, 2024-08-18:
Just trying the time setting via the write on the driver atm....

OneScuzzie2k, 2024-08-18:
Didnt work, the change show on the Modbus values but not the invertor. Set to start hour 2 start minute 31, but the invertor does not update. still shows as 2 and 30

OneScuzzie2k, 2024-08-18:
> I'll return back for a sec to the request problem best for you will be to set requests like this (instead of setting that min_span):
> 
> ```
> requests_fine_control:
> requests:
>   - start: 33022
>     end:  33095
>     mb_functioncode: 0x04
>   - start: 33116
>     end:  33179
>     mb_functioncode: 0x04
>   - start: 33206
>     end:  33282
>     mb_functioncode: 0x04
>   - start: 43000
>     end: 43150
>     mb_functioncode: 0x03
> ```
> 
> Notice the `requests_fine_control:` it's right above and aligned with the `requests:` and it will ensure the requesting will follow what's exactly set in the requests section. 😉
> 
> And remove the `min_span: 1`

This is what I had orginally on my old YAML

davidrapan, 2024-08-18:
With `requests_fine_control:` too? It's important.

> Didnt work, the change show on the Modbus values but not the invertor. Set to start hour 2 start minute 31, but the invertor does not update. still shows as 2 and 30

Show me. Also enable debug before you try to set it and then wait some time, try to set, wait some time. Disable debug and give the log file please.

OneScuzzie2k, 2024-08-18:
> With `requests_fine_control:` too? It's important.
>
>
> Show me. Also enable debug before you try to set it and then wait some time, try to set, wait some time. Disable debug and give the log file [...]

No just added this, let me try again....

OneScuzzie2k, 2024-08-18:
Adding this:
requests_fine_control:
requests:
  - start: 33022
    end:  33095
    mb_functioncode: 0x04
  - start: 33116
    end:  33179
    mb_functioncode: 0x04
  - start: 33206
    end:  33282
    mb_functioncode: 0x04
  - start: 43000
    end: 43150
    mb_functioncode: 0x03

has now stopped it working again....
[image]

davidrapan, 2024-08-18:
Try increse the first number: 33022 to 33029

davidrapan, 2024-08-18:
Oh wait a minute:
start: 43000
end: 43150
is also too long i think? It can only be 125 at max.
try to separate that one into 2

OneScuzzie2k, 2024-08-18:
> Try increase the first number: 33022 to 33029

Just rebooting, normally 33022 reads fine as its the invertor rime

davidrapan, 2024-08-18:
> Just rebooting, normally 33022 reads fine as its the invertor rime

Yeah I know I'm just guessing honestly. That long request is probably the issue, imo.

OneScuzzie2k, 2024-08-18:
- start: 43000
    end: 43150
    mb_functioncode: 0x03

davidrapan, 2024-08-18:
Maybe better will be to not to push that 125 limit and go 43000 - 43100.

When you are done with this show me then those new configurable items in the device view.

davidrapan, 2024-08-18:
Anyway according to sensors in the file you need only 43011 - 43025 and 43141 - 43150 anyway.

OneScuzzie2k, 2024-08-18:
start: 33022
    end: 33095
    mb_functioncode: 0x04
    start: 33116
    end: 33179
    mb_functioncode: 0x04
    start: 33206
    end: 33282
    mb_functioncode: 0x04
    start: 43011
    end: 43025
    mb_functioncode: 0x03
    start: 43141
    end: 43150 
    mb_functioncode: 0x03

OneScuzzie2k, 2024-08-18:
Last figures killed HA

davidrapan, 2024-08-18:
Didn't you forget the dashes? In front of every start?

OneScuzzie2k, 2024-08-18:
Ok backup, debuging enabled, setting the time to 6 hours and 15 Mins, let see what happens. I will upload debug

davidrapan, 2024-08-18:
Good. And show me the configurable items so I can check if it's all good and how it's supposed to be..

OneScuzzie2k, 2024-08-18:
Its updated Solarman figures again
[image]

but as before not changing invertor
[home-assistant_solarman_2024-08-18T19-11-46.689Z.log](https://github.com/user-attachments/files/16651373/home-assistant_solarman_2024-08-18T19-11-46.689Z.log)

davidrapan, 2024-08-18:
```
service_write_multiple_holding_registers: 43143, values: [6] failed, attempts left: 4. [error: unpack requires a buffer of 12 bytes]
```
:-/

OneScuzzie2k, 2024-08-18:
> ```
> service_write_multiple_holding_registers: 43143, values: [6] failed, attempts left: 4. [error: unpack requires a buffer of 12 bytes]
> ```
> 
> :-/

Means nothing to me! lol

davidrapan, 2024-08-18:
But wait it looks like that 43144 did go through...
service_write_multiple_holding_registers: 43144, response: 1

OneScuzzie2k, 2024-08-18:
Shows correct on HA but the invertor has not changed

[96 more replies]

closed 2024-08-30 by davidrapan