davidrapan/ha-solarman #101: HA keeps restarting when Integration starts up with no active inverter.

lenwar, 2024-08-19:
**Describe the bug**
I updated the Solarman integration yesterday late evening, while there was no sun and my inverters were 'powered down'/'offline'. After the update, HA obviously needed to restart.

HA kept restarting (kind of a 'boot loop' of HA), as long as the Solarman Integrations were active. During each restart I had a number of seconds (20-30 seconds or so), to disable the Solarman integrations. When there were disabled, HA kept working fine. When I enabled the integration (with the inverters still offline), HA kept restarting again.

**Attach the debug log**
(( not available yet - Will look in to soon - logging has already passed the rotation period. ))

**To Reproduce**
Steps to reproduce the behavior:
1. When the inverters are 'offline', restart Home Assistant.

**Expected behavior**
I expect Home Assistant to remain available/running after a restart.

**Screenshots**
(( N/A ))

**Metadata:**
Using all 'current' versions:

Solarman HACS version: v24.08.17
Home Assistant Core: 2024.8.2
Home Assistant OS: 13.0
Supervisor: 2024.08.0
Hardware type: Home Assistant Blue / Hardkernel ODROID-N2/N2+ (probably not relevant)

I'm using 5 dual-port Solarman/Deye micro inverters using the 2MPPT profile.
Firmware:   MW3_16U_5406_2.32-D1

**Additional context**
I 'think' I also had this problem with an earlier version, though I'm not entirely sure. I thought I could like the issue to another Integration (Dyson) back then.

davidrapan, 2024-08-19:
Hey @lenwar, thanks for your report. I will try to replicate it and then get back to you, promptly!

davidrapan, 2024-08-20:
Unfortunately I wasn't able to reproduce described issue and I will need more info about what's happening in your case. If you could maybe try to catch a log (debug one at best)?

lenwar, 2024-08-27:
Sorry for the late reply.

I haven't had the chance to generate the issue yet. (As you can imagine it's not practical to generate the issue as HA controls the light and a lot of other things in the house)

I'll try to disable the wifi of the inverters during the day soon.

closed 2025-12-24 by davidrapan