davidrapan/ha-solarman #103: Inverter Todays Production negative in the mornings

pschneider87, 2024-08-20:
When adding the Inverter Todays Production to the energy dashboard, everything works well over the day.
However, every day around 7am, I get a full negative spike for the previous days production.

[image]
[image]

Anyone knows what could cause this or how to solve it?

davidrapan, 2024-08-20:
You have to add into energy dashboard Total* sensors. not Today ones.

pschneider87, 2024-08-20:
I thought so too, but this will make the measurements even worse...

[image]

It will then add the total production each day in the morning, which makes all other measurements useless.

But I don't know if this is an issue with HA or this integration. I recognized with the new integration first

davidrapan, 2024-08-20:
No it's okay like this. It's cause it's a new sensor for the dashboard, subsequent days will be okay.

**But to be sure, show me the history graph of your total sensor.**

pschneider87, 2024-08-20:
it looks like it's forced to reset every day
[image]

davidrapan, 2024-08-20:
Ah i see, that could be an issue. What profile are you using?

pschneider87, 2024-08-20:
I'm using deye_2mppt.yaml

My Inverter Modell is not explicitly listed on https://github.com/davidrapan/ha-solarman/wiki/Supported-Devices, but except those total readings, all other sensors are working as expected

my modell is Deye SUN300G3-EU-230 (https://solarv.de/Deye-Micro-Inverter-SUN300G3-EU-230)  including the Die Relay Box (https://epp.solar/product/deye-externes-relais-sun-mi-relay-01-fur-mikrowechselrichter/)

davidrapan, 2024-08-20:
And that reset only happens to the Production sensor or even the rest of them? What version of the integration are you using?
I mean there should already be some checks in place to mitigate this strange behavior of your inverter. Does it drops all the way to the zero?

pschneider87, 2024-08-20:
it only happended to the production sensors.
But yesterday I've updated to 24.08.17
checking the statisics for today, it seems that this morning, the issue was not happening and the sensor is showing no value, which I guess is fine
[image]

davidrapan, 2024-08-20:
Yes! That's how it's supposed to look like for you particular device. There is an issue which you can see but we were able to to somehow correct it by placing constrain on the value that it has to be > 0 otherwise it's not a valid value. I mean it's not the best but at least it works!

pschneider87, 2024-08-20:
ok, good to know it's solved for the future. 
I will try to clean up my statistics manually then.

Thanks for the immediate support :-)

davidrapan, 2024-08-20:
> I will try to clean up my statistics manually then.

Yes that's a tedious job, sadly. I myself had to go trough that route many times so i feel you!

You are very welcome. 🎉

pschneider87, 2024-08-20:
I still hope for a "purge statistics" service/action in the future of HA :-)

closed 2024-08-20 by pschneider87

davidrapan, 2024-08-20:
> I still hope for a "purge statistics" service/action in the future of HA :-)

There is one. Deleting the database. 😆

Yes. I mean when you delete the device you are able to "Fix issue" in the Developer tools STATISTIC page on specific sensor. Which should delete the history. And then you can re-add device back. It should work.