davidrapan/ha-solarman #106: native_value not time data.

straga, 2024-08-21:
```
  File "/opt/iot/app/config/custom_components/solarman/time.py", line 63, in native_value
    return datetime.strptime(self._attr_native_value, "%H:%M").time()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 554, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/_strptime.py", line 333, in _strptime
    raise ValueError("time data %r does not match format %r" %
ValueError: time data '' does not match format '%H:%M'
```

davidrapan, 2024-08-21:
May I ask what profile are you using?

davidrapan, 2024-08-21:
[fix: Empty _native_value or _state return None](https://github.com/davidrapan/ha-solarman/commit/59e941a498420852566aa3364dbe7439b81bd7c4)

davidrapan, 2024-08-21:
[v24.08.21](https://github.com/davidrapan/ha-solarman/releases/tag/v24.08.21)

closed 2024-08-21 by davidrapan